liquidctl/liquidctl #226: Would you accept a PR for supporting the thermaltake floe riing?

lakinwecker, 2020-11-18:
https://github.com/chestm007/linux_thermaltake_riing supports it, but I prefer the interface this project is exposing, and the features this project supports.  It's not yet clear to me whether the same features can be implemented for my cooler, but if they can I think I'd be interested in looking into it.  Of course, that depends on whether you'd accept a PR of this sort.

jonasmalacofilho, 2020-11-18:
Absolutely!

---

License are compatible: as of [`linux_thermaltake_riing@d805e05f0e06`](https://github.com/chestm007/linux_thermaltake_riing/commit/d805e05f0e06beb92d6c95a1a3e9d78a307d51d1) all relevant files are licensed under the GPL-2.0-or-later.

lakinwecker, 2020-11-18:
The likelihood I have time in the near future is slim. So if someone else comes along and wants to work on this instead of me, they should feel free to do so. But I'll try and make time before the new year.

jonasmalacofilho, 2020-11-18:
That's fine.

But can you edit the issue title and description to clarify the specific device (or devices) you're interested in?

lakinwecker, 2020-11-19:
Done. I assume it's the same as many of their other devices, but that's the one I have.

jonasmalacofilho, 2021-02-27:
@lakinwecker is this still something you're interested in working on?

lakinwecker, 2021-02-27:
If I could find the time, yes. But I have no idea when I'll find the time, Feel free to close it and I'll update you when I do find the time.

jonasmalacofilho, 2021-02-27:
Well, feel free to open a PR anytime.  And reach out if you need any help getting started.

closed 2021-02-27 by jonasmalacofilho

redaxmedia, 2021-08-22:
Hey folks, I would like to contribute but have no idea how to start? I found https://github.com/MoshiMoshi0/ttrgbplusapi so this could be a good starting point.