liquidctl/liquidctl #227: Exporting to hwinfo

NitroNokia, 2020-11-21:
Is there some kind of documentation for exporting data to hwinfo?
Everything works, drivers are correct, liquiddump shows data. Can't seem to find easy way to export data out. Im using evga clc280.

status --debug
[debug.txt](https://github.com/jonasmalacofilho/liquidctl/files/5577605/debug.txt)

jonasmalacofilho, 2020-11-21:
Hi @NitroNokia,

HWiNFO supports integration through the Windows Register, and I made some experiments with.  Unfortunately that work has stalled, but please take a look at #101 for a proof of concept and more details.

---

I'm going to close this issue as duplicate of that other one.  Let's continue the conversation there.

closed 2020-11-21 by jonasmalacofilho

NitroNokia, 2020-11-23:
So now its working correctly, i think that the cause of the problem was that i was missing ms build tools or something like that. After installing those, and running LQiNFO.py everything working correctly.