liquidctl/liquidctl #229: Windows: Corsair Hydro H115i

voltagex, 2020-11-22:
I'd really like to change the fan speeds on my 115i without needing iCUE.

```
c:\>C:\Users\Adam\Desktop\liquidctl-1.4.2-bin-windows-x86_64\liquidctl.exe initialize
NotImplementedError: Operation not supported or unimplemented on this platform
[9124] Failed to execute script cli

c:\>C:\Users\Adam\Desktop\liquidctl-1.4.2-bin-windows-x86_64\liquidctl.exe list
Device ID 0: Corsair Hydro H115i

c:\>C:\Users\Adam\Desktop\liquidctl-1.4.2-bin-windows-x86_64\liquidctl.exe status
NotImplementedError: Operation not supported or unimplemented on this platform
[18188] Failed to execute script cli
```

Is this supposed to work?

jonasmalacofilho, 2020-11-22:
Hi!

Please show me the output of each command with the flag `--debug` appended.

voltagex, 2020-11-23:
```
PS C:\Users\Adam\Desktop\liquidctl-1.4.2-bin-windows-x86_64> .\liquidctl.exe --debug initialize; .\liquidctl.exe --debug list ; .\liquidctl.exe --debug status;
[DEBUG] __main__: running liquidctl v1.4.2 (AppVeyor; liquidctl-1.4.2-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:2220
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02fe
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:8d31
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c0a
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Hydro H115i
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:800a
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:8002
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:8d26
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b21:1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02ea
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:8d2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:2220
[DEBUG] __main__: device: Corsair Hydro H115i
Traceback (most recent call last):
  File "liquidctl\cli.py", line 327, in <module>
  File "liquidctl\cli.py", line 297, in main
  File "liquidctl\driver\asetek.py", line 145, in connect
  File "liquidctl\driver\usb.py", line 115, in connect
  File "liquidctl\driver\usb.py", line 241, in open
  File "usb\core.py", line 909, in get_active_configuration
  File "usb\core.py", line 113, in wrapper
  File "usb\core.py", line 247, in get_active_configuration
  File "usb\core.py", line 113, in wrapper
  File "usb\core.py", line 131, in managed_open
  File "usb\backend\libusb1.py", line 804, in open_device
  File "usb\backend\libusb1.py", line 652, in __init__
  File "usb\backend\libusb1.py", line 600, in _check
NotImplementedError: Operation not supported or unimplemented on this platform
[7300] Failed to execute script cli
[DEBUG] __main__: running liquidctl v1.4.2 (AppVeyor; liquidctl-1.4.2-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:2220
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[... 86 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b21:1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02ea
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:8d2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:2220
[DEBUG] __main__: device: Corsair Hydro H115i
Traceback (most recent call last):
  File "liquidctl\cli.py", line 327, in <module>
  File "liquidctl\cli.py", line 297, in main
  File "liquidctl\driver\asetek.py", line 145, in connect
[... cut ...]
  File "usb\core.py", line 113, in wrapper
  File "usb\core.py", line 247, in get_active_configuration
  File "usb\core.py", line 113, in wrapper
  File "usb\core.py", line 131, in managed_open
  File "usb\backend\libusb1.py", line 804, in open_device
  File "usb\backend\libusb1.py", line 652, in __init__
  File "usb\backend\libusb1.py", line 600, in _check
NotImplementedError: Operation not supported or unimplemented on this platform
[1124] Failed to execute script cli
PS C:\Users\Adam\Desktop\liquidctl-1.4.2-bin-windows-x86_64>
```

jonasmalacofilho, 2020-11-23:
This appears to be a problem at the driver level.

Did you notice that this driver requires a special driver, and did you follow the steps to install it?

- [Installing on Windows](https://github.com/jonasmalacofilho/liquidctl#installing-on-windows)
- linked from [Asetek 690LC liquid coolers](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/asetek-690lc-guide.md)
- linked from [Supported devices](https://github.com/jonasmalacofilho/liquidctl#supported-devices)

voltagex, 2020-11-24:
Ah sorry about that.

In my defence, it's only marked as E, not Z on the wiki:

[image]

I can confirm with WinUSB installed via Zadig it's working correctly.

```
Corsair Hydro H115i
├── Liquid temperature       44.1  °C
├── Fan speed                 660  rpm
├── Pump speed               3120  rpm
└── Firmware version      2.8.0.0
```

closed 2020-11-24 by voltagex

jonasmalacofilho, 2020-11-24:
Don't worry about it.

In fact, this was in part caused by Corsair's less-than-ideal naming scheme: you're looking at the wrong line, you have a H115i, not H115i _Platinum._

Anyway, I'm glad it's working now! : )