liquidctl/liquidctl #235: Setting the pump mode outside of initialize.

txf-, 2020-11-27:
```
def initialize(self, pump_mode='balanced', **kwargs):
        """Initialize the device and set the pump mode.
```

the initialize function seems to be the only place where pump mode is set. As the function comment states, initialize doesn't do anything once fans are configured. But it is a bit strange to have to call initialize every time I want to change pump mode. Is there some reason I'm missing that this is done this way?

jonasmalacofilho, 2020-11-28:
There are two reasons for this:
- not having a better *general* API for dealing with this type of device state (yet)
- it is important to put the cooler in a know state during initialization

The second is easily kept if we improve the first: just continue to default to `balanced` mode.

jonasmalacofilho, 2021-02-27:
Superseded by #297.

closed 2021-02-27 by jonasmalacofilho