liquidctl/liquidctl #236: Error running under testing/python (3.9) in Arch Linux

gwstorm, 2020-11-28:
Python 3.9 hit testing in Arch Linux on Nov 26th. Updating to it breaks liquidctl with the following output -

```
Traceback (most recent call last):
  File "/usr/bin/liquidctl", line 33, in <module>
    sys.exit(load_entry_point('liquidctl==1.4.1', 'console_scripts', 'liquidctl')())
  File "/usr/bin/liquidctl", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: liquidctl
```
Downgrading back to stable 3.8.6-1 returns normal functionality. I'm not sure if this is already known or a bit too preemptive, but if you need any further output or checks, I'd be happy to help. All dependencies are installed through a pacman/aur. No pip.

jonasmalacofilho, 2020-11-28:
I'm using Arch as well, I'll give it a try here as soon as I can.

In the meantime, did you remake and reinstall the `liquidct[-git]` package after upgrading to Python 3.9?

gwstorm, 2020-11-28:
I did rebuild liquidctl and I tried to install the git package, but it actually errored out as well. The git package does still successfully build under python 3.8 though. I don' t have that output under 3.9 at the moment. It ended with something about pytest. It looks like 3.9 has a *lot* of python packages adjusting and the only real sane way for me to get back to normal with gkraken was to downgrade everything back to stable. 

Everything is still running smoothly on stable though, so I don't feel that there's any extreme urgency to this. I'll happily offer up whatever help I can though as bouncing between stable and testing is only a matter of minutes for me.

jonasmalacofilho, 2020-11-28:
I was able to reproduce this, as well as confirm that a _successful_ rebuild (and reinstall) fixes the problem.

_(This is common and all Python packages in the official repositories will have been rebuilt by the time `python` hits `extra`)._

However, you mentioned that liquidctl isn't building on 3.9 due to errors during the `check()` phase.  Can you post the output here when you have some time?

Unfortunately I was testing this on a `archlinux:base-devel` container and my `check()` function fails on 3.8 as well, due to a myriad of permission issues.

gwstorm, 2020-11-28:
I just upgraded to testing again and tried a clean rebuild of `liquidctl-git` and this was the result, which is the same as what I received before -

```
==> Making package: liquidctl-git 1.4.2.r76.90d7ec5-1 (Fri 27 Nov 2020 11:22:12 PM PST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting pkgver()...
==> Starting build()...
running build
running build_py
creating build
creating build/lib
creating build/lib/liquidctl
copying liquidctl/cli.py -> build/lib/liquidctl
copying liquidctl/error.py -> build/lib/liquidctl
copying liquidctl/pmbus.py -> build/lib/liquidctl
copying liquidctl/keyval.py -> build/lib/liquidctl
copying liquidctl/version.py -> build/lib/liquidctl
copying liquidctl/util.py -> build/lib/liquidctl
copying liquidctl/extraversion.py -> build/lib/liquidctl
copying liquidctl/__init__.py -> build/lib/liquidctl
creating build/lib/liquidctl/driver
copying liquidctl/driver/usb.py -> build/lib/liquidctl/driver
copying liquidctl/driver/nvidia.py -> build/lib/liquidctl/driver
copying liquidctl/driver/kraken3.py -> build/lib/liquidctl/driver
copying liquidctl/driver/corsair_hid_psu.py -> build/lib/liquidctl/driver
copying liquidctl/driver/kraken2.py -> build/lib/liquidctl/driver
copying liquidctl/driver/rgb_fusion2.py -> build/lib/liquidctl/driver
copying liquidctl/driver/smart_device.py -> build/lib/liquidctl/driver
copying liquidctl/driver/base.py -> build/lib/liquidctl/driver
copying liquidctl/driver/nzxt_epsu.py -> build/lib/liquidctl/driver
copying liquidctl/driver/hydro_platinum.py -> build/lib/liquidctl/driver
copying liquidctl/driver/smbus.py -> build/lib/liquidctl/driver
copying liquidctl/driver/__init__.py -> build/lib/liquidctl/driver
copying liquidctl/driver/asetek.py -> build/lib/liquidctl/driver
==> Starting check()...
Traceback (most recent call last):
  File "/usr/bin/pytest", line 33, in <module>
    sys.exit(load_entry_point('pytest==6.1.2', 'console_scripts', 'pytest')())
  File "/usr/bin/pytest", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: pytest
==> ERROR: A failure occurred in check().
    Aborting...
error making: liquidctl-git
```

jonasmalacofilho, 2020-11-28:
Oh, it seems that `pytest` needs its rebuild too : )

You can rebuild it locally (for example, using the ABS), but then you would need to manually downgrade it when going back to stable.

P.S. our tests have been running on Python 3.9 in CI, and barring some Arch-specific issue, as soon as `pytest` can run, `check()` should pass without additional issues.

gwstorm, 2020-11-28:
Oh, well in that case it looks like everything will be just fine in the end. Sorry for all of that. Liquidctl is pretty vital for me so I get a bit nervous when issues come up!

Thanks for your time and prompt replies.

closed 2020-11-28 by gwstorm

jonasmalacofilho, 2020-11-28:
> Sorry for all of that. Liquidctl is pretty vital for me so I get a bit nervous when issues come up!

I completely understand.  And thanks for letting me know about the problem, regardless of whether it will (probably) be gone once 3.9 is out of `testing` (and we trigger a rebuild of the AUR packages).

By the way, I'll leave the issue pinned until that happens, just in case someone else also stumbles on this.

jonasmalacofilho, 2020-12-02:
Python 3.9 arrived in `extra` today; was able to rebuild liquidctl (with the `check` function unchanged) without issues.

gwstorm, 2020-12-02:
Confirmed that it worked for me. Full system upgrade and a rebuild of `liquidctl-git` and everything is great. 

Thanks for maintaining this package!

jonasmalacofilho, 2020-12-02:
Thanks!

For others coming here, `liquidctl` (non `-git`) should work just as well.