liquidctl/liquidctl #240: Add support for Corsair Hydro H100x

gabrik, 2020-12-03:
First thanks for this awesome tool.

It would be possible to add this Corsair AIO?

I can give all the needed help and test it.

My current setup is:

Debian sid - 5.9.0-4-amd64
System is AMD Ryzen 9 3900X on ASRock X570 PhantomGaming 4 

I can see the fans and the pump (I guess) from `lm-sensors'

Thanks!

jonasmalacofilho, 2020-12-04:
Hi @gabrik!

Since you're connecting the fans and pump to the motherboard, what is really necessary is support for that board.  Presumably for the Super I/O chip it's using.  Do you happen to know which is it?

Regardless, supporting this type of thing is something I'm still evaluating: one hand we're now experimenting with non-USB devices, so that's a maybe; on the other, Super I/O chips (really) don't support concurrent access, and given that limitation Linux hwmon drivers would be preferred over userspace tooks (like liquidctl) anyway.

gabrik, 2020-12-11:
Hi @jonasmalacofilho 

From sensors I can see this:

```
$ sensors
nct6798-isa-0290
Adapter: ISA adapter
in0:                   704.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     1.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                     1.10 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                     1.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                     1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                    1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  632.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  928.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                  912.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   618 RPM  (min =    0 RPM)
fan3:                  4090 RPM  (min =    0 RPM)
fan4:                   522 RPM  (min =    0 RPM)
fan5:                   358 RPM  (min =    0 RPM)
fan6:                  1829 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +28.5°C    sensor = thermistor
CPUTIN:                 +32.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                 +8.0°C    sensor = thermistor
AUXTIN1:                -62.0°C    sensor = thermistor
AUXTIN2:                +13.0°C    sensor = thermistor
AUXTIN3:                +31.0°C    sensor = thermistor
SMBUSMASTER 1:          +49.0°C  (high = +105.0°C, hyst = +95.0°C)
SMBUSMASTER 0:          +44.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
Vcore:         1.46 V  
Vsoc:          1.09 V  
Tctl:         +44.5°C  
Tdie:         +44.5°C  
Tccd1:        +39.0°C  
Tccd2:        +35.5°C  
Icore:        15.00 A  
Isoc:          6.25 A  
```

Also `pwmconfig` reports:

```
...
   hwmon0 is nct6798
   hwmon1 is k10temp
...
```

So I guess the chip is `nct6798`, also with `fancontrol` I can adjust the speed of the fans and the pump but it does not five a very fine control, or at least I wasn't able to make a nice curve, that is, for the time being I'm using the curve configured in the bios.

Please let me know if  you need me to run some specific command to get more information

jonasmalacofilho, 2020-12-11:
It's very unlikely that we'll be able to do much better than the nct6798 driver.  What we could explore is using `liquidctl` as a (possibly better) alternative to `fancontrol` for standard hwmon drivers.  In your opinion, what fine control knobs are lacking in `fancontrol`?

gabrik, 2020-12-15:
When using `fancontrol` I was able to set only a few parameters:

- Temperature at which fan should stop
- Temperature at which fan should start spinning
- Initial fan RPM
- Temperature at which fan should be at maximum speed

Then it does linear interpolation between the points to calculate the right PWM and hence fan speed.

It also does not have any way to know if a "fan" is actually a "pump".

In my opinion, it would be nice to the possibility to set more points in order to create a custom curve for both fans and pumps.

This is an example `fancontrol` configuration file for another machine where I'm using it

```
INTERVAL=2

FCTEMPS=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=/sys/class/hwmon/hwmon1/temp1_input /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=/sys/class/hwmon/hwmon1/temp1_input
FCFANS=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/fan3_input /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/fan1_input
MINTEMP=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=50 /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=40
MAXTEMP=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=70 /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=65
MINSTART=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=100 /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=150
MINSTOP=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=0 /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=0
MAXPWM=/sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm3=255 /sys/devices/pci0000:00/0000:00:1f.4/i2c-0/0-002d/hwmon/hwmon2/pwm1=255
```

jonasmalacofilho, 2020-12-16:
I need to look into `fancontrol`.  I know it exists, and I know what the hwmon interfaces expose, but I don't have any first time experience trying to use it.

> In my opinion, it would be nice to the possibility to set more points in order to create a custom curve for both fans and pumps.

That said, our CLI only exposes chip-native features.  While with our APIs it's possible to extend those features, like implementing more fine grained profiles than what the hardware supports, this wouldn't be accessible through the CLI.  The CLI hands off a profile for the device, and immediately exists: that profile will run on (and with the limitations of) the device.

Because of this, our CLI would still be severely constrained by the number of PWM inflection points that the chip supports, if it even natively supports profiles.  Can you post the output for `sensors -u 'nct6798-*'`?

gabrik, 2020-12-21:
@jonasmalacofilho sorry for the delay, here you go:

```
$ sensors -u 'nct6798-*'
nct6798-isa-0290
Adapter: ISA adapter
in0:
  in0_input: 0.096
  in0_min: 0.000
  in0_max: 1.744
  in0_alarm: 0.000
  in0_beep: 0.000
in1:
  in1_input: 1.672
  in1_min: 0.000
  in1_max: 0.000
  in1_alarm: 1.000
  in1_beep: 0.000
in2:
  in2_input: 3.392
  in2_min: 0.000
  in2_max: 0.000
  in2_alarm: 1.000
  in2_beep: 0.000
in3:
  in3_input: 3.312
  in3_min: 0.000
  in3_max: 0.000
  in3_alarm: 1.000
  in3_beep: 0.000
in4:
  in4_input: 1.832
  in4_min: 0.000
  in4_max: 0.000
  in4_alarm: 1.000
  in4_beep: 0.000
in5:
  in5_input: 1.096
  in5_min: 0.000
  in5_max: 0.000
  in5_alarm: 1.000
  in5_beep: 0.000
in6:
  in6_input: 1.344
  in6_min: 0.000
  in6_max: 0.000
  in6_alarm: 1.000
  in6_beep: 0.000
in7:
  in7_input: 3.392
  in7_min: 0.000
  in7_max: 0.000
  in7_alarm: 1.000
  in7_beep: 0.000
in8:
  in8_input: 3.216
  in8_min: 0.000
  in8_max: 0.000
  in8_alarm: 1.000
  in8_beep: 0.000
in9:
  in9_input: 1.656
  in9_min: 0.000
  in9_max: 0.000
  in9_alarm: 1.000
  in9_beep: 0.000
in10:
  in10_input: 1.024
  in10_min: 0.000
  in10_max: 0.000
  in10_alarm: 1.000
  in10_beep: 0.000
in11:
  in11_input: 0.632
  in11_min: 0.000
  in11_max: 0.000
  in11_alarm: 1.000
  in11_beep: 0.000
in12:
  in12_input: 1.032
  in12_min: 0.000
  in12_max: 0.000
  in12_alarm: 1.000
  in12_beep: 0.000
in13:
  in13_input: 0.928
  in13_min: 0.000
  in13_max: 0.000
  in13_alarm: 1.000
  in13_beep: 0.000
in14:
  in14_input: 0.912
  in14_min: 0.000
  in14_max: 0.000
  in14_alarm: 1.000
  in14_beep: 0.000
fan1:
  fan1_input: 0.000
  fan1_min: 0.000
  fan1_alarm: 0.000
  fan1_beep: 0.000
  fan1_pulses: 2.000
fan2:
  fan2_input: 609.000
  fan2_min: 0.000
  fan2_alarm: 0.000
  fan2_beep: 0.000
  fan2_pulses: 2.000
fan3:
  fan3_input: 4166.000
  fan3_min: 0.000
  fan3_alarm: 0.000
  fan3_beep: 0.000
  fan3_pulses: 2.000
fan4:
  fan4_input: 521.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 0.000
  fan4_pulses: 2.000
fan5:
  fan5_input: 359.000
  fan5_min: 0.000
  fan5_alarm: 0.000
  fan5_beep: 0.000
  fan5_pulses: 2.000
fan6:
  fan6_input: 1252.000
  fan6_min: 0.000
  fan6_alarm: 0.000
  fan6_pulses: 2.000
fan7:
  fan7_input: 0.000
  fan7_min: 0.000
  fan7_pulses: 2.000
SYSTIN:
  temp1_input: 26.000
  temp1_type: 4.000
  temp1_offset: 0.000
CPUTIN:
  temp2_input: 25.500
  temp2_max: 80.000
  temp2_max_hyst: 75.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
  temp2_beep: 0.000
AUXTIN0:
  temp3_input: 8.000
  temp3_type: 4.000
  temp3_offset: 0.000
AUXTIN1:
  temp4_input: -62.000
  temp4_type: 4.000
  temp4_offset: 0.000
AUXTIN2:
  temp5_input: 13.000
  temp5_type: 4.000
  temp5_offset: 0.000
AUXTIN3:
  temp6_input: 31.000
  temp6_type: 4.000
  temp6_offset: 0.000
SMBUSMASTER 1:
  temp7_input: 41.000
  temp7_max: 105.000
  temp7_max_hyst: 95.000
  temp7_alarm: 0.000
  temp7_beep: 0.000
SMBUSMASTER 0:
  temp8_input: 42.000
PCH_CHIP_CPU_MAX_TEMP:
  temp9_input: 0.000
PCH_CHIP_TEMP:
  temp10_input: 0.000
intrusion0:
  intrusion0_alarm: 1.000
  intrusion0_beep: 0.000
intrusion1:
  intrusion1_alarm: 1.000
  intrusion1_beep: 0.000
beep_enable:
  beep_enable: 0.000
  
  ```

jonasmalacofilho, 2020-12-25:
I'm the one that's sorry, I totally messed up and the raw output of `sensors` wouldn't show me what I wanted to see anyway.

---

Still, having liquidctl support the NCT6798 chip wouldn't really help you very much (like I mentioned, the PWM inflection points would be the same, since they are defined at the hardware level).  And the chip (itself) is already well supported on Linux.

Writing a better alternative to `fancontrol` (that is, a better software to dynamically leverage hwmon PWM devices) is currently outside the scope of the core of the project.  We do have [`extra/yoda`](https://github.com/liquidctl/liquidctl/blob/master/extra/yoda), but it's not a top priority for me.

You're welcome to join our team and work on that... is this something you would be interested in/able to do?

gabrik, 2020-12-28:
@jonasmalacofilho sorry I'm not able for the time being to join the team.
It is something I'm interested in but I do not have much spare time to dedicate to it. 

I may take a more deep look at it and come with some PR once I have more spare time.

jonasmalacofilho, 2020-12-28:
That's fine.

Anyway, I'm going to close this issue since there isn't anything specific to the NCT6798 we can do.

closed 2020-12-28 by jonasmalacofilho