liquidctl/liquidctl #242: Coursair H100i RGB on during boot but not once liquidctl is called

brandon-feder, 2020-12-10:
Hello. When my computer is booting, both the fans ( and the pump ) light up with there default RGB barf. However, once the liquidcfg service is started, only one fan (and the pump) light up while the other fan does not. Both fans are spinning, but only one lights up. I am not sure whether this is a hardware or software issue, but I was hoping that if it is a software issue, that someone would help. I am running Manjaro Linux if that matters. Any other info you need?

Thanks in advance,
Brandon Feder

jonasmalacofilho, 2020-12-10:
Hi @brandon-feder,

What specific variant of the H100i do you have?  Can you also post which specific liquidctl commands you are using, as well as their corresponding output with `--debug`?

MarshallAsch, 2020-12-10:
Not quite related to your immediate problem but there is a fix in progress for that "default RGB barf" that shows up while the computer is booting. (see #193 for more details on that.)

brandon-feder, 2020-12-11:
Hello. Thanks for your reply. I have a H100i RGB PLATINUM SE. The only commands I run are `sudo liquidctl initialize` and `sudo liquidctl set led color fixed ffffff`. Its output using the `--debug` flag is the following:

```
[DEBUG] liquidctl.cli: running liquidctl v1.4.2
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Platinum SE (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a12:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H100i Platinum SE (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c19/loc0
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in /var/run/liquidctl/vid1b1c_pid0c19/loc0/leds_enabled)
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in /var/run/liquidctl/vid1b1c_pid0c19/loc0/pump_mode)
[DEBUG] liquidctl.keyval: no data (file) found for fan1_mode
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)
[DEBUG] liquidctl.keyval: no data (file) found for fan1_duty
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 100% duty cycle
[DEBUG] liquidctl.keyval: no data (file) found for fan2_mode
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)
[DEBUG] liquidctl.keyval: no data (file) found for fan2_duty
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 100% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=8 (from /var/run/liquidctl/vid1b1c_pid0c19/loc0/sequence)
[DEBUG] liquidctl.keyval: stored sequence=9 (in /var/run/liquidctl/vid1b1c_pid0c19/loc0/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:48:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:ff:02:00:00:00:00:ff:01:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:12
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:48:11:0f:00:09:00:60:1c:14:02:ff:e8:03:3d:55:01:02:ff:e8:03:3d:5c:01:01:aa:00:00:aa:08:09:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:e6
Corsair H100i Platinum SE (experimental)
└── Firmware version    1.1.15
```

and

```
[DEBUG] liquidctl.cli: running liquidctl v1.4.2
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[... 28 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[... cut ...]
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:3d:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:c4
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:38:11:0f:00:07:00:34:1d:00:00:33:e8:03:33:56:01:00:33:e8:03:33:5a:01:01:aa:00:00:aa:0d:09:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:46
```

Also, thanks @MarshallAsch. I will look into this.

MarshallAsch, 2020-12-11:
all of that output looks correct. What fan types do you have connected? Is it the two ML fans that came with the cooler or did you switch to different fans? 

Is this a new issue (it worked before and has since stoped working) or is this the first time you are using `liquidctl`?

brandon-feder, 2020-12-11:
They are the fans that came with the computer, and this issue just started when I upgraded my motherboard. I have used liquidctl for a while now.

MarshallAsch, 2020-12-11:
Does it work properly using icue in windows (either as a dual boot or in a vm)?

brandon-feder, 2020-12-11:
Yes it does. In my Windows VM, I tried it and both fans and the cooler work just as expected.

MarshallAsch, 2020-12-11:
hmm thats odd. I just checked it on mine. (the non SE version) and everything worked as expected and it is using the same firmware version. I'll look at this some more over the next couple days and see what I can come up with. 

Aside from your motherboard upgrade did anything else change?

brandon-feder, 2020-12-11:
I changed nothing else. I possible moved some cables around but all the components are identical. Thanks for your help. And no rush. It's not that important that the RGB works if the fans are spinning.

jonasmalacofilho, 2020-12-11:
@brandon-feder

Back in #170 you were using a git version of liquidctl from before 1.4.1.  Could you rule out this issue being a regression in 1.4.1 or 1.4.2 (by going back to previous versions/builds)?

Apart from that your power saving/ErP motherboard settings may be different now, and could be wiping some state liquidctl was relying on being already set.  Do you think you can provide some USB [traffic captures](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md), preferably made from Linux with iCue in a VM so we can capture the device being enumerated and initialized by both Windows and iCue?

brandon-feder, 2020-12-13:
Hello. Sorry for not getting back to you sooner. I will try to get you this traffic captures by later today or tomorrow. I appreciate your help.

brandon-feder, 2020-12-14:
Hello. I downgraded liquidctl to a few previous versions. However, because support for my cooler is relatively knew, I did not go back that far. Id also like to note that liquidctl worked perfectly fine before I upgraded my motherboard and that unless there was an update in the hour or two it took to swap the boards, I doubt that the version is the problem. Either way, I can attach the Wireshark data. However, before I do that I just want to make sure there is no private information I should try to filter out. Is there? Thanks!

jonasmalacofilho, 2020-12-15:
To prevent most private information from leaking, make sure to filter the packets to only those from/to the cooler.

One simple way to do that is to first need to figure out the device's address number with lsusb.  Taking as an example my Kraken cooler:

```
$ lsusb
[... other devices ...]
Bus 001 Device 004: ID 1e71:170e NZXT Kraken X
    ^^^        ^^^
```

With this information you can capture only the desired bus (in this case by selecting the `usbmon1` input) and device (using filter `usb.device_address == 4`).

Alternatively you could filter your capture afterwards, with the same expression, by using File -> Export Specified Packets -> **Displayed**.

---

At this point the only private information that could remain is the cooler's serial number, either as a USB descriptor or in one of the packets, assuming the device even reports it.  If you're worried about this, please send me an email and I can try to accommodate you, either by scrubbing the data myself or by helping you do it.  Though I can't assure I'll be able to find *all* instances of a serial number (particularly because I don't know what the serial number looks like for these devices), and it may take me a while to even attempt to do that.

jonasmalacofilho, 2021-01-22:
@brandon-feder 

I'm not sure you got my reply to your email: the captures you sent me were of network traffic, not USB traffic.

MarshallAsch, 2021-01-22:
This is the same issue as #267 (I just noticed that that one is a duplicate). And it is because the `Corsair H100i Platinum SE` cooler has LL fans that have 16 LED's and not 4. It will be fixed once #269 gets merged.

commit 4162d2e8 (2021-01-22): Merge #269: hydro_platinum: handle the different number of LEDs/fan

Closes: #242
closed 2021-01-22 by jonasmalacofilho