liquidctl/liquidctl #244: Support for EVGA XC3 Ultra RTX 3090?

gardotd426, 2020-12-20:
Hey, so I was completely shocked to see the new support for the EVGA FTW3 1080 and the ROG Strix GPUs after adding the new SMBUS dependency, as well as motherboards and all sorts of new stuff. Well, that got me wondering if it would be possible to at least look into adding the support for the XC3 Ultra RTX 3090 from EVGA. I doubt you remember but I was one of those 3 or 4 people that had X*3 AIOs when you were initially adding support and was helping test out the experimental builds, and I didn't know if this would be easier or harder than that was, because that got done pretty quick. Anyway, it's no big deal, I'm just curious and I'm more than happy to help test. 

Here's my `i2cdetect -l`

```
i2c-3	smbus     	SMBus NCT67xx adapter at 02a0   	SMBus adapter
i2c-1	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-8	i2c       	NVIDIA i2c adapter 8 at f:00.0  	I2C adapter
i2c-6	i2c       	NVIDIA i2c adapter 6 at f:00.0  	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 1 at f:00.0  	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-0	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-7	i2c       	NVIDIA i2c adapter 7 at f:00.0  	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 5 at f:00.0  	I2C adapter
```

The PIIX4s are my G.Skill Trident Z RGB RAM sticks, just FYI.

jonasmalacofilho, 2020-12-20:
Oh, hi again!

With SMBus/I²C devices we're making sure that the drivers are developed by someone with access to the actual hardware, since testing this stuff properly is very important.

If can work on this, it shouldn't be very hard at all.  And we'll obviously help, including reviewing your code before you actually have to run it...  feel free to join our [developer's server on Discord](https://discord.gg/eJ5MswXSf9).

Basically, the first step is to capture the I²C traffic (usually to some address on "NVIDIA i2c adapter 1") with [NvAPISpy](https://gitlab.com/CalcProgrammer1/NvAPISpy) and document the protocol.  After that the implementation should be fairly trivial.

gardotd426, 2020-12-20:
Awesome, I'll check all that out and hopefully see you on the Discord

jonasmalacofilho, 2021-02-27:
@gardotd426 any news?  Do you need some help?

gardotd426, 2021-02-27:
Yeah, I tried to use NVapiSpy and it didn't do anything, it didn't work at all, and I told Adam (CalcProgrammer1) and he couldn't really help much. I guess I could try and give it another go, but yeah NVApiSpy won't work w/ Precision X1 I tried it about 8 times.

jonasmalacofilho, 2021-02-27:
I had some difficulty too, but it worked once I replaced it globally.

I think X1 may use a hardcoded path for the system DLL, instead of going through the system loader and various candidate paths.

gardotd426, 2021-02-27:
Yeah that was one of the suggestions Adam made, I'll give it a try globally, I'll just have to install Windows on an extra partition. I have Windows in a single-GPU passthrough VM, but I don't want to mess with anything like this in VFIO even though the GPU is basically running as if it were bare metal. I wiped the Windows install I tried this on previously because it was the only reason I even installed Windows in the first place, but I already have an iso on a Ventoy USB drive and was already needing to install Windows to update some firmware anyway, so yeah I'll give it another shot.

jonasmalacofilho, 2021-08-06:
@gardotd426 how is it going?

ZetaPhoenix, 2022-03-02:
@jonasmalacofilho Reference code for the EVGA Ampere controllers here: https://gitlab.com/CalcProgrammer1/OpenRGB/-/tree/master/Controllers/EVGAAmpereGPUController

gardotd426, 2022-03-02:
Well from what I've been told by the OpenRGB folks, Nvidia doesn't allow writing to the necessary i2c bus required in order for there to be RGB control. That said, I'll definitely give this a shot and report back. If anyone else has been able to get RGB control on an EVGA Ampere GPU (preferably an XC3 Ultra RTX 3090) that would be good to know

jonasmalacofilho, 2022-03-02:
That doesn't appear to be the case, at least at the hardware level:

https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/d8f28b546dbdec5826774852a49a2767ffdff3cb/Controllers/EVGAAmpereGPUController/EVGAAmpereGPUControllerDetect.cpp#L61

Maybe they meant that some platforms, and some of NVIDIA's drivers for those platforms, may be blocking that I2C bus?

Anyway, OpenRGB does appear to support that card, at least somewhere...

@ZetaPhoenix, any idea if there are any limitations on this card?

---

@gardotd426 , if I'm not mistaken @ZetaPhoenix is TheRogueZeta, one of maintainers (or at least a regular contributor) of OpenRGB.

ZetaPhoenix, 2022-03-02:
Um, no per-led control yet and Linux support may be possible with `i2ctransfer` like how the Gigabyte cards are handled in Linux.

gardotd426, 2022-03-02:
@jonasmalacofilho I'm typing this on my phone right now so I don't have easy access to the pull request thread and issue thread at OpenRGB but the support for the XC3 Ultra you referenced was Windows only. I remember commenting on the PR at the time to confirm that it still wasn't supported under Linux. I have to record a podcast with Intelligent Gaming right now but when I'm done I'll actually snag the relevant threads

ZetaPhoenix, 2022-03-02:
@gardotd426 are you on the OpenRGB discord? There is a thread I just created to discuss this.
https://discord.com/channels/699861463375937578/941201973192065125

jonasmalacofilho, 2022-07-03:
Triage: closing as the consensus is that this is not possible to implement with the current NVIDIA Linux drivers.  Having followed the discussions on OpenRGB, I still have a tiny bit of hope that a non-obvious I2C transfer type might work, but various people looking into this haven't found it yet...

closed 2022-07-03 by jonasmalacofilho

gardotd426, 2022-07-03:
I apologize for not being available to provide the testing work necessary to see if we can get this working. I'm the sole person behind the Regolith Desktop Environment port to Arch Linux, and v2.0 launched recently and I've been preparing for it for months and now I'm all over the place trying to get everything done (2.0 basically completely reworks how Regolith works on a fundamental level, major PITA).

Once I'm done I'll try and see if I can't get anywhere with this.

jonasmalacofilho, 2022-07-03:
> I'm the sole person behind the Regolith Desktop Environment port to Arch Linux [...]

Oh, that's really cool!  I definitely want to check it out when I have some time...

---

> Once I'm done I'll try and see if I can't get anywhere with this.

Sure, and feel free to ping me about it (either here or on Discord).  If there's a solution for the I2C issue, the rest looks like it wouldn't be hard to implement.

ZetaPhoenix, 2022-07-03:
FYI, this is related: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/41