liquidctl/liquidctl #245: all liquidctl command except "liquidctl list" doesn't work in my computer

JeongJae0815, 2020-12-23:


jonasmalacofilho, 2020-12-23:
Starting with the first error:

```
UnicodeEncodeError: 'ascii' codec can't [...]
```

`liquidctl` cannot be executed under `LC_ALL=C` and Python 3.6, because non-ASCII characters are used in the output (see #84).

Also consider that `LC_ALL=C` is not really supposed to be used indiscriminately/globally:

> Using LC_ALL is strongly discouraged as it overrides everything. Please use it only when testing and never set it in a startup file. 
> _[Locale - Debian Wiki](https://wiki.debian.org/Locale)_

---

The second error is utterly unexpected.  Can you run

```
PYUSB_DEBUG=debug LIBUSB_DEBUG=4 liquidctl status --debug
```

to gather more information and show me the output?

jonasmalacofilho, 2020-12-23:
Additionally, do you really have *two* H100i v2 coolers on that system?

MarshallAsch, 2020-12-25:
It also did not seem to have been run as root, did you also install  the `udev` rules to allow non root access? Could this be an odd permissions error?

jonasmalacofilho, 2020-12-25:
@MarshallAsch

That's possible... and it would make sense.

JeongJae0815, 2020-12-28:
@jonasmalacofilho @MarshallAsch 
 I really appreciate you responding quickly.

> Starting with the first error:
>
> ```
> UnicodeEncodeError: 'ascii' codec can't [...]
> ```
>
> `liquidctl` cannot be executed under `LC_ALL=C` and Python 3.6, because non-ASCII characters are used in the [...]
I can recognize my system's python version is lower than 3.6

jonasmalacofilho, 2020-12-28:
> I recognized my system's python version is lower than 3.6.(default python version in ubuntu16.04 is lower than 3.6),

Oh... I only noticed that the first screenshot showed Python 3.6 paths but, indeed, the later dumps don't specify it (and Ubuntu 16.04 is quite old by now).

> Since I've installed liquidctl outside of docker, everything does work well.

Good to know! : )

I'll close this then...

closed 2020-12-28 by jonasmalacofilho