liquidctl/liquidctl #246: Fan curve support for NZXT Kraken x61 on Windows?

ROODAY, 2020-12-24:
Looking through the issues I found this https://github.com/liquidctl/liquidctl/blob/master/extra/krakencurve-poc which says it supports the x62 but doesn't explicitly say x61, nor does it have a sensor option for Windows. Does anyone have a workaround, or could point me in the direction of getting it working? I don't mind having to write a little code, but I'm not sure where to start (never dealt with this type of stuff before). I followed the install guide and got it working (used the WinUSB driver and I append `--legacy-690lc` to all commands), I would just like to be able to set a software based curve and forget about it (I like having the fan as quiet as possible until it's needed). Would I also need to do extra steps to have it start on boot?

jonasmalacofilho, 2020-12-25:
We haven't been able to implement support for fan profiles on legacy Asetek 690LC devices like the X61.

These devices are painful to work with, but take a look at [#35 (comment)](https://github.com/liquidctl/liquidctl/issues/35#issuecomment-529792488).  Note however that's a very old issue, and so are any patches attached there.

---

A simpler alternative for you might be to run the fan profiles in software, with something like [`extra/yoda`](https://github.com/liquidctl/liquidctl/blob/master/extra/yoda).

ROODAY, 2020-12-25:
@jonasmalacofilho I don't mind running in software (what's the difference between software and hardware, just that hardware is faster/more accurate?) but my concern was that for extra/yoda and extra/krakencurve-poc it says that there are no system sensors supported for Windows. Will that be an issue, or will it still work out of the box? (i.e. be able to read temps and adjust fan/pump accordingly)

P.S. Happy Holidays!

jonasmalacofilho, 2020-12-25:
@ROODAY happy holidays!

By software I meant on your host/main CPU, instead of the controllers embedded CPU.  The main advantage of the latter is that it can be more reliable, since it's isolated from the rest of the system.  Also, `yoda` is experimental and not yet battle tested.

Still, while `yoda` wont have access to system sensors on Windows, you'll still be able to set up a profile based on the liquid temperature.  And "hardware" profiles also have the same limitation anyway.

ROODAY, 2020-12-26:
Is there a specific DLL version I need to get yoda to work with the X61, similar to how I needed to install the WinUSB driver for the X61? I followed the install instructions for setting up liquidctl with python and putting the libusb dll in my python install (I used the latest release and the VS2019/MS64 folder), but when I try running show sensors I get the following error:
```
λ python C:\liquidctl-1.4.2-bin-windows-x86_64\yoda.py show-sensors
Sensor identifier                                                    Temperature
--------------------------------------------------------------------------------
usb.core.USBTimeoutError: [Errno 10060] Operation timed out
```

Do I need to use the WinUSB driver instead? And if so, is there any config I need to do to tell yoda to use that driver instead?

jonasmalacofilho, 2020-12-26:
`yoda` has the same driver requirements as liquidctl.  The issue is likely caused by not having passed `--legacy-690lc`.

You'll need to do a cold boot (turn off the PSU and wait for any caps to completely drain) to make the cooler responsive again.  After that

```
> python C:\liquidctl-1.4.2-bin-windows-x86_64\yoda.py --legacy-690lc show-sensors
```

should display the internal coolant temperature sensor.

ROODAY, 2020-12-27:
Ah my bad, I missed that note in the yoda.py args. Last question, is this supposed to run indefinitely? I ended up running this (as close to the NZXT silent profile as I could remember):
```
λ python C:\liquidctl-1.4.2-bin-windows-x86_64\yoda.py --legacy-690lc --match kraken control pump with "(20,50),(40,65),(60,85),(80,100)" on _internal.liquid and fan with "(20,20),(40,50),(60,75),(80,100)" on _internal.liquid
```
but it never gave any output, is that because it's polling in the background? If so, is there a recommended way of running it as a service in Windows?

jonasmalacofilho, 2020-12-27:
> Last question, is this supposed to run indefinitely

Yes, since yoda is controlling the profile, it has run continuously.  Updated fan/pump speeds will be sent to the cooler at `--interval` intervals.

> If so, is there a recommended way of running it as a service in Windows?

Probably with Task Manager, although I admit I haven't given it much thought.

Do you know how system services are set up on Windows (from the perspective of program authors)?

ROODAY, 2020-12-27:
> Do you know how system services are set up on Windows (from the perspective of program authors)?

Unfortunately not, most of my experience is in web-dev so I haven't made any Windows apps. Sad that Windows doesn't have something as simple as systemd. I did find [this SO post](https://stackoverflow.com/questions/32404/how-do-you-run-a-python-script-as-a-service-in-windows) though, I'll comment again once I figure out a good method!

ROODAY, 2020-12-28:
@jonasmalacofilho I think I got it running as a service however I can't verify the status. From that SO post, I downloaded [NSSM](http://nssm.cc/download) (the pre-release version since I'm on the latest Win 10) and added it to my path. I then ran `nssm.exe install WinService`, and then installed with the following args:
```
path: c:\Users\rudyt\AppData\Local\Programs\Python\Python37\python.exe
Startup directory: c:\Users\rudyt\AppData\Local\Programs\Python\Python37
Arguments: C:\liquidctl-1.4.2-bin-windows-x86_64\yoda.py --legacy-690lc --match kraken control pump with "(20,50),(40,65),(60,85),(80,100)" on _internal.liquid and fan with "(20,20),(40,50),(60,75),(80,100)" on _internal.liquid
```
Then I went into Windows services and found the service it installed (I named it LiquidctlYoda) and set it to start. I heard the fan get quieter (I set it to 100% first so any change would be audible). So this makes me believe the service is working as intended.

However what I meant earlier by I can't verify the status, is that while yoda is running it seems to lock access to the device, as I get this error:
```
λ liquidctl status --legacy-690lc
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
```
I tried running in an administrator terminal as well and got the same error, but commands start working again if I stop the Yoda service. Is there a way around this/check the status of the device while Yoda is running? Also, do I still need to run `liquidctl initialize --legacy-690lc` at system start before yoda? (in which case I might just edit Yoda to run that command first)

jonasmalacofilho, 2020-12-28:
> Is there a way around this/check the status of the device while Yoda is running

I checked our driver and we apparently already do our best to release the device as soon and as often as possible.  So I don't think there's anything else we can do, it seems to be a Windows thing...

> Also, do I still need to run `liquidctl initialize --legacy-690lc` at system start before yoda?

Yes, it's still necessary.

ROODAY, 2020-12-28:
Dang, that's unfortunate. I tried getting NSSM to log output to a file so I could least check up on it from time to time but it caused issues when restarting yoda manually (not sure if that will really be an issue though as long as I don't touch it), so I guess that's the closest I can get to viewing status.

As for the part with initialize, that was an easy fix actually by just creating another service with NSSM that is just a bat file that runs initialize, and setting it as a dependency of the Yoda service so it will always run first. I think this is as close to perfect as I can get, thanks for all the help!

jonasmalacofilho, 2020-12-30:
I'm reopening this because I want to check the `PyUsbDevice.release()` behavior on Windows before the next release.

---

Additionally, these inconsistent permission errors are very weird.  Can you try to use`PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` to figure out what's different between runs with and without `--verbose`?

ROODAY, 2020-12-30:
@jonasmalacofilho Will do! I'm busy for the next couple days, just wanted to let you know I saw this!

ROODAY, 2021-01-08:
@jonasmalacofilho Sorry for the delay, I've been prepping for a move, which incidentally means I won't have access to my desktop for a couple weeks while the moving company ships it. In regards to your last comment, running with the `--verose` flag and those 2 environment variables you gave makes the the log file grow huge quickly. So the full set of changes from my normal working setup and this test are:
- enable `--vebose` flag
- add `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` as env vars
- Enable log file

 I ran into an issue where trying to restart the Yoda service I created (after enabling the above changes) failed as Yoda didn't stop gracefully (also after this failed restart it seems the service I made to run initialize keeps running over and over as my fan speed keeps spiking to 100%, probably Windows trying to restart Yoda). I'm starting to suspect it's an issue with the log file, perhaps something to do with the way NSSM/services accesses files causes issues. Is there anything specific I should look for in the logs? I tried searching for errors but none appear, I'm guessing because those access errors are coming from another liquidctl instance (when I run `liquidctl status --legacy-690lc` in a terminal while Yoda is running).

xX-Kryptos-Xx, 2021-01-20:
Great piece of software. Thank you (and contributors) for all of your hard work on this. Proprietary software from EVGA was not cutting it. I just wanted to chime in on my setup. Took a bit of configuration but it worked out nicely.

I have an EVGA CLC 280 (Asetek 690LC non-legacy) and I use LQiNFO (yoda) to display pump/fan/coolant temp status in Rainmeter. Works out nicely as Rainmeter can pull registry values.

Here is a screen of my 8700K with stats.
[image]

Note: I have the same issue as ROODAY as far as accessing status using liquidctl when yoda is running. Get the same permission denied error. Stop yoda, all works well. No biggie as I set fan/pump speed on boot and then run yoda when I log into Windows. This could be some type of accessibility conflict to the USB device as both yoda and liquidctl are trying to access is for information at the same time. (just speculating here).

Another Note: WinUSB thinks my 690LC device is removable (using Zadig to replace driver). Not sure if anyone else has this issue. Couldn't find out much on the subject through the internet space.

I would be more than happy to help if a similar setup would like to be accomplished.

jonasmalacofilho, 2021-01-20:
It could be that yoda is running with higher privileges, and then lower privileged liquidctl instances cannot take ownership of the device _interface_ (even though yoda should be releasing _it_ to the OS between updates).

I'll try to reproduce these permission errors here...  _(and this is where I rant that it's really inconvenient that we can't get proper log output from these services)._

---

@ROODAY 

> I wonder, would it be possible to integrate something like this to display the current fan speed/temp/etc as system tray icons?

Sure.  Any chance you can prepare a pull request for that?

You could also/alternatively integrate [`extra/windows/LQiNFO.py`](https://github.com/liquidctl/liquidctl/blob/master/extra/windows/LQiNFO.py) and centralize all monitoring in HWiNFO.

Just please note that, whatever option you pick, it should not affect users running operating systems other than Windows.

@xX-Kryptos-Xx 

> WinUSB thinks my 690LC device is removable (using Zadig to replace driver)

I think that's to be expected.

ROODAY, 2021-01-20:
@jonasmalacofilho Sure I can look into making a PR! I just need some time as I'm waiting for my computer to reach my new address, going to be about another week and a half before I can start.

ROODAY, 2021-02-10:
@jonasmalacofilho Is there something special required to use the release function added in e227b47? I updated both my liquidctl release and the yoda script, and I still get the `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)` error when trying to run `liquidctl status --legacy-690lc` while yoda is running

jonasmalacofilho, 2021-02-10:
Well, I suppose 

> It could be that yoda is running with higher privileges, and then lower privileged liquidctl instances cannot take ownership of the device _interface_ (even though yoda should be releasing _it_ to the OS between updates).

could still apply.

That said, I think it's more likely that `yoda` is not using the same/fixed liquidctl version.  Note that it too uses the liquidctl APIs that are available on the system (and that this has no relation with the liquidctl source code that was shipped with the script).

Running both programs with `--debug` would help to check that, and to find possible solutions for other problems.  But I remember that you have some difficulty getting output from the NSSM service...

ROODAY, 2021-02-10:
Hmm it could be possible that system services are running at a higher privilege and that could just be a design feature of Windows. I think Yoda should be using the new version on my system as I made sure the NSSM service was pointing to that release (and the old release was deleted anyways). I'll look around see if there's another way to hook into a running service and see it's output, I'm sure that exists somehow

jonasmalacofilho, 2021-02-10:
> pointing to that release (and the old release was deleted anyways).

This might not be sufficient, you should check the Python environment (global or virtual, depending on top of what your service runs).

jonasmalacofilho, 2021-02-11:
> Hmm it could be possible that system services are running at a higher privilege and that could just be a design feature of Windows.

Actually, I remember testing that after that patch control could be shared between Administrative and regular prompts.  But maybe I made a mistake (during the tests or in later changes)...

ROODAY, 2021-02-11:
> > pointing to that release (and the old release was deleted anyways).
> 
> This might not be sufficient, you should check the Python environment (global or virtual, depending on top of what your service runs).

Ah, does the newer version of liquidctl require a newer version of python?

jonasmalacofilho, 2021-02-11:
No, so far we still support the same Python 3.6+ as always.  This will only change once 3.6 reaches EOL, which IIRC is later this year.

What I meant by the Python environment is that `yoda` will find the liquidctl APIs it will use in the Python environment it's running on, either the global or a virtual one.  So it's not sufficient to remove the old release from where you downloaded it, you also need to make sure that what's on the Python environment where `yoda` runs is what you expect.

_(Usually these enviroments would have been manipulated with some flavor of `pip install` or `python3 setup.py install`)._

ROODAY, 2021-02-16:
@jonasmalacofilho So I updated liquidctl via pip and voila, I am now able to run `liquidctl --legacy-690lc status` while Yoda is running. However now I have a new issue that's even worse, the fan keeps jumping to 100% every couple seconds and then goes back down. At first I thought it was Yoda, so I tried stopping it via the service manager, but the service manger failed to shut it down (not sure if that's related to me attempting to log it with `--debug` enabled). I eventually shut it down by querying for its PID and killing it, then I set the fan speed with `liquidctl --legacy-690lc set fan speed 50`.

However the problem persists, every couple minutes or so the fan jumps back up to max, and won't go back down until I either set the speed manually, or run the status command (no clue why running status makes it jump back down). I'm guessing that it's something with releasing the device explicitly, perhaps when it's released the hardware tries to take over and defaults to max speed? I think I'm able to get a few logs, but I'm not sure what to look for.

Also, as I was writing this, the issue got even worse (so like over the course of 5 minutes). Now it won't stay at the low fan speed for longer than a few seconds before ramping up again. I'm going to try to downgrade liquidctl as a short term fix, but any ideas what could be causing this?

jonasmalacofilho, 2021-02-16:
I think the issue has to do with the fact that, on these legacy 690LC devices, `liquidctl status --legacy-690lc` has to send cooling settings to be able to retrieve information from the cooler.

To support this quirk across multiple liquidctl instances and executions, the driver stores some runtime information in the filesystem; on Windows that will be somewhere within `%TEMP%`.  But since yoda and liquidctl are apparently running with different users, they could see different environments and `%TEMP%` values, which could explain the seemingly confused behavior you're observing (each would send different cooling settings to the device).

You could try checking the `--debug` logs for the location each instance is using.  And, to work around the problem, you could run both with a common `%TEMP%` value, or run your `liquidctl status --legacy=690lc` invocations with the same user (presumably Administrator).

---

Another possible cause is that one of your yoda or liquidctl incantations missed (or is missing) `--legacy-690lc`.  But this would normally cause the device to immediately lock up.

bandaid, 2021-04-10:
Okay, I just posted a request for support on this model. Just found this request. Thanks for looking into it.

jonasmalacofilho, 2021-04-27:
Back in February I left this open because it had something to do with #274.

The main issue here is the fact the driver instances are used from different users.  This causes each instance to see a different environment, and miss the internal files saved by the other instance, which will also be a concern when fixing #274.

But until we implement driver instance synchronization (to fix #274), it's not something we're going to change.<sup>1</sup>  Thus, I'm closing this issue.

_For now,_ simultaneous access to a device must be done with the same user and environment.<sup>2</sup>

_<sup>1</sup> Even though saving internal files per user is causing us some issues, it has also good reasons (e.g. secure but unprivileged access) to be, that will need to be accommodated in the new synchronization model._  
_<sup>2</sup> And on the specific case of the Corsair Hydro Platinum/Pro XT coolers, simultaneous access should be avoided while #274 isn't fixed._

commit e227b470 (2021-01-23): asetek: release the entire device on Windows

Allow the simultaneous use of more than one instance of liquidctl (or
other liquidctl-based programs) on Windows.

While on Linux (and possibly on Mac and the BSDs) we can get by simply
releasing the interface, Windows requires the entire device to be
released for other programs to be able to access it.

Closes: #246 (yoda + liquidctl status on Windows)
closed 2021-04-27 by jonasmalacofilho