liquidctl/liquidctl #248: Does not recognize devices

wyrmss, 2020-12-27:
Hello, I report my problem.
When installing liquidctl, it does not recognize the devices, I leave you captures so that you can see what it can be.
I do not know English so I will try to be as graphic as possible, I pass you captures.
[image]

As you can see the id do appear, but it does not detect them, not even with sudo.

With OpencorsairLink it works perfectly for me, capture below.
[image]

I don't know what could be happening.
I also tried the latest version 1.4.2 and the same, I left this one because I have it to install by the opensuse package manager and I don't have to create a virtual environment with pip and it works for all users.

A greeting.

MarshallAsch, 2020-12-27:
Have you tried running the command with sudo? That should not be causing the devices to not be recognized but its worth a shot.

wyrmss, 2020-12-27:
Hello

`As you can see the id shows up, but it doesn't detect them, not even with sudo.`

If as I said before, I do not work as sudo or as root

A greeting.

jonasmalacofilho, 2020-12-27:
The issue is likely to be that the `usbhid` driver has been unbound, possibly by OpenCorsairLink (or other tools using libusb).

Try restarting or following [this troubleshooting section](https://github.com/liquidctl/liquidctl#device-not-listed-linux).

jonasmalacofilho, 2020-12-27:
Also note that we don't yet support Corsair Pro coolers on the current stable release or even the main development branch.

But it's coming...

jonasmalacofilho, 2020-12-27:
Sorry, one more comment: you're somehow using a very old version of liquidctl, please try to update to 1.4.2.

wyrmss, 2020-12-27:
> Es probable que el problema sea que el `usbhid`controlador se ha desatado, posiblemente por OpenCorsairLink (u otras herramientas que usan libusb).
> 
> Intente reiniciar o seguir [esta sección de solución de problemas](https://github.com/liquidctl/liquidctl#device-not-listed-linux) .

Uninstall OpencorsairLink and remove all your files and folders that were left after the uninstallation, restart the pc and the same, it does not detect them.

A greeting.

wyrmss, 2020-12-27:
> Lo siento, un comentario más: de alguna manera está usando una versión muy antigua de liquidctl, intente actualizar a 1.4.2.

Also use the new 1.4.2 with the same result, using this is because it is in the community opensuse repositories and not having to use a virtual environment and so all users can use it.

If it is an impetuous need for the latest version, desistlo this I have installed again the 1.4.2.

A greeting.

jonasmalacofilho, 2020-12-27:
Right, I just noticed that the PID for what you're calling a RM1000 doesn't match what we expect at all.  In fact, it seems to use the CorsairLink dongle, which RMi PSUs usually don't.

Then again, you said RM1000, not RM1000i.... if that's indeed correct (no "i" on your model), then we unfortunately don't support it.

---

The Corsair H115i Pro will ultimately need to wait for 1.5.0, but in the meantime you can give the code in #234 a try.

jonasmalacofilho, 2020-12-27:
Leaving this here in case it's a really a RM1000 (no "i")...

Can you [capture some USB traffic](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md) between the RM1000 and iCue/CorsairLink?  If they are similar to what we've seen for the RMi series, I could try enabling support for it in our driver.

---

P.S. Please join our [developer's Discord server](https://discord.gg/eJ5MswXSf9).  If it helps I can write in Portuguese and I can (mostly) understand Spanish.

wyrmss, 2020-12-27:
Hello

So I must deduce that there is no error, only that my hardware is not supported :(.

My Corsair **RM1000 1000W 80 Plus Gold Modular** psu is not supported?

I understand that the H115i Pro will have it in the next or next updates :).
If so, as I read in the thread that you reported to me, we will wait for the next version to go from OpencorsairLink to liquidctl.

Regards and thanks for everything.

wyrmss, 2020-12-27:
> Dejando esto aquí en caso de que sea realmente un RM1000 (sin "i") ...
> 
> ¿Puede [capturar algo de tráfico USB](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md) entre el RM1000 e iCue / CorsairLink? Si son similares a lo que hemos visto para la serie RMi, podría intentar habilitar el soporte en nuestro controlador.
> 
> PD: únete [al servidor Discord de](https://discord.gg/eJ5MswXSf9) nuestro [desarrollador](https://discord.gg/eJ5MswXSf9) . Si ayuda, puedo escribir en portugués y puedo (principalmente) entender el español.

I don't use windows, I'm really sorry, I can't use corsairlink or icue

jonasmalacofilho, 2020-12-27:
> I don't use windows, I'm really sorry, I can't use corsairlink or icue

That's too bad : \

closed 2020-12-27 by jonasmalacofilho

wyrmss, 2020-12-27:
> > No uso windows, lo siento mucho, no puedo usar corsairlink o icue
> 
> Eso es muy malo : \

So sorry.

My psu https://www.corsair.com/es/es/Categories%C3%ADas/Products/Units-of-source-of-alimentaci%C3%B3n/rm-series-config/p/CP-9020062- NA