liquidctl/liquidctl #249: Support for AURA LED Controller?

aluveitie, 2020-12-29:
Any plans to support Asus AURA LED Controller which is connected through USB?

jonasmalacofilho, 2020-12-29:
As long as someone with one of these controllers can work on the feature, merging/accepting it would be a no-brainer.

aluveitie, 2020-12-29:
OpenRGB would have a working implementation 
https://gitlab.com/CalcProgrammer1/OpenRGB/-/tree/master/Controllers/AsusAuraUSBController

jonasmalacofilho, 2021-01-13:
That's a good starting point.  Although, to be clear, I don't plan on working on this feature myself, at least in the short term.

jonasmalacofilho, 2021-02-27:
This is good first issue.  Anyone interested in working on this can join our [developer's Discord server](https://discord.gg/f5HKCAadsr) for help.

aluveitie, 2021-03-03:
There is also a working implementation for MacOS: https://github.com/notjosh/Aureal

jonasmalacofilho, 2021-08-06:
**Triage: close without fixing**

At this time, the maintainers don't have the hardware and specifications to work on this, and no one has demonstrated interest in collaborating to get this implemented.

closed 2021-08-06 by jonasmalacofilho