liquidctl/liquidctl #252: FreeBSD 13-CURRENT: ImportError: /usr/local/lib/python3.7/site-packages/hid.so: Undefined symbol "hid_get_input_report"

historical-theology, 2020-12-30:
I have a new installation of the latest FreeBSD 13-CURRENT (Dec. 24 snapshot) on an X370 motherboard. I have used variously plain (glorious) liquidctl and also the Gkraken frontend in Manjaro GNU/Linux without problem over the past few years on this same system to control my Kraken X62.

I installed 'py37-liquidctl' from the official FreeBSD repository ('pkg install py37-liquidctl'). When trying to run it, whether as root or not, I receive the following error:

```
root@coreybsd:~ # liquidctl list
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 63, in <module>
    import hidraw as hid
ModuleNotFoundError: No module named 'hidraw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.4.2', 'console_scripts', 'liquidctl')()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.7/site-packages/liquidctl/cli.py", line 75, in <module>
    from liquidctl.driver import *
  File "/usr/local/lib/python3.7/site-packages/liquidctl/driver/__init__.py", line 25, in <module>
    from liquidctl.driver import asetek
  File "/usr/local/lib/python3.7/site-packages/liquidctl/driver/asetek.py", line 23, in <module>
    from liquidctl.driver.usb import UsbDriver
  File "/usr/local/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 65, in <module>
    import hid
ImportError: /usr/local/lib/python3.7/site-packages/hid.so: Undefined symbol "hid_get_input_report"
```

I have read the following related report (Fedora GNU/Linux 32): https://github.com/liquidctl/liquidctl/issues/145

However, I have a fresh installation of all of the related components, including:

py37-liquidctl = 1.4.2
hidapi = 0.8.0.r1_3
py37-hidapi = 0.10.1
libudev-devd = 0.4.2_1

Thus, I suspect that this is a FreeBSD-specific issue.

I also receive the following error message:

```
corey@coreybsd:~ $ python3.7 -c "import usb; import hid; import hidraw"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/local/lib/python3.7/site-packages/hid.so: Undefined symbol "hid_get_input_report"
```

Thoughts, anyone?

Thanks. I love liquidctl; I simply want to get it running in FreeBSD :smile:

jonasmalacofilho, 2020-12-30:
I think this is caused by a mismatch between py37-hidapi and hidapi.  In particular, py37-hidapi expects hidapi to export that symbol, but it wont until a later version.  Take a look at trezor/cython-hidapi#90.

historical-theology, 2020-12-31:
@jonasmalacofilho Well, that is a bummer! I was not able to get `hidapi` to build from source (and I do not recommend bothering, since it required me to install a bunch of otherwise useless Linux compatibility cruft), so it would seem that I will be stuck until the package is updated to 0.9.X in the official FreeBSD repository. 

The good news is that should not take long. I will report back here when that time comes. Thank you for the swift and helpful reply.

jonasmalacofilho, 2021-01-13:
> The good news is that should not take long. I will report back here when that time comes. Thank you for the swift and helpful reply.

I'm going to close this issue, but please let me know when the situation is fixed in FreeBSD 13.

P.S. By the way, the problem is almost certainly affecting other programs that use cython-hidapi.

closed 2021-01-13 by jonasmalacofilho

historical-theology, 2021-03-20:
@jonasmalacofilho, I reported the fact that comms/hidapi is out of date to the port maintainer, who immediately updated it. Until the next quarterly update (April), comms/hidapi must be built from the latest ports tree (v. 0.10.1) before installing liquidctl, but then `liquidctl` and all other required packages can be installed from the most recent binary builds with `sudo pkg install liquidctl`.

RESOLVED :smile:

jonasmalacofilho, 2021-03-20:
Awesome, and thanks for letting me know!