liquidctl/liquidctl #259: [BUG] Linux with hidapi-0.10.1 is showing RGB Fusion controller twice

aravance, 2021-01-03:
## Description of Bug
#### System details
* Distro: Arch Linux
* `uname -r`
> 5.10.3-zen1-1-zen
* ``pacman -Q liquidctl hidapi `pacman -Qi liquidctl | awk -F'[:<=>]' '/^Depends/ {print $2}'` ``
> liquidctl 1.4.2-4
> hidapi 0.10.1-1
> python 3.9.1-1
> python-docopt 0.6.2-9
> python-pyusb 1.1.0-3
> python-hidapi 0.10.1-1

### Steps to reproduce
`liquidctl list`

### Expected
> Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
> Device ID 1: Corsair H115i Platinum (experimental)

### Actual
> Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
> Device ID 1: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
> Device ID 2: Corsair H115i Platinum (experimental)

### Workaround
I downgraded to hidapi-0.9.0-2 from package cache, and the devices now show up only once.

## Hardware Configuration
* x570 Aorus Master rev1.1
* Corsair h115i RGB Platinum

jonasmalacofilho, 2021-01-03:
Can you please run

```
$ python -c "import hid; print('\n'.join(map(repr, hid.enumerate())))"
```

and post the output?

aravance, 2021-01-03:
Worth note, hidapi 0.10.1 introduced a change to handling of usages and pages: https://github.com/libusb/hidapi/releases

I reported a similar issue with OpenRGB here: https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/924

### 0.9.0-2

> {'path': b'/dev/hidraw0', 'vendor_id': 1165, 'product_id': 33431, 'serial_number': '', 'release_number': 3, 'manufacturer_string': 'ITE Tech. Inc.', 'product_string': 'ITE Device(8595)', 'usage_page': 51456, 'usage': 13354, 'interface_number': 0}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 12080, 'usage': 12336, 'interface_number': 5}
{'path': b'/dev/hidraw2', 'vendor_id': 1133, 'product_id': 50501, 'serial_number': '', 'release_number': 512, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
{'path': b'/dev/hidraw3', 'vendor_id': 1133, 'product_id': 50501, 'serial_number': '', 'release_number': 512, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 0, 'usage': 0, 'interface_number': 1}
{'path': b'/dev/hidraw4', 'vendor_id': 1133, 'product_id': 50501, 'serial_number': '', 'release_number': 512, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 12080, 'usage': 12336, 'interface_number': 2}
{'path': b'/dev/hidraw5', 'vendor_id': 6940, 'product_id': 2610, 'serial_number': 'v0.6', 'release_number': 256, 'manufacturer_string': 'Corsa\x1cir Components Inc.', 'product_string': 'Corsair ST100 Headset Outpu\x1ct', 'usage_page': 5584, 'usage': 13350, 'interface_number': 3}
{'path': b'/dev/hidraw6', 'vendor_id': 6940, 'product_id': 3095, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'Corsair Inc.', 'product_string': 'H115i Platinum', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
{'path': b'/dev/hidraw7', 'vendor_id': 1133, 'product_id': 50490, 'serial_number': '', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
{'path': b'/dev/hidraw8', 'vendor_id': 1133, 'product_id': 50490, 'serial_number': '', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 12080, 'usage': 12336, 'interface_number': 1}
{'path': b'/dev/hidraw9', 'vendor_id': 1133, 'product_id': 50490, 'serial_number': '', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 0, 'usage': 0, 'interface_number': 2}
{'path': b'/dev/hidraw10', 'vendor_id': 6940, 'product_id': 2612, 'serial_number': '22900316E2204CAFFDF1CB5C001C00F5', 'release_number': 4, 'manufacturer_string': 'Corsair', 'product_string': 'Corsair ST100 LED Driver', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
{'path': b'/dev/hidraw11', 'vendor_id': 6940, 'product_id': 3098, 'serial_number': '45B0010BE00445AF6555055CC71B00F5', 'release_number': 9, 'manufacturer_string': 'Corsair', 'product_string': 'CORSAIR Lighting Node CORE', 'usage_page': 12080, 'usage': 12336, 'interface_number': 0}
{'path': b'/dev/hidraw13', 'vendor_id': 1133, 'product_id': 16479, 'serial_number': '405f-d7-86-94-8c', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 12080, 'usage': 12336, 'interface_number': 2}
{'path': b'/dev/hidraw12', 'vendor_id': 1133, 'product_id': 16505, 'serial_number': '4079-ea-c0-67-c6', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 0, 'usage': 0, 'interface_number': 2}

### 0.10.1-1

> {'path': b'/dev/hidraw0', 'vendor_id': 1165, 'product_id': 33431, 'serial_number': '', 'release_number': 3, 'manufacturer_string': 'ITE Tech. Inc.', 'product_string': 'ITE Device(8595)', 'usage_page': 65417, 'usage': 16, 'interface_number': 0}
{'path': b'/dev/hidraw0', 'vendor_id': 1165, 'product_id': 33431, 'serial_number': '', 'release_number': 3, 'manufacturer_string': 'ITE Tech. Inc.', 'product_string': 'ITE Device(8595)', 'usage_page': 65417, 'usage': 204, 'interface_number': 0}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 12, 'usage': 1, 'interface_number': 5}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 12, 'usage': 1, 'interface_number': 5}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 65347, 'usage': 514, 'interface_number': 5}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 65280, 'usage': 1, 'interface_number': 5}
{'path': b'/dev/hidraw1', 'vendor_id': 4152, 'product_id': 4781, 'serial_number': '', 'release_number': 281, 'manufacturer_string': 'SteelSeries ', 'product_string': 'SteelSeries Arctis 7', 'usage_page': 65280, 'usage': 10, 'interface_number': 5}
[... cut ...]
{'path': b'/dev/hidraw13', 'vendor_id': 1133, 'product_id': 16479, 'serial_number': '405f-d7-86-94-8c', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 65280, 'usage': 2, 'interface_number': 2}
{'path': b'/dev/hidraw13', 'vendor_id': 1133, 'product_id': 16479, 'serial_number': '405f-d7-86-94-8c', 'release_number': 16386, 'manufacturer_string': 'Logitech', 'product_string': 'USB Receiver', 'usage_page': 65280, 'usage': 4, 'interface_number': 2}

jonasmalacofilho, 2021-01-04:
> Worth note, hidapi 0.10.1 introduced a change to handling of usages and pages: https://github.com/libusb/hidapi/releases

That's what I figured, as these controllers expose two usages.

But on 0.9.2, hidapi-hidraw returns `usage: 0x342a`, instead of `0` (as documented at the time) or one of the actual usages (`0x10` or `0xcc`).

This complicates things, since I have to match the usage on 0.10.0, but can't possibly do so on 0.9.2...

Additionally, I also need to somehow check whether hidraw or libusb is in use, because (from the release notes and the current docs) only the former supports multiple usages.

jonasmalacofilho, 2021-01-04:
Oh, `usage: 0x342a` apparently came from uninitialized memory, since `usage` and `usage_page` were simply [never assigned to](https://github.com/libusb/hidapi/blob/hidapi-0.9.0/linux/hid.c).  : ( 

---

We need [cython-hidapi](https://github.com/trezor/cython-hidapi/) to export `hid_version_str` or `hid_api_version` (both new in 0.10.0, fixed in 0.10.1).

Additionally, we may also need some cython-hidapi method to tell us the backend in use, since that can vary at build time and (from what I can see) cannot be inferred later from the built modules.

jonasmalacofilho, 2021-01-04:
I added a Arch-specific quick fix patch to [liquidctl-git<sup>AUR</sup>](https://aur.archlinux.org/packages/liquidctl-git/).

Hopefully we can have a more portable solution by the time liquidctl 1.5 is released.

jonasmalacofilho, 2021-01-04:
This should now be permanently fixed in the master branch.  Can you give it a try and let me know?

(liquidctl-git<sup>AUR</sup> has been updated accordingly)

---

Filipe, I'm not going to ping you here, but if you end up reading this:

Since this issue is specific to this driver and to distros already running hidapi 0.10, and since we also already have a release scheduled for the end of the month, I don't think a 1.4.3 patch release is necessary.  But I can (easily) be convinced otherwise.  Also, if necessary, the 4286cc477ddf patch can also be (cleanly) applied on top of 1.4.2.

aravance, 2021-01-05:
I pulled the latest AUR without the patch, and it's not working as expected.

jonasmalacofilho, 2021-01-05:
Can you post the `--debug` output?

aravance, 2021-01-05:
Sorry, typo. Everything is confirmed working.

closed 2021-01-05 by aravance