liquidctl/liquidctl #260: Feature: Support fan curves for Smart Device V2 (and other devices)

DrPeroxide, 2021-01-06:
I noticed in the docs that the app currently only supports fixed duty values for the Smart Device. A lot of users like myself rely on Fan curves that can respond to changes to CPU or GPU temp to optimise cooling and noise levels.

I'm very invested in getting this feature added, so would be happy to work on it myself. Has anyone closer to the project had any thoughts as to how this should be designed/implemented?

jonasmalacofilho, 2021-01-07:
I'm glad to hear that you're interested in working on this, and it's obviously an useful feature.  But there are some complications...

First, these devices don't themselves support profiles, so they would need to run "in software", that is, on the main system's CPU.  This deviates from how liquidctl normally works, and requires a lot of care regarding error conditions and recovery.

Integrating the feature with liquidctl also poses some challenges in how it would be implemented on the CLI.  Perhaps #195 may be required for a reasonable user experience.

There are also issues regarding access to the sensors.  On Linux that's fairly easy to solve, but on Mac OS things get a bit harder, and on Windows... well, there's simply no standard way of getting sensor data (and very few non-proprietary tools we can integrate with).

---

For starts, take a look at [`extra/yoda`](https://github.com/liquidctl/liquidctl/blob/master/extra/yoda).  It's a proof-of-concept of a loop-based implementation of profiles for devices that don't support them (or using sensors not directly available to the devices).  It has some significant limitations though, like not being able to read sensors for other liquidctl devices, and not yet having a super robust error handling.

Then, take a look at the _very old_ [`integrate-yoda`](https://github.com/liquidctl/liquidctl/tree/integrate-yoda) branch.  It's not functional, but shows what I was starting to play with in terms of integrating yoda into the main liquidctl CLI, and which commands and options I was planning to use (but that's by no means a solved problem).

Finally, take a look at the comments on #72 for additional information related to this feature, like some rough/early ideas on how to get sensor data on Windows.

jonasmalacofilho, 2021-01-24:
@DrPeroxide

I'm going to close this for now since I haven't heard back from you.

But if you still intend to work on this, please let me know.  Also, join our developers' Discord server: <https://discord.gg/PQAbEhAvV9>.

closed 2021-01-24 by jonasmalacofilho