liquidctl/liquidctl #262: Create Completions for Fish Shell

hmajid2301, 2021-01-11:
Would it be possible to add a completions file for fish shell alongside bash ?

Some Examples: https://github.com/fish-shell/fish-shell/tree/master/share/completions

jonasmalacofilho, 2021-01-13:
I don't personally know enough about the fish shell to work on this, but it would be nice to have.

That said, it seems that the completion file would be distributed with the fish shell... so any work should probably be submitted to that repository.

MarshallAsch, 2021-01-25:
@hmajid2301 
I was looking this up to see how similar it is to bash completions and it seems that it can be included here and installed the same way that the other completion is installed. 

https://fishshell.com/docs/current/#tab-completion

These seem like some good starting points:
https://medium.com/@fabioantunes/a-guide-for-fish-shell-completions-485ac04ac63c
https://fishshell.com/docs/current/cmds/complete.html

If you want to work on this I'd happily go over how I generated the completion script for bash if that helps.

closed 2021-02-27 by jonasmalacofilho