liquidctl/liquidctl #265: Don't forget to initialize when you upgrade from Catalina to Big Sur

zk2004mb, 2021-01-20:
~~Hi!~~
~~I just upgrade from 10.15.7 to 11.1 this afternoon. However, the liqudctl seems stop working.~~
~~I can still see my device:~~

~~(base) user ~ % liquidctl list~~
~~Device ID 0: NZXT RGB & Fan Controller (experimental)~~

~~But when I type  liquidctl status~~
~~It will just be stuck there. Also when I try to change the fan speed, the code will be executed but the actual speed won't change.~~

Problem solved. Just re-initialize the device and it's good to go.
I keep the message here in case someone might in the same situation as me.

jonasmalacofilho, 2021-01-21:
Updating to Big Sur didn't require a reboot?

In general you should always `liquidctl initialize` your devices after booting up the system.

closed 2021-01-22 by jonasmalacofilho

zk2004mb, 2021-01-22:
Actually I never had any problem after a reboot without initialize. That's why I didn't figure that out in the first place.

jonasmalacofilho, 2021-01-22:
That's because that device can draw power from the 5V Stand-by rail and, generally, will keep itself initialized even after the host has been shutdown (as long as the PSU is still on).  Still, generally a good idea to always initialize after a new boot.

Anyway, thanks for the info!