liquidctl/liquidctl #267: Setting LED colour on Corsair H100i Platinum SE only sets colour for pump and one fan.

Priva28, 2021-01-22:
Pretty much the title. Fan/pump control works great as expected but setting RGB colour seems to only be half working. 

Running `sudo liquidctl initialize all` shows this:
```
Corsair H100i Platinum SE (experimental)
└── Firmware version    1.1.15
```

Running `sudo liquidctl set led color off` works and turns all of the LEDs off for the pump, fan 1 and fan 2.

But running `sudo liquidctl set led color fixed ffffff` only sets the colour for the pump and fan 1 LED while fan 2 LED remains off.

I'm running a plain Debian 10 Buster.
[image]

jonasmalacofilho, 2021-01-22:
Can you provide the output of `liquidctl list --debug`?

Additionally, do you know how many LEDs are in the pump and in each fan?

Priva28, 2021-01-22:
Sure:
```
[DEBUG] liquidctl.cli: running liquidctl v1.4.2
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Platinum SE (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1462:7c75
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25e3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bc2:231a
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a86:7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 1462:7c75
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:10a2
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: Corsair H100i Platinum SE (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c19
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw6
└── Driver: HydroPlatinum using module hidraw
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

MarshallAsch, 2021-01-22:
This should be an easy fix that Im working on. The H100i Platinum SE uses the LL fans that have 16 leds on each of them.

MarshallAsch, 2021-01-22:
This is a duplicate of #242. 

closing.

PR #269 (2021-01-22): Corsair H100i Platinum SE Number of LEDs fix
Closes #267 and #242.

Changed support to work with a different number of LED's that come with the different stock Hydro Platinum AIO coolers.

This changes one of the options that is passed to the constructor for the `HydroPlatinum` driver. Changed `rgb_fans`  boolean option in the HydroPlatinum driver to a `fan_leds` integer option that is set to the number of LEDs that are on each fan that comes with the cooler.

For the SE cooler that comes with LL fans (16 leds each) this bumps the total number of LEDs to 48, which is too many for to fit in the [... cut]
closed 2021-01-22 by MarshallAsch