liquidctl/liquidctl #270: Is there a way to speed up the status query?

jabra98, 2021-01-23:
I'm trying to integrate it into another program, but an invocation through a syscall takes like 700ms . Is there a way to speed things up?

jonasmalacofilho, 2021-01-23:
What device and OS are we talking about, exactly?

Also, are you using our Python APIs, or executing the `liquidctl` CLI?

jabra98, 2021-01-24:
I'm using the Kraken x62 on Arch Linux. Yeah, I use the liquidctl CLI. Is there an C/C++ API aswell?

jonasmalacofilho, 2021-01-24:
No, but in theory you could integrate a Python interpreter and use the liquidctl APIs from there.

---

That said, a better alternative might be for you to use the `krx62` kernel driver from [liquidtux](https://github.com/liquidctl/liquidtux) (packaged for Arch in [liquidtux-dkms-git<sup>AUR</sup>](https://aur.archlinux.org/packages/liquidtux-dkms-git/)), and poll the hwmon data from sysfs.

*Just a quick disclaimer: that driver has a disgusting data race, but since it still works (I personally use it) on X86-64 I haven't been bothered to fix it yet.*

While using the liquidctl APIs would already speed things up a bit (no need to load Python, probe for devices or connect to the X62 every time), they still need to assume worst-case conditions: in practice, the liquidctl APIs never use an already enqueued HID report, and always wait for a *new* one to arrive.  The kernel driver, on the other hand, can do things a bit more efficiently.

jabra98, 2021-01-24:
Alright, thanks for clarifying.

closed 2021-01-24 by jonasmalacofilho