liquidctl/liquidctl #274: Corsair H100i PRO XT stops responding to read requests.

planeturban, 2021-01-26:
I've noticed this twice now; after a while (don't think it's X time but rather X read requests) my H100i PRO XT stops responding. The reason for me thinking it's X reads is that both times this has happend I've had a `while 1; do sleep; liquidctl status; done` loop running. 

I can write, but the process hangs after this:

```
root@pve:~# liquidctl initialize -g                                                                                                                                                                                                                                                                                 
[DEBUG] liquidctl.cli: running liquidctl v1.4.2                                                                                                                                                                                                                                                                     
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])                                                                                                      
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i PRO XT (experimental)                                                                                                                                                                                                                              
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])                                                                                                                                                                                              
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0024                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0024                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.cli: device: Corsair H100i PRO XT (experimental)                                                                                                                                                                                                                                                  
[DEBUG] liquidctl.keyval: data in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0                                                                                                                                                                                                                                        
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/leds_enabled)                                                                                                                                                                                                        
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/pump_mode)                                                                                                                                                                                                              
[DEBUG] liquidctl.keyval: no data (file) found for fan1_mode                                                                                                                                                                                                                                                        
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)                                                                                                                                                                                                                              
[DEBUG] liquidctl.keyval: no data (file) found for fan1_duty                                                                                                                                                                                                                                                        
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 100% duty cycle                                                                                                                                                                                                                                             
[DEBUG] liquidctl.keyval: no data (file) found for fan2_mode                                                                                                                                                                                                                                                        
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)                                                                                                                                                                                                                              
[DEBUG] liquidctl.keyval: no data (file) found for fan2_duty                                                                                                                                                                                                                                                        
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 100% duty cycle                                                                                                                                                                                                                                             
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced                                                                                                                                                                                                                                               
[DEBUG] liquidctl.keyval: loaded sequence=25 (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/sequence)                                                                                                                                                                                                             
[DEBUG] liquidctl.keyval: stored sequence=26 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/sequence)                                                                                                                                                                                                               
[... cut ...]               
    _device_set_speed(dev, args, **opts)                                                                                                                                                                                                                                                                            
KeyboardInterrupt                                                                                                                                                           
```

jonasmalacofilho, 2021-01-26:
The use of two simultaneous instances of liquidctl could explain this, as the sequence numbers may be received by the device out of order, and we already have previous reasons to think it can sometimes lock up when that happens.

Can you confirm/reproduce this two instances scenario (a loop + another instance), and provide the `--debug` output of both at the time when the device freezes?

Assuming that's it, we may need to implement some a two-phase commit (or some other stronger synchronization method) for the sequence number.

jonasmalacofilho, 2021-01-26:
Oh, can you switch to the current code in the master branch before trying to reproduce the problem?

planeturban, 2021-01-26:
Sure:

```
root@pve:~/liquidctl# ps -ef | grep liq                                                                                                                                                                                                                                                                             
root      5811 25202  0 12:57 pts/1    00:00:00 grep liq                                                                                                                                                                                                                                                            
root@pve:~/liquidctl# python3 -m liquidctl.cli initialize -g                                                                                                                                                                                                                                                        
[DEBUG] __main__: running liquidctl v1.4.2 (6dd4f3de3a18)                                                                                                                                                                                                                                                           
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)                                                                                                                                                                                                           
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0                                                                                                                                                                                                                                                                 
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)                                                                                                              
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c20                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Pro XT (experimental)                                                                                                                                                                                                                              
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)                                                                                                                                                                                                        
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c20                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002                                                                                                                                                                                                                                                            
[DEBUG] __main__: device: Corsair H100i Pro XT (experimental)                                                                                                                                                                                                                                                       
[DEBUG] liquidctl.keyval: data in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0                                                                                                                                                                                                                                        
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/leds_enabled)                                                                                                                                 [... cut ...]
root@pve:~/liquidctl# git status                                                                                                                                                                                                                                                                                    
On branch master                                                                                                                                                                                                                                                                                                    
Your branch is up to date with 'origin/master'.                             

```

jonasmalacofilho, 2021-01-26:
I actually wanted to compare the various `loaded sequence` and `stored sequence` lines.  But that's ok for now.

---

I reviewed the code and we definitively need to implement a `RuntimeStorage.load_add` method that atomically adds to the current value, returning the previous value, and rewrite `hydro_platinum._sequence` in terms of it.

In the meantime, the only workaround is to avoid running multiple simultaneous instances of liquidctl.

planeturban, 2021-01-26:
So, there's a lock up in the hid device itself? Since I don't have any other instances runnig now?

Here are the full outputs:

```
root@pve:~/liquidctl# python3 -m liquidctl.cli initialize -g                                                                                                                                                                                                                                                        
[DEBUG] __main__: running liquidctl v1.4.2 (6dd4f3de3a18)                                                                                                                                                                                                                                                           
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)                                                                                                                                                                                                           
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0                                                                                                                                                                                                                                                                 
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)                                                                                                              
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c20                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Pro XT (experimental)                                                                                                                                                                                                                              
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)                                                                                                                                                                                                        
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c20                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024                                                                                                                                                                                                                                                            
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002                                                                                                                                                                                                                                                            
[DEBUG] __main__: device: Corsair H100i Pro XT (experimental)                                                                                                                                                                                                                                                       
[DEBUG] liquidctl.keyval: data in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0                                                                                                                                                                                                                                        
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/leds_enabled)                                                                                                                                                                                                        
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/pump_mode)                                                                                                                                                                                                              
[... cut ...]                                                              
    _device_set_speed(dev, args, **opts)                                                                                                                                                                                                                                                                            
KeyboardInterrupt                                                                                                            
```

jonasmalacofilho, 2021-01-26:
Once the device locks up, it will probably stay that way until powered down (which may require turning off the PSU for a couple of minutes).  But if the device is brought back to a working state, it should not lock up again as long as you only run a single liquidctl instance at a time.

MarshallAsch, 2021-01-26:
There is also a `RESET` command that can be sent to the device if you want to write yourself some code to send the message that has resolved the same problem on my H100i Platinum. (not caused by liquidctl). https://github.com/liquidctl/collected-device-data/blob/master/Corsair%20H100i%20RGB%20Platinum/04%20-%20capture%20-%20marshallasch/protocol.md#special-restart-command

planeturban, 2021-01-27:
Actually, I might just do that. Get som new skills. :)

Thanks.

jonasmalacofilho, 2021-02-15:
@planeturban 

Can you test the code in PR #285 (branch [`274-keyval-file-locking`](https://github.com/liquidctl/liquidctl/tree/274-keyval-file-locking)) and see if it fixes the issues you were having with the Pro XT coolers and more than one liquidctl instance running?

Just make sure that all instances of liquidctl (and of the liquidctl APIs) that are running simultaneously have been updated.

planeturban, 2021-02-15:
Ran the following test for 5(!) hours (forgot to exit :D). 

`watch -n .3 "clear; python3 -m liquidctl.cli status"` and `watch -n .1 "python3 -m liquidctl.cli set fan2 speed  20 0 35 20 40 50 49 20 50 95"` in two different sessions..

Without any issues. Seems ok.

jonasmalacofilho, 2021-02-16:
Thank you, that's good to hear! : )

_(I just hope these devices are able to deal with multi-packet commands from different instances getting intertwined.  Otherwise our atomic `_sequence` wont be enough and we'll need to lock the entire device._

_This potential issue shouldn't be a big deal with the H100i Pro XT, but could be a more significant problem on models with more fans and/or LEDs, like the H100i Platinum SE or H150i Pro XT)._

Anyway, I'll give our current fix a bit longer to mature before releasing it in 1.5.1.  In the meantime, please try to stress test the fix as much as possible, and let me know if you notice any problems.

planeturban, 2021-02-17:
Just got back in the office some 30 minutes ago. Had it running all day and night. After 8 hours is locked up with the following error

```
ERROR: Unexpected OS error with Corsair Hydro H100i Pro XT (experimental): OSError: Could not write to device                
```

Could be my machine. I'll do some digging with hid modules..

planeturban, 2021-02-17:
Tried reloading the hid module with no luck. However, I'd say this case with excessive reads and writes to the controller is not a real world scenario. So I'll suggest we'll leave it as is.

Attaching debug logs. 

```
root@pve:~/liquidctl# python3 -m liquidctl.cli -g initialize all
[DEBUG] __main__: running liquidctl v1.5.0 (6dd4f3de3a18)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found IÂ²C bus i2c-0
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c20
[DEBUG] liquidctl.driver.hydro_platinum: instanced driver for Corsair Hydro H100i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 8087:0024
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] __main__: device: Corsair Hydro H100i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/leds_enabled)
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/pump_mode)
[DEBUG] liquidctl.keyval: ['/run/user/0/liquidctl/vid1b1c_pid0c20/loc0', '/var/run/liquidctl/vid1b1c_pid0c20/loc0']
[DEBUG] liquidctl.keyval: fan1_mode
[DEBUG] liquidctl.keyval: loaded fan1_mode=0 (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/fan1_mode)
[DEBUG] liquidctl.keyval: ['/run/user/0/liquidctl/vid1b1c_pid0c20/loc0', '/var/run/liquidctl/vid1b1c_pid0c20/loc0']
[DEBUG] liquidctl.keyval: fan1_profile
[DEBUG] liquidctl.keyval: loaded fan1_profile=[(20, 10), (30, 20), (40, 30), (45, 80), (50, 95)] (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/fan1_profile)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to follow profile [(20, 10), (30, 20), (40, 30), (45, 80), (50, 95), (60, 100), (60, 100)]
[DEBUG] liquidctl.keyval: ['/run/user/0/liquidctl/vid1b1c_pid0c20/loc0', '/var/run/liquidctl/vid1b1c_pid0c20/loc0']
[DEBUG] liquidctl.keyval: fan2_mode
[DEBUG] liquidctl.keyval: loaded fan2_mode=0 (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/fan2_mode)
[DEBUG] liquidctl.keyval: ['/run/user/0/liquidctl/vid1b1c_pid0c20/loc0', '/var/run/liquidctl/vid1b1c_pid0c20/loc0']
[DEBUG] liquidctl.keyval: fan2_profile
[DEBUG] liquidctl.keyval: loaded fan2_profile=[(20, 0), (35, 20), (40, 50), (49, 20), (50, 95)] (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/fan2_profile)
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to follow profile [(20, 0), (35, 20), (40, 50), (49, 50), (50, 95), (60, 100), (60, 100)]
[DEBUG] liquidctl.keyval: ['/run/user/0/liquidctl/vid1b1c_pid0c20/loc0', '/var/run/liquidctl/vid1b1c_pid0c20/loc0']
[DEBUG] liquidctl.keyval: pump_mode
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=13 (from /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/sequence)
[DEBUG] liquidctl.keyval: replaced with sequence=14 (stored in /run/user/0/liquidctl/vid1b1c_pid0c20/loc0/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:70:14:00:ff:05:ff:ff:ff:ff:ff:00:00:00:00:00:00:00:00:00:00:00:00:01:00:00:00:00:00:07:14:1a:1e:33:28:4c:2d:cc:32:f2:3c:ff:3c:ff:14:00:23:33:28:80:31:80:32:f2:3c:ff:3c:ff:00:00:00:00:00:2c
[INFO] __main__: Could not write to device
Traceback (most recent call last):
  File "/root/liquidctl/liquidctl/cli.py", line 321, in main
    _print_dev_status(dev, dev.initialize(**opts))
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 211, in initialize
    res = self._send_set_cooling()
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 419, in _send_set_cooling
    return self._send_command(_FEATURE_COOLING, _CMD_SET_COOLING, data=data)
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 376, in _send_command
    self.device.write(buf)
  File "/root/liquidctl/liquidctl/driver/usb.py", line 435, in write
    raise OSError('Could not write to device')
OSError: Could not write to device
[ERROR] __main__: Unexpected OS error with Corsair Hydro H100i Pro XT (experimental): OSError: Could not write to device
```

jonasmalacofilho, 2021-02-17:
Can you still read from the device?  And can you change the code in `usb.py:435` to print the value of `res`?

jonasmalacofilho, 2021-02-17:
Can you also confirm the commit that matches the code that you're running?

`[DEBUG] __main__: running liquidctl v1.5.0 (6dd4f3de3a18)` is showing stale metadata.

jonasmalacofilho, 2021-02-17:
And please feel free to join our developer's Discord channel: <https://discord.gg/f5HKCAadsr>.

planeturban, 2021-02-17:
```
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:88:14:00:ff:05:ff:ff:ff:ff:ff:00:00:00:00:00:00:00:00:00:00:00:00:01:00:00:00:00:00:07:14:1a:1e:33:28:4c:2d:cc:32:f2:3c:ff:3c:ff:14:00:23:33:28:80:31:80:32:f2:3c:ff:3c:ff:00:00:00:00:00:18
[INFO] __main__: Could not write to device. Res: -1
Traceback (most recent call last):
  File "/root/liquidctl/liquidctl/cli.py", line 321, in main
    _print_dev_status(dev, dev.initialize(**opts))
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 211, in initialize
    res = self._send_set_cooling()
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 419, in _send_set_cooling
    return self._send_command(_FEATURE_COOLING, _CMD_SET_COOLING, data=data)
  File "/root/liquidctl/liquidctl/driver/hydro_platinum.py", line 376, in _send_command
    self.device.write(buf)
  File "/root/liquidctl/liquidctl/driver/usb.py", line 435, in write
    raise OSError('Could not write to device. Res: %s' % res)
OSError: Could not write to device. Res: -1
[ERROR] __main__: Unexpected OS error with Corsair Hydro H100i Pro XT (experimental): OSError: Could not write to device. Res: -1
```

(I'll join the discord later this evening (CET)).

MarshallAsch, 2021-02-18:
@jonasmalacofilho  could the issue that @planeturban is having now be related to the multiple commands not being sent atomically, or the messages being sent out of order.

jonasmalacofilho, 2021-02-19:
Yes, that is likely.  Unfortunately locking entire devices is a pretty big change, with some serious ramifications, and will need to be a minor (instead of patch) release.

Given that #285 doesn't actually solve any immediate issues, I'm leaning towards not including it in 1.5.1, as I had originally planned.  It's not a simple cherry-pick, as there are a few merge conflicts, and despite all of our testing, it could still break something.

PR #285 (2021-02-17): added basic file locking capabilities for unix and windows
Closes #274

This should solve issues with multiple instances of liquidctl running at the same time having issues with concurrent access of the data file. 

This is being resolved by using lock files to make the functions for the `RuntimeStorage` class atomic. 

Also adds the function `load_store()` which can get a value, update it and then save it atomically.

- [x] write tests to validate that the calls are atomic
- [x] write tests to validate that multiple `load()`'s can happen at the same time
- [x] update all usages to use the ato [... cut]