liquidctl/liquidctl #278: Corsair H100i PRO XT doesn't work anymore

Jdelefosse, 2021-01-29:
Hi,

I just discovered this wonderful soft two days ago. I wanted to get rid of iCue so I jump on liquidctl

I have a Corsair H100i PRO XT, and at the beginning it worked just fine.
Yesterday, I updated from 1.4.2 to 1.5.0 without problems, then I did some stress tests (OCCT and Prime95), and at some point my computer crashed (weirdly because my max temps for the CPU was at 76°C).
Since then I didn't manage to make liquidctl work, I even went back to the 1.4.2 version in case.

I have this error:
`C:\>liquidctl list --verbose
Device ID 0: Corsair H100i PRO XT (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c20
├── Release number: 0x0100
├── Bus: hid
├── Address: \\?\hid#vid_1b1c&pid_0c20#7&1e4e78f5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
└── Driver: HydroPlatinum using module hid

C:\>liquidctl initialize all
ERROR: Unexpected error with Corsair H100i PRO XT (experimental)
Traceback (most recent call last):
ValueError: source code string cannot contain null bytes`

I installed iCue to see if it was bugging as well but iCue seems to work fine.

jonasmalacofilho, 2021-01-29:
> Yesterday, I updated from 1.4.2 to 1.5.0 without problems,[...] at some point my computer crashed [...] Since then I didn't manage to make liquidctl work, I even went back to the 1.4.2 version in case.

That's quite odd.  How are you getting liquidctl, are you using our supplied executables or are you installing it with `pip`?

> ```
> ValueError: source code string cannot contain null bytes`
> ```

Can you provide the output of the same command with an additional `--debug` flag?

Jdelefosse, 2021-01-29:
I'm using the supplied executable.

Here the output with the debug flag:
`liquidctl initialize all --debug
[DEBUG] __main__: running liquidctl v1.4.2 (AppVeyor; liquidctl-1.4.2-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device b58e:9e84
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i PRO XT (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 1462:7c92
[DEBUG] liquidctl.driver.usb: probing drivers for device b58e:9e84
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2514
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2640
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:4060
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device b58e:9e84
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1462:7c92
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:43ee
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c07c
[DEBUG] __main__: device: Corsair H100i PRO XT (experimental)
[DEBUG] liquidctl.keyval: data in C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030
[DEBUG] liquidctl.keyval: stored leds_enabled=0 (in C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\leds_enabled)
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\pump_mode)
[DEBUG] liquidctl.keyval: loaded fan1_mode=2 (from C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\fan1_mode)
[DEBUG] liquidctl.keyval: loaded fan1_duty=100 (from C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\fan1_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 100% duty cycle
[DEBUG] liquidctl.keyval: loaded fan2_mode=2 (from C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\fan2_mode)
[DEBUG] liquidctl.keyval: loaded fan2_duty=100 (from C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\fan2_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 100% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[ERROR] __main__: Unexpected error with Corsair H100i PRO XT (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 300, in main
  File "liquidctl\driver\hydro_platinum.py", line 171, in initialize
  File "liquidctl\driver\hydro_platinum.py", line 360, in _send_set_cooling
  File "liquidctl\driver\hydro_platinum.py", line 318, in _send_command
  File "liquidctl\driver\hydro_platinum.py", line 88, in _sequence
  File "liquidctl\keyval.py", line 108, in load
  File "liquidctl\keyval.py", line 75, in load
  File "ast.py", line 59, in literal_eval
  File "ast.py", line 47, in parse
ValueError: source code string cannot contain null bytes`

jonasmalacofilho, 2021-01-29:
Oh, that makes sense.

The crash corrupted a file where liquidctl saves some temporary state.  And apparently I didn't succeed in making our parsing resilient enough to these types of issues.

Can you please attach (you may need to zip it) the file in

```
C:\Users\CUISSE~1\AppData\Local\Temp\liquidctl\vid1b1c_pid0c20\loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030\sequence
```

so I can make this more robust?

After that you can simply delete the file and liquidctl (1.4.2 and 1.5.0) should work again : )

---

That device really doesn't need the WinUSB driver (in fact, liquidctl will only try to access it through the stock HID driver).

Jdelefosse, 2021-01-29:
Here is the folder zipped.
[loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030.zip](https://github.com/liquidctl/liquidctl/files/5893433/loc1_1_0_20_7_1_4_78_5_0_0000_4_1_55_2_16_11_88_001111000030.zip)

And yes after deleting it, it's actually working again thanks.

commit e88d581c (2021-01-29): keyval: handle corrupted data

Closes: #278
commit 6188b0ce (2021-01-29): keyval: handle corrupted data

Closes: #278
closed 2021-01-29 by jonasmalacofilho