liquidctl/liquidctl #287: Provide a winget package.

DaEgi01, 2021-02-19:
Please provide a winget package that is always up to date with the newest release on github :)

jonasmalacofilho, 2021-02-19:
It's very cool to see Windows finally having a first-party package manager.

https://docs.microsoft.com/en-us/windows/package-manager

This will require creating an installer for liquidctl: winget operates on first-party installers, and is not in itself a packaging tool or format.

We might also want to sign these installers (at least for the releases), which probably involves an annual fee.  And, at the very least, it would be wise to *disable* the installer pipeline for pull requests.  I think most users will implicitly trust installers a lot more.

DaEgi01, 2021-02-20:
I can't currently help out, even if I'd love to.
This awesome project of yours saves me every day from that awful CAM software,
yet I just have too many side projects already going.

Regarding the packaging, maybe https://docs.microsoft.com/en-us/windows/msix/overview would be an option for windows?
Also I really think you can and should avoid any annual fees :)

jonasmalacofilho, 2021-02-20:
> Regarding the packaging, maybe https://docs.microsoft.com/en-us/windows/msix/overview would be an option for windows?

Inno and Nullsoft are also (documented to be) well supported.

> Also I really think you can and should avoid any annual fees :)

Just to clarify my previous comment, code signing is not documented to be a requirement.

But unsigned applications (or, more precisely, not signed with MS tools and using their certification scheme) could result in a degraded experience (e.g. will a "unsigned application" warning pop up every time the application is updated?).

There's also some ad-hoc data to suggest that "unsigned" applications can also  be subject to more false positive detentions in Defender and other AV software.

Finally, and more importantly, there's the issue I mentioned of implicit trust in installers; I really don't want to see users tricked into installing nefariously modified copies of our project.

jonasmalacofilho, 2021-09-10:
Related: #375

jonasmalacofilho, 2021-10-05:
No longer applies since we're no longer providing executable files for Windows (#370).

closed 2021-10-05 by jonasmalacofilho