liquidctl/liquidctl #288: [Feature] Add IPC support for external integration

jammehcow, 2021-02-22:
Would be awesome to have some way of communicating with the service/application from another application (e.g. a custom fan controller or a polybar script etc.) Whether that's IPC or some other communication protocol it would be super useful. I'm happy to open a PR for it and work this through since I would like to use it.

jonasmalacofilho, 2021-02-22:
First, did you saw that we currently have issue #123 open for providing machine-readable output from `liquidctl status` (probably `liquidctl initialize` and `liquidctl list` as well)?

In fact, I did some benchmarks a couple of days ago and now have a better idea of which structure to output.

This would already cover some use cases.  The other `liquidctl` commands generate no primary output; and exit codes can be used to check for unexpected errors.

---

You mentioned integrating with the [liquidctl] "service"... but we don't currently provide one.

_(This has been considered in the past, to integrate `extra/yoda`, `extra/windows/LQiNFO.py`, `extra/liquiddump` and other similar "extra" features that either require or benefit from a long running process and connection to a device._

_But I wasn't able to quickly come up with a simple design [considering issues like defining responsibilities, protocol, integration with liquidctl, distribution, etc.] and nobody else has tried to work on it since)._

What would be your particular use case?

P.S. I imagine that you also already checked our Python APIs, and that using them is also infeasible (or impractical) in your case.

jammehcow, 2021-02-23:
I didn't see the Python API, that's actually exactly what I'm looking for. Ignore this. Thanks for the reply!

closed 2021-02-23 by jammehcow