liquidctl/liquidctl #292: Lower speed settings for NZXT Smart Device

RMateusPT, 2021-02-26:
Hi,
I was looking for an alternative to keep my fans and leds working without using NZXT CAM Software. Although thare are some missing features, like monitoring CPU temps and adjusting fans accordingly or change led colors based on CPU/GPU temperatures. 

Anyway, my real issue, is that I know that the fans I'm using goes down to about 700rpm, but even when set to 5%, they are working at around 940rpm.

I'll leave some outputs
```
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe list
Device ID 0: NZXT Smart Device V2

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device V2
├── Fan 1 duty       50  %
├── Fan 1 speed    1001  rpm
└── Noise level      48  dB
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device V2
├── Fan 1 duty      40  %
├── Fan 1 speed    940  rpm
└── Noise level     48  dB

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device V2
├── Fan 1 duty       5  %
├── Fan 1 speed    940  rpm
└── Noise level     48  dB
```

Additional information about the system:

- Affected device: NZXT Smart Device V2
- Operating system: Windows 10
- Liquidctl version: liquidctl v1.5.0 (liquidctl-1.5.0-bin-windows-x86_64)

Thank you for your time.
Best regards,
Ricardo Mateus

jonasmalacofilho, 2021-02-26:
> I know that the fans I'm using goes down to about 700rpm,

In what conditions did you reach those speeds?  And are those DC or PWM fans?

Finally, can you include the output for `.\liquidctl.exe status --debug`?

RMateusPT, 2021-02-26:
Hi, 
These are PWM fans, the model is RF-AF12C-RB
When using NZXT CAM, the lower speed i could get was @ 20%, and the fan speed was around 700rpm. 

Here's the output @ 20%
```
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe --debug status
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1e71:2006
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 67:02:3a:00:3f:00:18:57:32:53:32:30:31:20:03:00:01:00:00:00:00:00:00:00:c3:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:14:32:32:00:00:00:00:00:14:32:32:00:00:00:00:00:30:00:00:00:00:00:00:00
NZXT Smart Device V2
├── Fan 1 duty      20  %
├── Fan 1 speed    963  rpm
└── Noise level     48  dB
```
And @ 50%
```
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe --debug status
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1e71:2006
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 67:02:3a:00:3f:00:18:57:32:53:32:30:31:20:03:00:01:00:00:00:00:00:00:00:ff:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:32:32:32:00:00:00:00:00:32:32:32:00:00:00:00:00:30:00:00:00:00:00:00:00
NZXT Smart Device V2
├── Fan 1 duty       50  %
├── Fan 1 speed    1023  rpm
└── Noise level      48  dB
```

Thank you.

jonasmalacofilho, 2021-02-26:
Can you should me the output of `.\liquidctl.exe set fan1 speed 20 --debug`?

I would also like to known the `status` output (no debug infos necessary) for 100% duty cycle.

---

It's always  possible that there's a subtle bug in liquidctl, but we really don't enforce any minimum duty cycle in software, and the device appears to think it's operating at a lower duty cycle.

This leads me to think that device hasn't properly recognized the type of fans.  Did you `initialize` the device before using it, and what did you hear during that process (for example, did the fans sped up and then slow down)?

Finally, do you any any other types of fans on the same physical chain/device channel?

RMateusPT, 2021-02-26:
Hi,
Yes I did initialize the device first, and yes it did cycle through the speed up and slow down for then fans. 
For now, there's only 1 fan connected to this device. 

Here's the output
```
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe --debug set fan1 speed 20
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1e71:2006
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device V2
[INFO] liquidctl.driver.smart_device: setting fan1 duty to 20%
[DEBUG] liquidctl.driver.usb: writting report 0x62 with 63 bytes: 01:01:14:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device V2
├── Fan 1 duty      20  %
├── Fan 1 speed    941  rpm
└── Noise level     50  dB

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe --debug set fan1 speed 100
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1e71:2006
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device V2
[INFO] liquidctl.driver.smart_device: setting fan1 duty to 100%
[DEBUG] liquidctl.driver.usb: writting report 0x62 with 63 bytes: 01:01:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device V2
├── Fan 1 duty      100  %
├── Fan 1 speed    1481  rpm
└── Noise level      48  dB

```
I have a smart device V1 (it's disconnected now) but if needed, I can connect it and make some debugs to see if there's any differences. I can confirm that the fan works at lower speeds on NZXT CAM. The CAM software is closed when using liquidctl

RMateusPT, 2021-02-26:
Hi again. You might be right. The issue might be with the Smart Device V2. Right now, even with CAM, I cant get the fan to work lower than 960RPM. I will re-install the Smart Device V1 and check again and will provide further data. 

Thank you.

jonasmalacofilho, 2021-02-26:
The device stills appears to think it's operating at the lower duty cycle (when requested),

> ```
> ├── Fan 1 duty      20  %
> ```

and the correct speed change when set to 100% also shows that it's using the appropriate control mode (DC) for this fan.

There must be some protocol quirk that we're missing, like a special flag to enable low speeds on DC control.  Can you please capture the USB traffic between the device and CAM, while setting different fan speeds?

_You can find some documentation on how to do that in [Capturing USB traffic](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md#capturing-usb-traffic-on-a-native-windows-host).  And feel free to join our [developer's Discord server](https://discord.gg/f5HKCAadsr) and ping me there._

Please take notes of the various duty values set in CAM (in order), as well the approximate resulting speeds in rpm.  This will help me to find find the relevant information in the traffic captures.

RMateusPT, 2021-02-26:
I've connected the V1. and it works as before. This seems to be a limitation with the V2 . 
With the V2, liquidctl doesn't detect the Fan1 as a DC nor shows values for Current and Voltage.

I'll try capturing USB traffic, i'll leave for now the output. 
```
PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status
NZXT Smart Device (V1)
├── Fan 1                  DC
├── Fan 1 current        0.06  A
├── Fan 1 speed           785  rpm
├── Fan 1 voltage        4.24  V
├── Fan 2                   —
├── Fan 3                   —
├── Firmware version    1.0.7
├── LED accessories         0
└── Noise level            53  dB

PS C:\liquidctl-1.5.0-bin-windows-x86_64>

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe status --debug
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 1e71:1714
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 21 bytes: 04:36:00:00:00:00:00:0b:5b:00:00:01:00:00:07:10:01:00:00:1e:00
[DEBUG] liquidctl.driver.usb: read 21 bytes: 04:36:00:00:00:00:00:0b:5b:00:00:01:00:00:07:20:00:00:00:1e:00
[DEBUG] liquidctl.driver.usb: read 21 bytes: 04:34:00:02:ed:00:00:04:18:00:06:01:00:00:07:01:00:00:00:1e:00
NZXT Smart Device (V1)
├── Fan 1                  DC
├── Fan 1 current        0.06  A
├── Fan 1 speed           749  rpm
├── Fan 1 voltage        4.24  V
├── Fan 2                   —
├── Fan 3                   —
├── Firmware version    1.0.7
├── LED accessories         0
└── Noise level            53  dB

PS C:\liquidctl-1.5.0-bin-windows-x86_64>

PS C:\liquidctl-1.5.0-bin-windows-x86_64> .\liquidctl.exe set fan1 speed 100 --debug
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[... 67 lines cut ...]
[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1462:7c94
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 046d:c534
[DEBUG] liquidctl.driver.usb: found HID device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ada
[DEBUG] liquidctl.driver.usb: found USB device 046d:c534
[DEBUG] liquidctl.driver.usb: found USB device 1e71:1714
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ee
[DEBUG] liquidctl.driver.usb: found USB device 1022:149c
[DEBUG] liquidctl.driver.usb: found USB device 1462:7c94
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] __main__: device: NZXT Smart Device (V1)
[INFO] liquidctl.driver.smart_device: setting fan1 duty to 10%
[DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:00:00:0a:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
PS C:\liquidctl-1.5.0-bin-windows-x86_64>
```

jonasmalacofilho, 2021-02-26:
> Something might have changed with a firmware update or something.
> I've connected the V1. and it works as before. This seems to be a limitation with the V2 .

Well, there has been a recent firmware update...

> Updated SmartDevice2 firmware version to 1.12 in order to accommodate a minor hardware update
> *—<https://www.reddit.com/r/NZXT/comments/kdtice/nzxt_cam_4170_released>*

And maybe the minimum fan speed threshold was accidentally affected.  You could report that to NZXT and ask them to release a bug fix for the firmware.

But are you still able to shutdown the SD V2 fans completely with CAM or liquidctl (by setting them to 0%)?

---

> With the V2, liquidctl doesn't detect the Fan1 as a DC nor shows values for Current and Voltage.

As far as we know the SD V2 doesn't report that information, unfortunately.  But it's not really significant that they are missing from our output.

_(Please let us know if CAM shows some or all of this extra information; if that's the case I could try to find this data in the traffic captures you'll provide)._ 

> I'll try capturing USB traffic, i'll leave for now the output.

Thanks!

RMateusPT, 2021-02-26:
> Well, there has been a recent firmware update...
>
>
> And maybe the minimum fan speed threshold was accidentally affected. You could report that to NZXT and ask them to release a bug fix for the [...]

Probably, I've installed the SD V2  yesterday, NZXT sent it as replacement for the SD V1, duo to another problem I'm having, it's not working properly with my Motherboard hearders (MSI B550) CAM loses connectivity with the USB Device. It did not solved the issue. I'm able to get it work, connecting it to an USB port on the Case or the back IO (I'm working with their support on that) 

> But are you still able to shutdown the SD V2 fans completely with CAM or liquidctl (by setting them to 0%)?

Yes I can. On both SD V1 and SD V2.
 
> > With the V2, liquidctl doesn't detect the Fan1 as a DC nor shows values for Current and Voltage.
> 
> As far as we know the SD V2 doesn't report that information, unfortunately. So it's not really significant that they are missing from our output.
> _(Please let us know if CAM shows some or all of this extra information; if that's the case I could try to find this data in the traffic captures you'll provide)._

I really don't need that info, but it's nice to have. NZXT CAM don't show it either.

RMateusPT, 2021-02-26:
I've captured the USB packets with Wireshark. This is for the SDV1. 
[wireshark_SDV1.zip](https://github.com/liquidctl/liquidctl/files/6052306/wireshark_SDV1.zip)

jonasmalacofilho, 2021-02-27:
Thanks!

Just to double check the SDV2 capture: you first set fan1 to 50%, then to 15%; both times using CAM.  Is that correct?

> I'm really thinking that the culprit is indeed the SD V2, when it comes to fan speeds and not Liquidctl.

Certainly appears so, either due to a firmware bug or, since the device is new, a hardware defect or new limitation.

Try to contact NZXT, and please let me know what they reply (and if they issue a firmware update).

---

Slightly off-topic, but I also noticed something rather odd (with the SDV2): the first time the serial number descriptor is requested, a fictitious 00000000001A value is returned; but just a tenth of second later, a new request is made, and a proper (looking) serial number is fetched.  And requests appear to be identical, including the language ID.

jonasmalacofilho, 2021-03-08:
@RMateusPT do you have any news here?

RMateusPT, 2021-03-08:
I'm sorry. I was having other issues with my computer, it seems my USB were not even working correctly duo to a Issue with my main board and CPU. Anyway, this is their answer:

`this is something that is known with these particular fans are DC fans so the ability to finely control the RPM on these is not completely guaranteed and they may be a bit higher with the new Smart Device due to the lack of PWM fine tuning.`

So, even if it was possible with the Smart Device V1, it seems that it's not an issue to be solved on the V2. I expected a more technical answer with a reason for this. 

I think we can close this, I'm very pleased with the support and I'm sorry that I took your time troubleshooting an issue that was not even related to your software. 

Best regrads,
Ricardo Mateus

closed 2021-03-08 by RMateusPT