liquidctl/liquidctl #293: NZXT H series H210i, SmartDevice 2, does not show leds

synic, 2021-02-26:
The LEDs are connected to the SmartDevice, and are on, however, it doesn't list them in liquidctrl:

```
[~]$ liquidctl list
Device ID 0: NZXT Smart Device V2
[~]$ liquidctl status
NZXT Smart Device V2
├── Fan 1 duty       50  %
├── Fan 1 speed    1217  rpm
├── Fan 2 duty       50  %
├── Fan 2 speed    1050  rpm
├── Fan 3 duty       50  %
├── Fan 3 speed    1138  rpm
└── Noise level      54  dB

[~]$ 
```

I am using Ubuntu 20.10

jonasmalacofilho, 2021-02-26:
Can you include the output of `liquidctl status --debug`?

Are those LEDs official NZXT accessories, and are they lit (presumably in their default static white mode)?

Finally, did you execute `liquidctl initialize`?

synic, 2021-02-26:
here is the status with `--debug`:

```
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: ignoring 0-0019, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: ignoring 0-0018, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: found HID device 0b05:1939
[DEBUG] liquidctl.driver.usb: found HID device 1209:2303
[DEBUG] liquidctl.driver.usb: found HID device 1209:2303
[DEBUG] liquidctl.driver.usb: found HID device 1209:2303
[DEBUG] liquidctl.driver.usb: found HID device 1209:2303
[DEBUG] liquidctl.driver.usb: found HID device 046d:4051
[DEBUG] liquidctl.driver.usb: found HID device 046d:4051
[DEBUG] liquidctl.driver.usb: found HID device 046d:4051
[DEBUG] liquidctl.driver.usb: found HID device 046d:4051
[DEBUG] liquidctl.driver.usb: found HID device 046d:4051
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 0451:82ff
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 0451:8440
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 0451:82ff
[DEBUG] liquidctl.driver.usb: found USB device 05a3:9331
[DEBUG] liquidctl.driver.usb: found USB device 0451:8442
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1209:2303
[DEBUG] liquidctl.driver.usb: found USB device 0b05:1939
[DEBUG] liquidctl.driver.usb: found USB device 8087:0029
[DEBUG] liquidctl.driver.usb: found USB device 1e71:2006
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 67:02:22:00:2e:00:16:51:53:57:39:31:38:20:03:00:02:01:01:00:00:00:00:00:d2:04:17:04:9d:04:00:00:00:00:00:00:00:00:00:00:32:32:32:00:00:00:00:00:32:32:32:00:00:00:00:00:38:00:00:00:00:00:00:00
NZXT Smart Device V2
├── Fan 1 duty       50  %
├── Fan 1 speed    1234  rpm
├── Fan 2 duty       50  %
├── Fan 2 speed    1047  rpm
├── Fan 3 duty       50  %
├── Fan 3 speed    1181  rpm
└── Noise level      56  dB
```

Yeah, it's just the one LED strip that came with the system, and yes, it's just in the bright white mode.

I did run `liquidctl initialize all`

jonasmalacofilho, 2021-02-26:
Thank you.

Weren't the connected accessories reported in the output from `initialize all`?

synic, 2021-02-26:
@jonasmalacofilho you are right, it's listed there. I just got messed up because it's not in the status output. Trying to set the LED colors does indeed work, though:

```
[~]$ liquidctl initialize all
NZXT Smart Device V2
├── Firmware version                      1.5.0  
└── LED 1 accessory 1    HUE 2 LED Strip 300 mm  

[~]$ liquidctl status        
NZXT Smart Device V2
├── Fan 1 duty       50  %
├── Fan 1 speed    1265  rpm
├── Fan 2 duty       50  %
├── Fan 2 speed    1125  rpm
├── Fan 3 duty       50  %
├── Fan 3 speed    1185  rpm
└── Noise level      54  dB

[~]$ liquidctl set led1 color fixed af5a2f
[~]$ liquidctl set led1 color fixed af5a2f# liquidctl set led2 color fading 350017 ff2608 --speed slower
[~]$ liquidctl set led1 color fading 350017 ff2608 --speed slower
```

Thank you!

closed 2021-02-26 by MarshallAsch