liquidctl/liquidctl #298: Synchronized access to devices

jonasmalacofilho, 2021-02-28:
Some devices require synchronized access and don't cope with receiving out-of-order packets (#274).

Fixing this will require our drivers to implement some form of locking of the devices, probably cooperative.  The most obvious alternative is file locking, but there's the issue of what happens when there are multiple `liquidctl` instances running *with different user privileges.*

In this scenario, using `$XDG_RUNTIME_DIR` (or a reasonable substitute on other platforms) could result in the locks not been honored; a similar problem is already observed in #246.  On the other hand, using a common and universally writable location could potentially violate security expectations, particularly when `liquidctl` and access to the devices is reserved for root/Administrator: for example, an unauthorized processes would be able to DoS `liquidctl` instances running as root.

Another issue and/or related problem is that, due to specifics of POSIX and other locks, file locking by itself only ensures that different *processes* are synchronized.  And we also need to synchronize any *threads* of a given process that may try to access a particular device.

Finally, whatever solution is picked, it should try as much as possible to avoid breaking existing workflows *that work well.*  Because of this, it might be reasonable to only implement synchronization for the devices that really need it... where, if necessary, we can argue that the breaking changes were justified to fix a severe bug.

P.S. The issue of devices locking up due to out-of-order packets isn't very common, and we had no (significant) issues until the Platinum/Pro XT coolers.  Additionally, in the first versions of liquidctl we mostly used libusb for everything, which required us to claim the particular interface we were going to use, which also ensured exclusive access at the kernel level.

jonasmalacofilho, 2021-06-04:
Update: this is currently stalled because I don't have a good way to handle different users and environments when picking where to place the device lock.  And even if we could synchronize the device without a file lock, the devices for which this feature is crucial (#274) also require storing some non-volatile information that must be agreed on by all users.

intelfx, 2021-12-12:
> P.S. The issue of devices locking up due to out-of-order packets isn't very common, and we had no (significant) issues until the Platinum/Pro XT coolers

Not locking up, but returning completely bogus data is very common in my experience when two `liquidctl` processes are trying to read sensors out of a HX1000i PSU.

intelfx, 2021-12-12:
> an unauthorized processes would be able to DoS liquidctl instances running as root

Perfect is the enemy of good. Is it really a concern that someone would be able to DoS a liquidctl process (considering that the absolute majority of hardware supported by liquidctl belong to the PC enthusiast market, which is 99.9% single user machines)?

jonasmalacofilho, 2021-12-13:
Let me start my saying that I really appreciated your input, even though it might not look like it in the next few paragraphs.  And you motivated me to come back and work on this.

---

> Perfect is the enemy of good. Is it really a concern that someone would be able to DoS a liquidctl process (considering that the absolute majority of hardware supported by liquidctl belong to the PC enthusiast market, which is 99.9% single user machines)?

First, I think you glossed over the "for example" part, as well as what came before it.  DoS is perhaps a poor example, in the sense that it seems unlikely to happen in practice, but the thesis was there are undesirable security implications of placing the lock in an universally writable location (which, in turn, would be the most obvious way to ergonomically support multi-user systems).

Additionally, "99.9% single user machines" is not accurate, at least not in the context of the sentence before the one you quoted.  All of these 99.9% machines run general purpose operating systems and have, almost always, at least two "users", or distinct privilege levels and corresponding filesystem permissions: the `root`/`Administrator` plus one or more unprivileged users.

And because liquidctl may be executed manually, during boot, and/or from within the shell (as some sort of extension to display sensor data), it is fairly common[^1] for both privileged and unprivileged execution to happen on the same machine.  Access to the devices liquidctl manages on Linux is also nearly universally restricted to root, at least by default.[^2]

More importantly, by "unauthorized processes" I didn't mean only those installed _intentionally_ by other users on the same machine.  The machine owner may want to guard hardware monitoring and life support devices from processes that shouldn't interact with them, even if they running under their own unprivileged user; for example, by restricting access to them to root or some other special-purpose user.  And if they choose to do that, liquidctl must not bypass it, either directly or indirectly.

>  Not locking up, but returning completely bogus data is very common in my experience when two liquidctl processes are trying to read sensors out of a HX1000i PSU.

If you're running Linux, don't you also have issues with the `corsair-psu` kernel driver?

> I had to work around it locally by wrapping liquidctl in a crude flock wrapper, but it would be very nice to have some sort of synchronization in liquidctl itself.

A somewhat similar idea I have for solving this is to split functionality between a single server instance and various clients.  But, besides being a _massive change,_ there are some open questions:

- should the server be started at boot?  (but how exactly are we going to test, maintain and distribute all platform-specific code/scripts to make that happen?)
- alternatively, should we have the client start the server if one isn't already running? (but what happens if the server was started by a non-root user: where does the server saves its data [if the server shuts down but is later restarted by a different user, the new instance must still see the old data]?)

[^1]: Based on the history of comments to various issues in this repository.
[^2]: But I imagine most Linux users install our udev rules.

intelfx, 2022-04-09:
(I have just realized that there is a new reply to this discussion.)

I still maintain that you are overthinking it.

> DoS is perhaps a poor example, in the sense that it seems unlikely to happen in practice, but the thesis was there are undesirable security implications of placing the lock in an universally [...]

Which exactly security implications do you have in mind? I do not see any.

Regardless, if you are concerned with unprivileged processes DoSing privileged ones, I suggest using POSIX record locks (`fcntl(F_SETLK)`) on the specific device node that is being accessed (i. e. `/dev/bus/usb/XXX/YYY`).

Exclusive POSIX record locks can only be placed on a fd open for writing, in which case the process (that has R/W access to the usb device node) can already do pretty much everything it wants and it is pointless to defend against such processes.

jonasmalacofilho, 2022-04-09:
> Which exactly security implications do you have in mind? I do not see any.

The security implication is that a unprivileged user may influence a process running as root, which I think violates a common expectation of users, scrips and programs using the liquidctl CLI or APIs.  But, no, I don't see practical ways to abuse this, much less in scale (and liquidctl isn't super popular).

Still, I don't think we should assume it can't happen, just because we can't imagine it.  (Or, as the maintainer, I don't feel comfortable making that assumption).

> I suggest using POSIX record locks (`fcntl(F_SETLK)`) on the specific device node that is being accessed (i. e. `/dev/bus/usb/XXX/YYY`).

That's a natural location to place the lock, but it wouldn't work on Windows, AFAIK.  (I mean it because it depends on the "everything is a file" abstraction; I know that Windows has a somewhat similar alternative to POSIX locks).  Or I am missing something?

I'm not sure if it would work on mac OS either, because what I've seen hidapi return as a path there doesn't (IIRC) really lock like a path [I could place a lock on]:

```
IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS11@14500000/USB2.0 Hub@14500000/AppleUSB20Hub@14500000/AppleUSB20HubPort@14540000/NZXT USB Device@14540000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice

# from https://github.com/liquidctl/liquidctl/issues/219#issuecomment-726982682
```

> Exclusive POSIX record locks can only be placed on a fd open for writing, in which case the process (that has R/W access to the usb device node) can already do pretty much everything it wants and it is pointless to defend against such processes.

Again not seeing a practical attack, but I would like to point out that this statement assumes that permissions are the same for all liquidctl devices, which doesn't have to be the case (e.g. the RGB controller would be writable by all, but the CPU cooler could be restricted to root).

intelfx, 2022-04-09:
Oh, I had no idea liquidctl was actually cross-platform.

Maybe it would still make sense to handle the (presumably) common case of liquidctl running on Linux, given that the proposed alternative (splitting liquidctl into client and server) is a task of huge scope with no indication of it happening any time soon?

> Again not seeing a practical attack, but I would like to point out that this statement assumes that permissions are the same for all liquidctl devices, which doesn't have to be the case (e.g. the RGB controller would be writable by all, but the CPU cooler could be restricted to root).

No, I did not imply any such thing.

Different devices are represented by different node files, after all. So I don't see a way how an unprivileged process that has access to the hypothetical RGB controller can affect a privileged process that is trying to access the hypothetical CPU cooler.

jonasmalacofilho, 2023-09-05:
It seems that most third-party tools that support Corsair coolers on Windows have converged into using a global `Global\CorsairLinkReadWriteGuardMutex` mutex to synchronize with themselves or any other (compatible) tools running on the same system.[^1]

[^1]: https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/605#note_1434949606