liquidctl/liquidctl #299: Gigabyte Aorus RX5700XT Graphic card

evghenix, 2021-02-28:
it can be controlled with Gigabyte RGB Fusion 2.0

The device appears to support the following features:

 - [x] monitoring of temperatures or other environment sensors
 - [x] monitoring of fan or pump speeds or duty cycles
 - [ ] monitoring of voltages, currents or power
 - [ ] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [x] configurable lighting of embedded LEDs
 - [ ] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [ ] USB
- [x] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ ] HID
- [ ] USB
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [x] unknown

Please also include any useful additional information, such as USB vendor and product IDs, the output of `lsusb -v` (Linux) or `system_profiler SPUSBDataType` (Mac OS), links to external resources or already collected traffic data.

```
    $ system_profiler SPPCIDataType

    Radeon RX 5700 XT:

      Name: ATY_GPU
      Type: gpu-controller
      Driver Installed: Yes
      MSI: Yes
      Bus: PCI
      Slot: Slot-1@0,1,0/0,0/0,0/0,0
      Vendor ID: 0x1002
      Device ID: 0x731f
      Subsystem Vendor ID: 0x1458
      Subsystem ID: 0x2316
      Revision ID: 0x00c1
      Link Width: x16
```

Finally, I can help with:¹

- [ ] testing changes on Linux
- [x] testing changes on Windows
- [x] testing changes on Mac OS
- [ ] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2021-02-28:
This obviously requires understanding the protocol that's used to interact with this device.

Unlike with NVIDIA GPUs, I don't know how to sniff this traffic on Windows, and I'm not even sure it's I²C (although that's likely).  On the other hand, at least we can try to reach out to AMD on this one.

in addition to figuring out the protocol, and implementing our specific driver, we also need to implement the generic interface between `liquidctl` and the kernel GPU driver...

- on Linux
- and, for it to be useful for @evghenix , on Mac OS
- [optionally, for completeness, on Windows and BSDs too]

We'll definitively need help with all of these tasks, so I encourage everyone interested and/or able to contribute.

evghenix, 2021-02-28:
Hi @jonasmalacofilho I cant try to sniff the traffic but I have no clue how to do that. Would be great if anyone can help me with this. Also, as I have multiboot system I can try to install Linux as well.

jonasmalacofilho, 2021-03-02:
Thanks!

I looked into this a bit and it seems that the bus is indeed I²C.

I talked to Adam @ OpenRGB and he thinks it should be possible to capture the relevant calls to ADL with API Monitor.  Looking at the ADL documentation, I would initiallly focus on [`ADL2_Display_WriteAndReadI2C()` and friends](https://gpuopen-librariesandsdks.github.io/adl/group__I2CDDCEDIDAPI.html).

That said, I don't have a lot of experience with this type of debugging on Windows; try to hop into the `#reverse-engineering` channel on [OpenRGB's Discord server](https://discord.gg/xVkFxZUK).  And apparently some cards/applications use something other than ADL[1].

Oh, also feel free to join our [Discord server](https://discord.gg/f5HKCAadsr) (for developers).

---

Unfortunately, development for Linux will be blocked for the time being, due to limitations on the kernel level[2].

[1] https://discord.com/channels/699861463375937578/702322217622700052/816197094065831937
[2] https://github.com/twifty/aura-gpu/issues/4

jonasmalacofilho, 2021-03-02:
Adam @ OpenRGB:

> probably would start by verifying if it has a single LED zone or multiple LED zones
> a ton of Gigabyte GPUs (with single LED zones) use the same protocol
> but we don't know much about the ones with multiple zones

jonasmalacofilho, 2021-08-06:
**Triage: close without fixing (due to the platforms mentioned in this issue)**

At this time, the maintainers don't plan to pursue SMBus/I²C support for platforms other than Linux, and Mac/Windows are the only platforms of interest for this issue's author.  Contributions in the area of SMBus/I²C support for Windows, Mac and/or BSDs would be most welcome.

But if there are users interested in Gigabyte Aorus RX5700XT support _on Linux,_ please [open a new issue](https://github.com/liquidctl/liquidctl/issues/new?assignees=&labels=new+device&template=new_device.md&title=Gigabyte%20Aorus%20RX5700XT) for it and let us know.

closed 2021-08-06 by jonasmalacofilho