liquidctl/liquidctl #300: rm1000i Total PSU power show output power instead of input power

babafgd, 2021-03-01:
Hi, 

I have a Corsair rm1000i PSU and notive that the total power is the sum of all power, not the input power. 

ie : output power = input power * efficiency.

The input power needed by the PSU is the sum of all needed power divided by efficiency. 

That is the power delivered by the outlet, what we need for the complete system. 

Additional information about the system:

- Affected device: (tested with Corsair PSU rm1000i)
- Operating system: (linux Ubuntu 18.04.4 LTS)
- Liquidctl version: (liquidctl v1.5.1 python pip3 install)

example : read from Power outlet wattmeter : 1022W (confirmed with Corsaire iCue tool) 

read from liquidctl : 
```
Corsair RM1000i                                                                                                                                                                                                    
├── Current uptime                   0:18:01                                                                                                                                                                       
├── Total uptime            24 days, 0:18:01                                                                                                                                                                       
├── Temperature 1                       66.2  °C                                                                                                                                                                   
├── Temperature 2                       31.2  °C                                                                                                                                                                   
├── Fan control mode                Hardware                                                                                                                                                                       
├── Fan speed                            968  rpm                                                                                                                                                                  
├── Input voltage                     230.00  V                                                                                                                                                                    
├── Total power                       928.00  W                                                                                                                                                                    
├── +12V OCP mode                 Multi rail                                                                                                                                                                       
├── +12V output voltage                11.84  V                                                                                                                                                                    
├── +12V output current                78.00  A                                                                                                                                                                    
├── +12V output power                 922.00  W                                                                                                                                                                    
├── +5V output voltage                  5.02  V                                                                                                                                                                    
├── +5V output current                  1.38  A                                                                                                                                                                    
├── +5V output power                    6.50  W                                                                                                                                                                    
├── +3.3V output voltage                3.27  V                                                                                                                                                                    
├── +3.3V output current                0.56  A                                                                                                                                                                    
└── +3.3V output power                  1.50  W      
```

jonasmalacofilho, 2021-03-01:
If I understood correctly, you are suggesting that we should replace "Total power" by "Input power".

Unfortunately, the device only provides us with the total [output] power.  In order for us to estimate the input power we would likely need to lift the efficiency curves from the PSU manuals, assuming they are accurate.  Another possibility would be ask Corsair for the data in a machine readable format...

Anyway, this is actually a pretty good idea, and I would very much like to have this in liquidctl.  But first someone needs to do the hard work of either [successfully] reaching out to Corsair or lifting *all* the efficiency curves from the manuals.

babafgd, 2021-03-01:
Hi,

You don’t need so big work on efficiency curb as Corsair do the job for you.

On the windows tool you get access to all three : input, output, efficiency.

So I guess only the good Data address is needed.

For « total power » it’s 0xEE.

I have both windows and Linux on the computer, So I can try to see if I
found it.

I would be happy to help !

jonasmalacofilho, 2021-03-01:
So far I've only seen the output power (with `MFR_SPECIFIC_EE`) in the captures I have.  But I suppose the device could support either or both standard `READ_IIN` or `READ_PIN` commands, or some other `MFR_SPECIFIC_??` command that would allow us to properly compute the input power.

_(Note: PMBus has a `READ_POUT` command, but Corsair chooses to use a custom `MFR_SPECIFIC_EE` instead)._

Try to [capture the USB traffic](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md) and send me the PCAP files.

babafgd, 2021-03-01:
Will do !

babafgd, 2021-03-01:
Hi,

So first a picture to show what we are looking for

[image: image.png]

As you see PowerIn is bigger than PowerOut, the ratio between them is
efficiency.
Power in -> what is at the wall socket
Power out -> what is needed inside the Computer (out of the PSU)
from liquidctl I have seen only powerOut info, that I have already
confirmed with a Wattmeter.

On my application, I'm looking for the powerIn, the Computer consumption.

Then about the wireshark, see attachment as raw data of few seconds (I have
seen data are updated each seconds)

The powerIn and out values change regularly, 72, 78, 84, 91 are regular
values for powerIn.

Thanks

jonasmalacofilho, 2021-03-01:
Please don't use GitHub's reply-by-email feature.  If you check your issue on the repository, you'll see that what gets to the other side (=me) is completely borked.  For example, I can't see the attachment(s).

---

> As you see PowerIn is bigger than PowerOut, the ratio between them is efficiency.
I'm obviously not disputing that `POUT = PIN * efficiency`.

Let's look at the captures and see what the device sends to iCue, and continue from there.

_(again, the attachments were stripped, so I haven't seen the Wireshark capture yet)_

babafgd, 2021-03-01:
I'm still new to github :) Here is the picture 

[image]

And here is the capture : 

[CorsairLink-rm1000i.zip](https://github.com/liquidctl/liquidctl/files/6064540/CorsairLink-rm1000i.zip)

let me know How I can support more.

jonasmalacofilho, 2021-03-02:
The traffic you captured doesn't show any use of standard PMBus `READ_IIN` or `READ_PIN` commands by (presumably) iCue.

Still, I decided to give it a shot.  Can you please test c87788f?

babafgd, 2021-03-02:
Hi, 

So as said I still have to learn a few things. 

I have tested on my ubuntu linux and made the few commadn line : 

```
git clone https://github.com/liquidctl/liquidctl
cd liquidctl/
sudo apt install python3-pytest
sudo pytest-3
```

I had 3 errors at this step, not sure they are related to liquidctl : 
```
FAIL tests/test_ddr4.py::test_tse2004_ignores_not_allowed_buses
FAIL tests/test_nvidia.py::test_evga_pascal_finds_devices
FAIL tests/test_nvidia.py::test_rog_turing_does_not_find_devices
```

Then I tested the actual version  of liquictl : 

```
sudo python3 -m liquidctl.cli initialize
sudo python3 -m liquidctl.cli status
```

and got ` Total power                        926.00  W`

So I presume my version has been compiled and it's working as expected from git sources. 

Then I get the commit you point out : 

```
git checkout c87788f
sudo python3 -m liquidctl.cli initialize
sudo python3 -m liquidctl.cli status
```

and I got this error : 

`ERROR: Unexpected error with Corsair RM1000i: ZeroDivisionError: division by zero`

looks like something is wrong, happy to know if I made a mistake. 

Thanks

jonasmalacofilho, 2021-03-02:
> I had 3 errors at this step, not sure they are related to liquidctl :

I would need the complete output from the errors to know what's going on.  But if I had to guess, it's probably a missing dependency or kernel module.

---

> `ERROR: Unexpected error with Corsair RM1000i: ZeroDivisionError: division by zero`

I messed up and forgot to handle division by zero.  That said, the fact that input power is zero isn't great news...

Can you comment out the culprit

https://github.com/liquidctl/liquidctl/blob/c87788ff11dfc6b91a847f2db85d7714058559cf/liquidctl/driver/corsair_hid_psu.py#L123

and show me the complete output with `--debug`?

```
$ sudo python3 -m liquidctl.cli status --debug
```

babafgd, 2021-03-02:
Hi Jonas, 

here is the output with commented line and `--debug`

```
minerstat@Rom1-Rig2:~/liquidctl$ sudo python3 -m liquidctl.cli status --debug
[DEBUG] __main__: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1997:2433
[DEBUG] liquidctl.driver.usb: found HID device 1997:2433
[DEBUG] liquidctl.driver.usb: found HID device 1997:2433
[DEBUG] liquidctl.driver.usb: found HID device 1997:2433
[DEBUG] liquidctl.driver.usb: found HID device 1997:2433
[DEBUG] liquidctl.driver.usb: found HID device 5131:2007
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1c0d
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair RM1000i
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1c0d
[DEBUG] liquidctl.driver.usb: found USB device 5131:2007
[DEBUG] liquidctl.driver.usb: found USB device 1997:2433
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] __main__: device: Corsair RM1000i
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:97:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:ee:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:ee:9f:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:d2:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:d2:39:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:d1:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:d1:39:95:20:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:8d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:8d:cc:f0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:8e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 03:8e:8d:f0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 03:f0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
├── Total power (output)               318.00  W
├── +12V OCP mode                  Multi rail
├── +12V output voltage                 11.95  V
├── +12V output current                 26.50  A
├── +12V output power                  316.00  W
├── +5V output voltage                   5.02  V
├── +5V output current                   1.25  A
├── +5V output power                     6.00  W
├── +3.3V output voltage                 3.30  V
├── +3.3V output current                 0.56  A
└── +3.3V output power                   1.50  W
```

jonasmalacofilho, 2021-03-02:
Apparently neither `READ_IIN` (input current) or `READ_PIN` (input power) are supported by the device.

And from your traffic capture, there is no other relevant data that iCue fetches from the device.  It seems that iCue has the efficiency tables...

_(You did capture the traffic from/to iCue, right?  Or was the capture made of another software?)_

babafgd, 2021-03-02:
Hi, No I just notice, I was using Corsair link software.

I made a new capture with iCue software if it change anything. 

The power was around 122 this time

[CorsairLink-rm1000i-icue.zip](https://github.com/liquidctl/liquidctl/files/6071481/CorsairLink-rm1000i-icue.zip)

jonasmalacofilho, 2021-03-02:
Same thing, just `READ_VIN` is fetched for the input side.

We need the efficiency curves, either from the manuals (not that much work, but may not be very accurate and/or precise) or by reaching out to Corsair.  Can you help with this?

P.S. For completeness our driver, we need the curves for all RMi and HXi models.

babafgd, 2021-03-02:
Hi, 

Ok I will see what I can get from Corsair. I'm not convince they do it with curve calculation as it sounds weird to me. But I will ask them for some curves /data.

jonasmalacofilho, 2021-03-03:
Data for 80 PLUS certification is available at <https://www.clearesult.com/80plus/manufacturers/115V-Internal> and related pages.  But most devices have only been tested at ~115V input (where they are less efficient than at ~230V input).

babafgd, 2021-03-03:
Hi Jonas, 

I found the efficiency curves in the Rmi & Hxi Manual from Corsair. I have added also Ax series as I plan to move to an ax1500i soon. 

https://www.corsair.com/corsairmedia/sys_master/productcontent/RMi_Manual.pdf
https://www.corsair.com/corsairmedia/sys_master/productcontent/HXi-Manual.pdf
https://www.corsair.com/corsairmedia/sys_master/productcontent/AXi_Series_Manual.pdf
https://www.corsair.com/corsairmedia/sys_master/productcontent/AX1500i_Manual.pdf
https://www.corsair.com/corsairmedia/sys_master/productcontent/AX1600i_C14_Manual_Web.pdf

I can spend some time putting curve in a nice table, what would be the preffered output ? 

Thanks

babafgd, 2021-03-03:
Hi, 

another comment, on another project I found a similar PSU monitor reding feature, and the have implemented simple curve table for the powerIn calculation : 

https://github.com/Hagbard-Celin/cpsumon/blob/master/libcpsumon/src/cpsumon.c

jonasmalacofilho, 2021-03-03:
> I have added also Ax series as I plan to move to an ax1500i soon.

We don't support the AXi series (yet?).

> I can spend some time putting curve in a nice table, what would be the preffered output ?

We only need the efficiency values at 10%, 20%, 50% and 100% (the rest is interpolated anyway).  And the following format would be fine:

```
Model    Input (V)  Efficiency @ 10%   @ 20%   @ 50%  @ 100%
RM1000i        115             .8634   .9063   .9149   .8828
RM1000i        230 ...
...
```

Note that the efficiencies at 115 V can be lifted from the link I posted above.   But please let me know if you find any huge discrepancies between the 80 PLUS data and what's on the manual.

jonasmalacofilho, 2021-03-03:
For reference, I'll use this data and [`collected-device-data/psu-efficiency/interpolate`](https://github.com/liquidctl/collected-device-data/blob/master/psu-efficiency/interpolate) to get interpolations that behave like:

[image]

```
input/rated power = 0.1064 × load² + 1.013 × load + 0.01364
```

babafgd, 2021-03-03:
Hi, 

I did the exercice and the two curves from Corsair manual and 80 plus certificatino does not match, but are close enough for a first step 

[image]

babafgd, 2021-03-03:
Note : the Horizontal scale is logarithm 10 base.

jonasmalacofilho, 2021-03-03:
@babafgd I no longer can see your comment that started with

> I did the exercice and the two curves from Corsair manual and 80 plus certificatino does not match, but are close enough for a first step

What happened?

babafgd, 2021-03-03:
Hi, I should have made a mistake as I have seen double message and removed one ... 

Here is the Manual picture and the correlation I did : 

[image]

[image]

jonasmalacofilho, 2021-03-03:
Indeed, there are measurable differences:

```
Model    Input (V)  Efficiency@10%   @20%   @50%  @100%
RM1000i        115*           .863   .906   .915   .883  // from 80 PLUS data
RM1000i        115            .852   .892   .905   .876
RM1000i        230            .854   .895   .910   .886
```

The two bottom lines are data I lifted from the RMi manual in a fairly accurate way, by drawing guide lines on Inkscape and dividing distances.

[image]

---

That said, the data from the manual, good or bad, is certainly more than enough for what we want to do.

And given the differences from the 80 PLUS data, I suppose we can safely reduce the (absolute) precision to 0.5% or even 1%, making it a lot easier and faster to compile by eye (or just with vertical guides).

jonasmalacofilho, 2021-03-03:
> But I will ask them for some curves /data.

Did you reach out to them?

jonasmalacofilho, 2021-03-04:
```
Model    Input (V)  Efficiency@10%   @20%   @50%  @100%
RM1000i        115            .852   .892   .905   .876
RM1000i        230            .854   .895   .910   .886
RM850i         115            .85    .89    .91    .88
RM850i         230            .86    .91    .92    .90
RM750i         115            .83    .89    .91    .88
RM750i         230            .85    .90    .92    .90
RM650i         115            .83    .89    .91    .88
RM650i         230            .84    .90    .91    .89
HX1200i        115            .86    .91    .92    .90
HX1200i        230            .88    .92    .94    .92
HX1000i        115            .88    .91    .92    .89
HX1000i        230            .88    .92    .93    .90
HX850i         115            .87    .90    .92    .89
HX850i         230            .88    .91    .93    .91
HX750i         115            .87    .91    .92    .89
HX750i         230            .88    .92    .93    .91
```

babafgd, 2021-03-04:
That sounds really good ! I have asked in the corsair forum about USB data. in other hand I found thread of people asking for more details and the official answer is that Corsair cannot disclose third party IP. I would assume that with the table you made from 80 plus data and the good interpolation we can be close to the manual Curve and it should be good enough for a first approach. I will then be able to confirm it on a rm1000i with different load and a physical meter at wall.

jonasmalacofilho, 2021-03-05:
Can you give commit ~~9bee8da0bacd~~ 7f53e6c90fc1 a try?

jonasmalacofilho, 2021-03-05:
Note to myself, about the changelog: if theses changes get merged I'll also need to mention the renaming of "Total power" to "Total power output".

babafgd, 2021-03-05:
Hi, 

I was fighting a little with git, but finally get it working and here is the output of the 7f53e6c :

Corsair RM1000i
├── Current uptime                    3:31:04
├── Total uptime             27 days, 4:31:04
├── Temperature 1                        53.2  °C
├── Temperature 2                        19.0  °C
├── Fan control mode                 Hardware
├── Fan speed                             920  rpm
├── Input voltage                      230.00  V
├── Total power output                 918.00  W
├── Estimated input power             1030.73  W
├── Estimated efficiency                   89  %
├── +12V OCP mode                  Multi rail
├── +12V output voltage                 11.86  V
├── +12V output current                 77.00  A
├── +12V output power                  912.00  W
├── +5V output voltage                   5.02  V
├── +5V output current                   1.25  A
├── +5V output power                     6.00  W
├── +3.3V output voltage                 3.27  V
├── +3.3V output current                 0.56  A
└── +3.3V output power                   1.50  W

I have check at the wall plug and it's around 1010W So we are pretty close. I will test beginning of next week to make some more test and load correlation. 

Thanks a lot for the effort !

jonasmalacofilho, 2021-03-05:
> I have check at the wall plug and it's around 1010W

How precise and accurate is that meter supposed to be?

Still, 1010 W makes more sense with a 80 Plus Gold rated and fairly new 1000 W PSU at 230 V input (assuming the rating matches reality, but let's not open that can of worms).

In fact, based on the efficiency graph in the manual, the RM1000i wouldn't have passed 80 Plus Gold at 230 V.

_(Well, it really isn't listed on the official 80 Plus website, so maybe it *wasn't* certified at 230 V?)_

Anyway, your measurement suggests that the graph in manual underestimates the efficiency, at least at 230 V.  Can you also validate the input power and efficiency estimates from iCue at medium to high load levels?

jonasmalacofilho, 2021-03-06:
I gave this more thought and, while I appreciate any additional validation you decide do, I think for now we should stick with the advertised efficiency data from the manuals.

> At this point it is important to remember that efficiency, and consequently power draw, are functions of more than just the total power output. Thus, the data in the user manuals is probably significantly less precise than it appears to be, and we believe the same could be true for the values displayed by iCue and CorsairLink.
> —*[Differences in efficiency data](https://github.com/liquidctl/liquidctl/blob/master/docs/corsair-hxi-rmi-psu-guide.md#appendix-differences-in-efficiency-data) in our guide for these PSUs*

In the end, the data we're using is public and official, and thus our estimates are easy for users to verify.  We couldn't say the same if we were to use *ad hoc* collected data.

Still,

> we encourage Corsair to make more of its efficiency data public, which would hopefully allow liquidctl to present more precise estimates.

---

Unless bugs are found, I'll close this issue in a few days; the changes (so far) have already been merged into our main branch.

babafgd, 2021-03-08:
Hi, 

I think we made enough progress here until we have more accurate data from corsair. at least it would fit my need. Thanks a lot of the effort.

closed 2021-03-08 by jonasmalacofilho