liquidctl/liquidctl #303: Lighting channel names not being recognized in Smart Device V2 driver

CaseySJ, 2021-03-04:
There may be a regression in `liquidctl 1.5.1` where the lighting channel names `led1`, `led2`, etc. are not being recognized on macOS:
```bash
casey@X570-Taichi ~ % liquidctl --version
liquidctl v1.5.1 (homebrew; liquidctl 1.5.1)

casey@X570-Taichi ~ % liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)

casey@X570-Taichi ~ % liquidctl -v status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     24.2  °C
├── Fan speed               945  rpm
├── Pump speed             1853  rpm
└── Firmware version      6.0.2  
```
`KeyError: led1`:
```bash
casey@X570-Taichi ~ % liquidctl -v set led1 color spectrum-wave
INFO: 'led1'
Traceback (most recent call last):
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/cli.py", line 328, in main
    _device_set_color(dev, args, **opts)
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/cli.py", line 218, in _device_set_color
    dev.set_color(args['<channel>'], args['<mode>'], color, **opts)
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/driver/kraken2.py", line 175, in set_color
    byte2 = mod2 | _COLOR_CHANNELS[channel]
KeyError: 'led1'
ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72): KeyError: 'led1'
```
`KeyError: led` in more detail:`
```bash
casey@X570-Taichi ~ % liquidctl -v -g set led1 color spectrum-wave
[DEBUG] liquidctl.cli: running liquidctl v1.5.1 (homebrew; liquidctl 1.5.1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: found HID device 05ac:024f
[DEBUG] liquidctl.driver.usb: found HID device 05ac:024f
[DEBUG] liquidctl.driver.usb: found HID device 05ac:024f
[DEBUG] liquidctl.driver.usb: found HID device 04ca:008a
[DEBUG] liquidctl.driver.usb: found HID device 04ca:008a
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 04ca:008a
[DEBUG] liquidctl.driver.usb: found USB device 05ac:024f
[DEBUG] liquidctl.driver.usb: found USB device 1e71:170e
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
[INFO] liquidctl.cli: 'led1'
Traceback (most recent call last):
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/cli.py", line 328, in main
    _device_set_color(dev, args, **opts)
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/cli.py", line 218, in _device_set_color
    dev.set_color(args['<channel>'], args['<mode>'], color, **opts)
  File "/usr/local/Cellar/liquidctl/1.5.1/libexec/lib/python3.9/site-packages/liquidctl/driver/kraken2.py", line 175, in set_color
    byte2 = mod2 | _COLOR_CHANNELS[channel]
KeyError: 'led1'
[ERROR] liquidctl.cli: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72): KeyError: 'led1'
casey@X570-Taichi ~ % 
```

From the docs:
https://github.com/liquidctl/liquidctl/blob/master/docs/nzxt-hue2-guide.md
[image]

jonasmalacofilho, 2021-03-04:
Hi @CaseySJ, good to see you again!

The Kraken X2 coolers only have `logo`, `ring` and `sync` lighting channels ([guide](https://github.com/liquidctl/liquidctl/blob/master/docs/kraken-x2-m2-guide.md#rgb-lighting)).  Can you check if those work?

jonasmalacofilho, 2021-03-04:
> The problem is that my Smart Device V2 is not being recognized by macOS itself. Not a liquidctl issue.

Oh, that makes sense... lighting commands intended for the SD2 were then reaching the Kraken2 driver, hence the KeyError.

closed 2021-03-04 by jonasmalacofilho

CaseySJ, 2021-03-04:
@jonasmalacofilho 

Hello again! Alas, it seems there is a fault in my Hackintosh where the Smart Device V2 is not being recognized by macOS. So I was trying to configure lighting channels for the Kraken instead of the Smart Device V2!!!

CaseySJ, 2021-03-04:
Let me fix the Smart Device V2 (could be a hardware connection issue) and then I test see if there's any I/O error as mentioned by @sgtcoder.

Nevertheless, please close/delete this issue as it's User Error.

jonasmalacofilho, 2021-03-04:
> and then I see if there's any I/O error as mentioned by sgtcoder.

Thanks!