liquidctl/liquidctl #304: Can't Detect NZXT in Xubuntu 20

sgtcoder, 2021-03-05:
When running `sudo liquidctl list` my NZXT Smart Device V2 is not showing up. I tried `sudo liquidctl initialize all` and the device still did not show up.

NZXT Cam 4 in Windows 10 Detects All of My Devices
- NZXT Kraken X (X42, X52, X62 or X72)
- NZXT Smart Device (V1)
- NZXT Smart Device (V2)

```
uname -r
5.11.0-1.1-liquorix-amd64
```

```
cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

```
sudo liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
Device ID 1: NZXT Smart Device (V1)
```

```
lsusb
Bus 001 Device 009: ID 1e71:200e NZXT
Bus 001 Device 008: ID 1e71:1714 NZXT
Bus 001 Device 006: ID 1e71:170e NZXT
```

```
sudo lsusb -v -d1e71:200e

Bus 001 Device 009: ID 1e71:200e NZXT 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x200e 
  bcdDevice            1.0c
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 8D851375525
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     463
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

```
sudo liquidctl list --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: found HID device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: found HID device 1e71:200e
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 174c:3074
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1e71:200e
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 1e71:1714
[DEBUG] liquidctl.driver.usb: found USB device 1e71:170e
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 174c:2074
[DEBUG] liquidctl.driver.usb: found USB device 046d:082d
[DEBUG] liquidctl.driver.usb: found USB device 8087:0a2b
[DEBUG] liquidctl.driver.usb: found USB device 0951:16be
[DEBUG] liquidctl.driver.usb: found USB device 0951:16d3
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 49973535343
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: Kraken2
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver

Device ID 1: NZXT Smart Device (V1)
├── Vendor ID: 0x1e71
├── Product ID: 0x1714
├── Release number: 0x0200
├── Serial number: 496F266C3934
├── Bus: hid
├── Address: /dev/hidraw4
└── Driver: SmartDevice
[DEBUG] liquidctl.cli: hierarchy: _CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver
```

Additional information about the system:

- Affected device: NZXT Smart Device V2
- Operating system: Xubuntu 20.04.2 LTS
- Liquidctl version: 1.5.1

jonasmalacofilho, 2021-03-05:
It seems that your NZXT Smart Device V2 has a new product id (PID) `200e`.

_(Which is odd considering that, supposedly, the old PID (`2006`) recently got a firmware update to address hardware differences in newer units, which shouldn't be necessary if the newer units were going to use a new PID anyway)._

Do you think you can provide some USB [traffic captures](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md)?

If not, you could try to add a new entry to [`SmartDevice2.SUPPORTED_DEVICES`](https://github.com/liquidctl/liquidctl/blob/7a15ec66059e4c223c8198c053d828507128a0cb/liquidctl/driver/smart_device.py#L315-L332), enabling the driver for the new PID.  It should *probably* work...  please let me know either way.

sgtcoder, 2021-03-05:
@jonasmalacofilho Let me try what you provided. It's actually a RGB and Fan Controller, from my understanding is the same as Smart Device V2. I just received the device fresh from NZXT today: https://www.nzxt.com/products/rgb-fan-controller

jonasmalacofilho, 2021-03-05:
That's good to know, we actually differentiate between the two when reporting data to the user.

Anyway, `0x200e` is still a new PID for the RGB & Fan Controller (which we previously new with PID `0x2009`).

sgtcoder, 2021-03-05:
@jonasmalacofilho 

I added this and it is initializing, but also have to do under root so I can do sudo because it's not in udev yet either (in order to set).

Thank you for the quick response.

```
(0x1e71, 0x200e, None, 'NZXT RGB & Fan Controller', {
    'speed_channel_count': 3,
    'color_channel_count': 2
}),
```

jonasmalacofilho, 2021-03-05:
Already added to the main branch.  Thanks!

closed 2021-03-05 by jonasmalacofilho

sgtcoder, 2021-03-07:
@jonasmalacofilho not sure if you want me to open up a new ticket, but I am getting an error when trying to access this device in Windows 10. It shows up in liquidctl list, but when trying to intialize it gives me this error: 

```
ERROR: Unexpected error with NZXT RGB & Fan Controller: UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
NZXT RGB & Fan Controller
```

jonasmalacofilho, 2021-03-07:
No need for now: unlike that old issue with lots of subscribers, only us two and a handful of other people will get notifications from new comments here.

Can you show me the output with `--debug`?

And what terminal shell are you using (Windows Command Prompt, Powershell, other?)?

Finally, if not running our PyInstaller-based executable, what Python version are you using?

sgtcoder, 2021-03-07:
That works and makes sense. I need to fix my notifications (they used to send to pushover).

I did try Git BASH since I am a BASH guy, but a batch script in command prompt integrated better (to put a shortcut in the shell:apps). I also have some feedback on the documentation for both Windows and Linux since I had to do a few things more.

I couldn't use your compiled binary yet because it doesn't have our latest code compiled. So I installed through pip, then I installed through GIT so I can have that patch. The Python from the Windows Store was giving me some permission issues. So I downloaded Python 3.8 from the actual site.

It's no longer showing that error if I run manually (probably after the first time, I saw the error show up on boot, I will disable from boot and run manually so I can see the --debug). I do see this on initialize on Linux and Windows for only that new device, not sure why though: https://gyazo.com/9f323921220e90a11456ec3f6d6e0c2c

sgtcoder, 2021-03-07:
Here is the debug:

```
liquidctl initialize all --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.1 (d43c2522d044)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16d3
[DEBUG] liquidctl.driver.usb: found HID device 0951:16be
[DEBUG] liquidctl.driver.usb: found HID device 1e71:200e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT RGB & Fan Controller
[DEBUG] liquidctl.driver.usb: found HID device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.cli: device: NZXT RGB & Fan Controller
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x60 with 63 bytes: 02:01:e8:01:01:e8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x60 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 11:01:55:ff:6f:06:66:80:55:52:20:14:15:87:0e:20:01:01:0c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 61:03:55:ff:6f:06:66:80:55:52:20:14:15:87:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 21:03:55:ff:6f:06:66:80:55:52:20:14:15:87:02:0b:00:00:00:00:00:09:09:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT RGB & Fan Controller
├── Firmware version                     1.12.0
├── LED 1 accessory 1          AER RGB 2 120 mm
├── LED 2 accessory 1    HUE 2 Underglow 300 mm
└── LED 2 accessory 2    HUE 2 Underglow 300 mm

[DEBUG] liquidctl.cli: device: NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: writting report 0x01 with 64 bytes: 5c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x01 with 64 bytes: 5d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

jonasmalacofilho, 2021-03-07:
Thanks!

> I couldn't use your compiled binary yet because it doesn't have our latest code compiled

Oh, right.

> I am sorry, that issue was only because I was trying to find a way for the following screenshot to disappear.

Hm, apparently whatever you tried messed up the ability of the terminal emulator to handle unicode.

Anyway, glad to know you solved it.

> I do see this on initialize on Linux and Windows for only that new device, not sure why though

That's fine, it just means that it must be the only device you own which reports data during the initialization phase.

> Also, had to install this (don't run filter, only install the package): libusb-win32-devel-filter-1.2.6.0.exe in order for liquidctl to work.

Why?  Was this about missing a backend for PyUSB?

In that case, why weren't the official DLLs from the latest libusb release enough?

>> Pre-build liquidctl executables for Windows already include Python and libusb, but when installing from PyPI or the sources both of these will need to be manually set up.  The libusb DLLs can be found in [libusb/releases](https://github.com/libusb/libusb/releases) (part of the `libusb-<version>.7z` files) and the appropriate (e.g. MS64) `.dll` and `.lib` files should be extracted to the system or python installation directory (e.g. `C:\Windows\System32` or `C:\Python36`).
>> —*from our README*

sgtcoder, 2021-03-07:
Thank you!

```That's fine, it just means that it must be the only device you own which reports data during the initialization phase.```
I was not aware about that. Then technically that's a good feature! I ordered more NZXT RGB fans and another controller.

```Why? Was this about missing a backend for PyUSB?```
It mentioned about missing backend and then later about permission to write to the device.

```In that case, why weren't the official DLLs from the latest libusb release enough?```
Sorry, I didn't see this part and skimped over it. I ended up uninstalling that filter and doing a cold boot and everything still works, so that is strange. It might have been because I changed the method of how I installed Python (and under custom install you have to chose to add environment path). If this issue comes up again, I have saved the libusb package that I will try in future. This is good and sorry for "wasting" your time then as it appears.

Thank you again for everything. I was going to switch to MSI components in the future, but they have no support for Linux, and OpenRGB bricks the devices to they disabled it.

Because of this project, the customization, fine tuning, being able to run a fully integrated on both Windows and Linux seamlessly without having to cold boot into Windows first, essentially getting rid of the NZXT Cam bloat on boot in Windows, I am sticking with NZXT and very excited about this project and will try to contribute when possible.

jonasmalacofilho, 2021-03-08:
Oh, ok, don't worry about it.

And thanks!

sgtcoder, 2021-03-12:
I don't suppose you know how to setup a systemd service unit when the computer wakes in Ubuntu? After waking my system the lights all turn back to white and reset where it needs to be initialized and reconfigured again. The oneshot method only works at initial boot, but not when waking up computer.

jonasmalacofilho, 2021-03-12:
Perhaps you can make systemd (re)start the configuration service when the device is found by udev, by making the by making the device depend on the service, we an udev rule:

```
ACTION=="add", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", ATTRS{serial}=="<serial number>" ENV{SYSTEMD_WANTS}="liquidcfg.service"
```

(this is something I found out when looking into [Making systemd units wait for devices](https://github.com/liquidctl/liquidctl/blob/master/docs/linux/making-systemd-units-wait-for-devices.md))

jonasmalacofilho, 2021-03-12:
Alternatively, it should be possible to make the service run at every boot *and* resume, which could be more convenient if you have multiple supported devices.  But of the of my head I'm not sure how...

sgtcoder, 2021-03-12:
Thank you. I did see the udev rules. I had some issues in the past where the script would load too soon, so I tried running it after. It was all good, but I just added a 4th device, and now I am having resume issues. I will look at the udev rules, but ideally I would like it to have it load on resume and start. Initialize is also starting to get slow too.

jonasmalacofilho, 2021-03-12:
> I will look at the udev rules, but ideally I would like it to have it load on resume and start.

Please let me know what you find and, if possible, send me a PR for a new document in `docs/linux/`.

> Initialize is also starting to get slow too.

Since the devices are actually resetting to default settings after resuming, the first trick I could mention (don't initialize all devices all the time) is a bad idea.

The second trick is to initialize the devices in parallel.  This isn't a feature of the liquidctl CLI, but you can can execute multiple `liquidctl initialize <filter options> &` instances in the background, instead of a single `liquidctl intiialize all`.

sgtcoder, 2021-03-12:
I will get back to you on the document since there are some nice things we can add.

Going off of this StackOverflow, I got it to load on resume (using systemd): https://askubuntu.com/questions/92218/how-to-execute-a-command-after-resume-from-suspend

```
[Unit]
Description=AIO resume service
Before=sleep.target
StopWhenUnneeded=yes

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStop=/usr/bin/config_liquidctl

[Install]
WantedBy=sleep.target
```

I am unsure why settings are being wiped out now during sleep, that's kind of bothersome. I will dig deeper into this, at least it's working now. I upgraded to the latest NZXT Internal USB hub so I have 4 more USB ports, so might be something with that.

I will also look into udev rules or another option to hook in after the 4 devices are "found" (for the simplest way possible). Also, when is initialize needed? I thought every boot? Every boot I think is fine, but on resume we certainly wouldn't want to wait for the service to load.

jonasmalacofilho, 2021-03-12:
This isn't going to help much but, technically, you need to initialize a device any time it may be in an uninitialized state, or in a state that differs from critical assumptions made by liquidctl.

In practice this depends on a lot of factors, the most significant simply being how the device works (which is not in our control).  But also the OS, the motherboard, the power saving settings, and (rarely) hacks that liquidctl may need to use for a particular device...

Initializing after every boot is a good rule of thumb, but the actual necessity can vary from after every suspend to RAM to only after the PSU has been turned off for a few seconds, and anything in between.  And, sometimes, it's not necessary at all.

Without getting into specifics about each device, protocol and driver implementation, another rule of thumb is that if the device lost your settings, then it's probably a good idea to reinitialize it.

---

Now actually getting into the specifics:

- NZXT Kraken X (X42, X52, X62 or X72): does *not* need to be initialized, but looses settings when the +5VSB (stand-by) rail goes down
- NZXT Smart Device (V1): must be initialized anytime the settings are lost, as at that point the device no longer knows the fan types and number of connected LEDs; and the settings are also lost when +5VSB goes down
- NZXT RGB & Fan Controller: AFAIK same as the SD V1, but I don't have it so I can't confirm

The +5VSB rail *may* go down when the system is suspended to disk or shutdown; whether it will actually do so usually depends on the power saving settings on your motherboard.  And, obviously, the rail will go down if the PSU is turned off (for long enough to any residual charge to be dissipated).

sgtcoder, 2021-03-12:
I will get back to you on your comment. I found the reason why the devices are loosing the settings. The new NZXT USB hub uses SATA power vs molex. The SATA rails turn off in sleep and power off whereas the molex power stays on. I am unsure what to do now since the molex version has been discontinued and also only has 3 USB vs 4.

jonasmalacofilho, 2021-03-12:
Hm, I forgot about the hub, and it too can cut power to the devices connected to it.

By the way, the embedded controllers in the Kraken and SD V1 are powered by +5VSB *from the USB connection.*  And the SD V2 probably works the same way.

sgtcoder, 2021-03-12:
I realized that the VSB is through USB. The NZXT USB 3 is powered through SATA which I preferred, but the SATA shuts power off during sleep and off. The original NZXT USB Hub is powered through a molex and the power stays on always (unless unplugged or no power).

I ended up removing one of the V2 controllers for now. Also, the V2 controllers take about 5 seconds to initialize. The V1 and Kraken are instant (less than 1 sec).

On a side note, is there a way to determine if the device initialized or not prior to running the initialize (at least I only have to run it on boot to be safe)? I also was aware of the & for the command, but that made no difference since the V2 takes about 5 seconds while the rest are less than 1 second.

jonasmalacofilho, 2021-03-12:
> I ended up removing one of the V2 controllers for now. 

You could merge the USB cables from both V2 controllers into a single connector; each USB 2.0 header has two ports, and each controller only uses one.  All you need to do is extract the pins from one connector (without damaging them) and insert them into the other one.

> is there a way to determine if the device initialized or not

Not as a liquidctl feature.  We can detect and issue warnings for some devices (or maybe just one), but that's the exception: most devices will simply do wrong things, or even lock up, if you try anything but initializing it (while in an uninitialized state).

What you might be able to do with the SD V1, depending on your hardware setup, is to check the output of `status` and see whether the device has already detected your specific fans and LED accessories.  But this will only work if you setup is different than the controller's default assumptions (which I don't remember of the top of my head).  And it's obviously a hack without any stability guarantee.

But your main issue is with the much slower SD V2/RGB & Fan Controller.  Unfortunately, we haven't tested these corner cases or figured out how to detect the uninitialized state.  (And I don't have a device to test these things here).

---

P.S. I refreshing my hwmon kernel drivers for the Kraken X.2 and Smart Device V1.  Let me know if you can test them, particularly during suspend/resume, since suspending is broken on my machine due to some other issue (probably with the motherboard).

sgtcoder, 2021-03-12:
```
You could merge the USB cables from both V2 controllers into a single connector
```
Thank you, that is a good point. I only have 2 USB headers and I kind of need like 5 of them total. Either way, I went back to the old one. They also make splitters too that don't require external power. Not sure what the power limit is though.

If you lived in the US, I would just ship you my secondary V2 device.

I can definitely help test certain things out. What am I testing specifically? Just so you know I don't lose power on the molex rail so power will be kept to the USB devices now.

sgtcoder, 2021-03-14:
Also, I am still getting ```ERROR: Unexpected OS error with NZXT RGB & Fan Controller: OSError: read error``` error in Windows. I did copy the libusb-1.0.dll file to the C:\Windows\System32 folder and it still gives that error. But, when I install that filter, it works.

dferrandizmont, 2021-03-17:
Sorry for posting, but i'm having same problem with my NZXT RGB & FAN 

```
uname -r
5.11.6-051106-generic
```

```
cat /etc/os-release
NAME="Ubuntu"
VERSION="20.10 (Groovy Gorilla)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.10"
VERSION_ID="20.10"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
```

```
lsusb
Bus 001 Device 005: ID 1e71:200e NZXT NZXT USB Device
Bus 001 Device 004: ID 1e71:1715 NZXT Kraken M22
```

```
sudo lsusb -v -d1e71:200e

Bus 001 Device 005: ID 1e71:200e NZXT NZXT USB Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x200e 
  bcdDevice            1.0c
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 8D821597525
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
[... 92 lines cut ...]
[DEBUG] liquidctl.driver.usb: found USB device 046d:c539
[DEBUG] liquidctl.driver.usb: found USB device 2109:2817
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 0b05:189d
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1e71:200e
[DEBUG] liquidctl.driver.usb: found USB device 1e71:1715
[DEBUG] liquidctl.driver.usb: found USB device 0d8c:0014
[DEBUG] liquidctl.driver.usb: found USB device 0a12:0001
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
Device ID 0: NZXT Kraken M22
├── Vendor ID: 0x1e71
├── Product ID: 0x1715
├── Release number: 0x0200
├── Serial number: 8D871762534
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: Kraken2
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver

```

jonasmalacofilho, 2021-03-17:
@dferrandizmont the product ID for this new variant of the RGB & Fan Controller has already been added to the driver: f19913c75af9.

The change will be in the next release and, in the meantime, you can also use the code in the main branch.

dferrandizmont, 2021-03-17:
@jonasmalacofilho thanks for the quick reply, working like a charm with the code in the main branch. 

Thank you so much!

jonasmalacofilho, 2021-03-21:
@sgtcoder 

> I can definitely help test certain things out. What am I testing specifically? Just so you know I don't lose power on the molex rail so power will be kept to the USB devices now.

Essentially, I want to ask you to test the `nzxt-smartdevice` driver in [liquidtux](https://github.com/liquidctl/liquidtux), for the SDV1.

It's a Linux hardware monitoring driver, so fan monitoring – speeds, current and voltage – and control are made available in `/sys/class/hwmon/hwmon?/*`.¹  As that's a standard interface, you can use tools like `sensors`, `fancontrol`, Psensor, Freon, etc... basically, anything that works with lm-sensors; and, of course, you can also use plain `cat`/`echo` too.

Anyway, I'm especially interested to know whether it re-initializes² your device after resuming *from a suspended state that causes it to reset,* as well as the `dmesg | grep smartdevice` output of your entire test session.  Please also record if the device reset (if you can figure out based on the LED settings) and if you can hear the initialization routine (where all fans spin up to 100%, then go back down to their set speeds).

Since you're on Xubuntu, it'll probably be easier to use the manual build route.  But there's a `dmks.conf` in there too, if you want to get fancy and use it daily without having to manually recompile after every kernel update.

Oh, and at this point both drivers have been extensively tested (there's one for Kraken X.2 coolers too), so apart from specific issues during suspend/resume, I expect them to work fairly well.  I'm using both drivers daily.

If you can do this, it will be a great help.  I just can get a reliable way to test suspend/resume on my machine; QEMU locks up even with a stock kernel, and my physical machine either resumes immediately (some BIOS issue) or the proprietary Nvidia driver crashes.

_¹ LED control is not supported (or planned)._
_² In this context "initialization" means making it work again, not restoring your fan/lighting settings._

sgtcoder, 2021-03-21:
@jonasmalacofilho

I ended up ditching the V1 controller and upgrading to V2 for everything. I still have the V1 smart device that I can always link up to test.

As far as the resume state and having to initialize, was only because the new NZXT Internal USB 3 Hub (that I could only stand to use for 2 days) isn't wired to receive power from the VSB on the motherboard. Due to this serious flaw, I also removed it (I can test only if I put it back in my machine). Otherwise, only when I unplug my computer or flip the power switch to OFF on the PSU is the only time my devices need to be re-initialized. To call it safe, I only run this on boot for Linux/Windows.

Either way, I am welcome to help. And curious about that other project (I saw) and about the dkms.conf file :-)

Also, here is my complete build of what I am running: https://pcpartpicker.com/b/4VQqqs

jonasmalacofilho, 2021-03-22:
@sgtcoder 

Do you want to join our [Discord server](https://discord.gg/f5HKCAadsr), or chat via IRC?

> Either way, I am welcome to help.

Thanks!

> the new NZXT Internal USB 3 Hub [...] (I can test only if I put it back in my machine)

What you could do (and it's probably more practical) to test that part is, while suspended, connect and disconnect the Smart Device to the USB port.  But a simple suspend-resume cycle, even if not powering down the device, would already help a bit.

> And curious about that other project (I saw) and about the dkms.conf file :-)

Regarding using `dkms.conf`, maybe [this](https://help.ubuntu.com/community/DKMS) can help (but it may help to adjust the configuration; ours may be a bit too specific for building a *package* for ArchLinux, not for manual/direct installation).

> Also, here is my complete build of what I am running: https://pcpartpicker.com/b/4VQqqs

Cool.  We can also add support for the 1080, it should use the same protocol as the 1080 FTW (ACX), which is already supported.

commit f19913c7 (2021-03-05): smart_device: enable new variable of RGB & Fan Controller

Closes: #304