liquidctl/liquidctl #305: Smart Device V1 does not support temperature-related fan speed

GER13117, 2021-03-08:
Hello guys,
So I'm not sure if this is a limitation of  the V1 device, a bug or a mistake I made by myself, but when I call `liquidctl --match smart set fan2 speed 35 0  40 20  45 40  65 40  70 70  85 80  90 100` I get the error-message `ERROR: Error: operation not supported by NZXT Smart Device (V1)`. 
I know the Error-Message says that the V1 Device does not support this operation, but I'm not sure because the documentation I think shows, that this should be possible. 

This happens only to me on Linux (Garuda) on my Hackintosh (same PC) the Smart Device doesn't even gets detected.

jonasmalacofilho, 2021-03-08:
This is a limitation of all Smart Devices: they don't have any temperature sensors.

You would need something like [`extra/yoda`](https://github.com/liquidctl/liquidctl/blob/master/extra/yoda), or your own version of that (either using the APIs or with shell scripting), to achieve temperature-controlled fan speeds with these devices.

---

>  because the documentation I think shows, that this should be possible.

Can you point me to where this is suggested, as that's probably a bug in the documentation.

> on my Hackintosh (same PC) the Smart Device doesn't even gets detected.

As a rule of thumb on Hackintoshes, this usually means your USB setup isn't complete.

But if you have already checked that the OS can enumerate the device, then please post the output of `liquidctl list --debug`.

GER13117, 2021-03-08:
Thanks for the quick answer! I read the part where I thought I read it again, I just interpreted too much... So it's everything fine. On the Hackintosh I will check again.

Greetings from Germany

jonasmalacofilho, 2021-03-14:
I'm going to close the issue.

Feel free to open a bug report if, after checking/adjusting the USB configuration, Mac OS starts to see the device but liquidctl still can't.

closed 2021-03-14 by jonasmalacofilho