liquidctl/liquidctl #306: Liquidctl not detecting devices on Ubuntu 20.04 clean install

ChrisFhey, 2021-03-13:
Installed liquidctl on a clean Ubuntu 20.04, but no devices are detected. I am using a Corsair Commander Mini with 5 fans attached.

Running `liquidctl initialize` or `liquidctl status` gives the following error:

```
Error: no devices matches available drivers and selection criteria
```

Additional information about the system:

- Affected device: Corsair Commander MINI
- Operating system: Ubuntu 20.04 LTS
- Liquidctl version: Ubuntu `liquidctl` package, version `1.5.1`

MarshallAsch, 2021-03-13:
The corsair commander mini is not a device that liquidctl currently supports, although we would be happy to add it to the project.

@ChrisFhey would you be able to open a new issue using the new device template to give us some more information about the device?

ChrisFhey, 2021-03-13:
Sure thing. My apologies, I assumed the Commander devices were already part of Liquidctl.
I'll make a new issue, and close this one then.

closed 2021-03-13 by ChrisFhey

MarshallAsch, 2021-03-13:
Im hoping that it will use the same communication protocol, that should make it fairly trivial to implement.

sanmai, 2021-05-10:
Superseded by #307