liquidctl/liquidctl #307: <Corsair> <Commander Mini>

ChrisFhey, 2021-03-13:
### Description

Controller for Corsair LINK.
It has four Corsair Link Digital ports, six fan control connectors, four temperature inputs, and a port for connecting CORSAIR Link LED lighting strips.

It is mainly used in combination with corsair iCue to control fans and RGB. (Although I am currently using it for fan speed control exclusively)

### The device appears to support the following features:

 - [x] monitoring of temperatures or other environment sensors
 - [x] monitoring of fan or pump speeds or duty cycles
 - [ ] monitoring of voltages, currents or power
 - [x] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [x] configurable lighting of embedded LEDs
 - [x] configurable lighting of accessories like RGB fans or LED strips

### The device is connected to the host system using:

- [ ] USB
- [ ] PCI-E
- [x] onboard (the motherboard) _(through the USB2.0 header)_
- [ ] other (please elaborate)
- [ ] unknown

### The device communicates with the host system using:

- [ ] HID
- [x] USB _(I think)_
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [ ] unknown

### lsusb output

[lsusb_corsair_link.txt](https://github.com/liquidctl/liquidctl/files/6135639/lsusb_corsair_link.txt)

---

Finally, I can help with:¹

- [x] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [ ] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

MarshallAsch, 2021-03-13:
Excellent thank you @ChrisFhey  :) 

Would you be able to capture some traffic between icue and the device? We have some [usb capturing docs] docs that will walk you through capturing the traffic from a windows vm on linux. 

Specifically:
- capture the icue starting up (you can do this by having wireshark capturing the traffic before the vm starts up)
- setting the fan speed to a fixed %
- setting the fan speed to a curve
- led setting if you have LED's, you mentioned that your not using them 
- can you confirm that there is a setting to tell the device what type of fan is connected (DC, PWM)

 [usb capturing docs]: https://github.com/liquidctl/liquidctl/blob/master/docs/developer/creating-vm-for-capture.md

ChrisFhey, 2021-03-13:
Absolutely. I'll look into it as soon as I can.

However, I'm not sure if I can provide data for setting the fan speeds as I have no way to control the Commander Mini on Ubuntu right now, and as such my fans are stuck on max RPM at the moment.

MarshallAsch, 2021-03-13:
yup using a windows VM would be the way to go to get that

MarshallAsch, 2021-03-13:
You can also join [this](https://discord.gg/CFwbjqPx) discord server if you need some help collecting the trace

ChrisFhey, 2021-03-13:
I've just finished setting up the VM for testing, and I've discovered that the Commander Mini I have is not being detected by iCue at all. (Which is strange, as I think I used it with iCue when I was running windows on that PC.)

However, I can connect to it using Corsair Link. Is that potentially something you can use as well, or am I out of luck until I figure out if iCue can support the Mini after all?

MarshallAsch, 2021-03-13:
So I actually have not used corsair link at all I figured that link and icue were slightly different versions of the same thing, but  google says otherwise. 

captures from Corsair Link should work. I didn't realize that the device is that much older then the other lighting node / commander products. 

Once you get some captures I can take a look at them and see what I can figure out. (If are able to limit the capture to only the one device that would help. (you can do that by selecting the specific usbmon interface rather then selecting 0)

MarshallAsch, 2021-03-13:
>  I used it with iCue when I was running windows on that PC

this https://forum.corsair.com/forums/showthread.php?t=176371 says you didn't unfortunately, which also means that it is probably a different protocol :(.

ChrisFhey, 2021-03-13:
Yes, I saw something similar, sadly. But I am confused about it though, as I really thought I used it with iCue on that PC still.

Either way, would you still be interested in me trying to capture data for the Link?

MarshallAsch, 2021-03-14:
yup I am still expecting it to be usb hid traffic. As long as we are able to see what the traffic looks like and be able to confirm what was actually done then we should be able to implement a driver for it. 

While doing the capture can you also try to keep track of the order that you did things in, what the fan speeds were set to, what link showed them as, what the temperature probs  temps are. and things like that. 

Jonas is better than me at pulling that information out of the traces without knowing what the real values are, however I'm not quite that good yet and will happily take any extra information you are able to provide.

ChrisFhey, 2021-03-14:
@MarshallAsch I've done a couple of captures. I hope they turned out all right.
Should I just upload the results here, or where would you like them?

MarshallAsch, 2021-03-14:
You can toss them here or in a PR to the device data repository [liquidctl/collected-device-data](https://github.com/liquidctl/collected-device-data)

ChrisFhey, 2021-03-14:
I'll opt for doing it here then, if you don't mind.

[commander_mini_capture.zip](https://github.com/liquidctl/liquidctl/files/6136820/commander_mini_capture.zip)

It includes the capture file, and a txt file detailing the steps I took in Corsair Link.
Let me know if you need more info.

MarshallAsch, 2021-03-14:
I have taken a quick look at that and it uses a completely different protocol. 
Are you able to confirm how many fans you have connected, what types they are, what ports they are plugged into, and if the temperature probes are connected

ChrisFhey, 2021-03-14:
I've 5 fans connected at the moment. 
They're all either Corsair AF or SP series fans, and their spec sheet on the Corsair website states they're not PWM fans. And they have no RGB.
That's all I can tell you about them I'm afraid, as I bought these quite a while ago. 

The fans are connected to ports 1, 2, 4, 5, and 6 on the Commander Mini.
Other than the fans, nothing is connected to the Mini at the moment.
I don't have any additional hardware available to test, as I only use the Mini for fan control in my Ubuntu pc.

jonasmalacofilho, 2021-03-18:
The USB IDs match those of the Cooling Node, an older device.  And that also more closely matches your description of the device, with a single CLINK port (the Commander Mini has four).

I also don't think that's how the CLINK port works; as far as I know, it would need to be connected to a Lighting Node (or a device more capable than it) for it to *indirectly* allow control over strip/fan LEDs.  But I don't really know how this works...

Anyway, please clarify if it is indeed a Cooling Node.  Because if that's the case, then there's a lot of code to be shared with #142, at least [according to OpenCorsairLink](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/device.c#L66-L79).

ChrisFhey, 2021-03-19:
Hello Jonas, sorry for the delay in my reply.

I'm not sure what a Corsair Cooling Node is, but that is indeed what Ubuntu declares the device as if I perform an lsusb:

`Bus 001 Device 002: ID 1b1c:0c04 Corsair Link Cooling Node`

However, it is as far as I know still a Corsair Commander MINI, and is reported as such by Corsair Link itself as well:

[image]

jonasmalacofilho, 2021-03-20:
That's interesting, thanks for clearing that out.

It could be just another episode of Corsair confusing their own names (like with the Pro coolers that have "Platinum" descriptors), but it could also mean that both Cooling Node and (current) Commander Mini share the same protocol.

@MarshallAsch do you see any similarities between these captures and the Commander Core?  (I assume when you said it was a different protocol, you were comparing against the Commander Pro)

ChrisFhey, 2021-03-20:
No problem. Let me know if there's anything else I can help you guys with. :)

MarshallAsch, 2021-03-20:
I can check my notes to be sure about the specifics but the protocol has no overlap with any of the other devices I have looked at (the commander pro family of devices, h100i platinum, commander core). It seems to be 64 byte messages and only the first 6 or so have data, I have identified a couple of the messages but haven't quite been able to determine the format of them.

jonasmalacofilho, 2021-03-20:
It appears to somewhat match the Cooling Node implementation in OpenCorsairLink:

```
Request:
00h-01h    Message length
01h-02h    Sequence number (this is annoying)
02h-03h    Opcode
03h-len    <data>

Response:
00h-01h    Sequence number echo
01h-02h    Opcode echo
02h-...    <data>
```

- [CoolIT opcodes (OCL)](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/include/protocol/coolit.h#L64)
- [Porting drivers from OpenCorsairLink (to help navigate OCL)](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/porting-drivers-from-opencorsairlink.md)

sanmai, 2021-05-09:
FWIW OpenCorsairLink can detect some features of the device.
```
$ opencorsairlink --device 0
Dev=0, CorsairLink Device Found: Commander Mini!

Vendor: Corsair
Product: Commander Mini
Firmware: 0.0.61
Temperature 0:  7.50 C
Temperature 1: 34.62 C
Temperature 2: 178.24 C
Temperature 3: 180.32 C
Temperature 4: 25.68 C
```
I have two thermal sensors connected (1 and 4 in the list), and their readings match other sensors in the system.

sanmai, 2021-05-09:
The driver seems to be CoolIT, similar to what other devices like H100i use:

https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/device.c#L133-L145

sanmai, 2021-05-09:
Here's a [procedure to read firmware ID](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/protocol/coolit/core.c#L93). It is almost right because the program shows 0.0.61 where it is 1.1.6 actually.

sanmai, 2021-05-09:
More to the issue: Corsair H110i GT reportedly has the same USB IDs: 0x1B1C/0x0C04. This certainly does not make things easier.

jonasmalacofilho, 2021-05-09:
Thanks for confirming that the CoolIT driver from OpenCorsairLink works.  This means that porting it is probably the easiest way forward.

Now we need a volunteer for this, preferably someone with access to one of these devices.  Here are some relevant links to help them get started:

- Previous art: PR #147 ("Added experimental support for H110i GT")
- [Porting drivers from OpenCorsairLink](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/porting-drivers-from-opencorsairlink.md)
- [Style-guide and other documentation for developers](https://github.com/liquidctl/liquidctl/tree/main/docs/developer)
- [If necessary, ask for help on Discord server](https://discord.gg/GyCBjQhqCd)

@sanmai any chance you do it?

sanmai, 2021-05-09:
I tried the patch from #147, and it seems like things went too far since then. It won't work from get go (I get `key must be valid Python identifier`).  Let me see if I can help it.

sanmai, 2021-05-09:
I can confirm that driver introduced in #147 works for the device:
```
Corsair H110i GT (experimental)
└── Firmware version    1.1.6  

Corsair H110i GT (experimental)
├── Liquid temperature    34.8  °C
├── Fan 1 speed           1308  rpm
├── Fan 2 speed           1905  rpm
└── Pump speed               0  rpm
```

jonasmalacofilho, 2021-05-10:
Good to know.

Although it still needs a bit of work because it can be merged.  From the comments on the PR:

> That said, the code appears to be functional, just a bit more complex than necessary. So it probably just needs a bit of refactoring (besides some tests and documentation) before being ready to be merged. At the very least it's a good reference.

jonasmalacofilho, 2022-04-05:
It would be great if someone could clean the old PR #147 and re-submit it.  Otherwise I'll probably close this issue and #142 in a few months, as most of these devices are quite old by now, and likely starting to fail.

jonasmalacofilho, 2022-07-03:
Triage: as previously anticipated, closing since no one has picked up the old PR.

closed 2022-07-03 by jonasmalacofilho