liquidctl/liquidctl #308: Corsair Hydro H80i v2: Able to set, but not read fan speed

MennoLabs, 2021-03-14:
$ liquidctl  status
Corsair Hydro H80i v2
├── Liquid temperature       30.8  °C
├── Fan speed                   0  rpm
├── Pump speed               1350  rpm
└── Firmware version      2.8.0.0  

$ liquidctl  set fan speed 100
<sound like airplane taking off>

$ liquidctl  set fan speed 20
<quiet again>

- Affected device: Corsair Hydro H80i v2
- Operating system: Linux 5.8.0-44 (Kubuntu 20.10)
- Liquidctl version: v1.5.0

jonasmalacofilho, 2021-03-14:
I'm pretty sure this device uses a fan splitter.  If that's the case, only one of the receptacles, usually the first (=the closest to the cooler), will have 4 pins and will be able to read the fan speed.  Is your fan connected to it?

MennoLabs, 2021-03-14:
I have just opened the case. It's been built to spec for me and all the cables are very tight, but I was able to locate a splitter. One had 3 pins, the other 4 pins. Swapped them, but still no fan speed.
But I also tried opencorsairlink, which also does not give a fan speed.
So this might be hardware related, either defective or not connected right.

closed 2021-03-14 by MennoLabs

jonasmalacofilho, 2021-03-14:
Just to double check things no our side, can you show me the complete output of

```
$ liquidctl status --debug
```

MennoLabs, 2021-03-14:
$ liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.0
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature,
VengeanceRg
b, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro,
CorsairHidPsu,
HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2,
SmartDevice, S
martDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 145f:0176
[DEBUG] liquidctl.driver.usb: found HID device 145f:0176
[DEBUG] liquidctl.driver.usb: found HID device 046d:c03d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc,
Moder
n690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 2109:0810
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 2109:3431
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Hydro H80i v2
[DEBUG] liquidctl.driver.usb: found USB device 046d:0819
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 046d:c03d
[DEBUG] liquidctl.driver.usb: found USB device 145f:0176
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair Hydro H80i v2
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.asetek: set clear to send = True
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 2), {}
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.usb: explicitly claim interface
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 1), {}
[DEBUG] liquidctl.driver.usb: writting 4 bytes: 14:00:00:00
[DEBUG] liquidctl.driver.usb: read 32 bytes:
00:00:01:25:00:06:00:9d:05:46:1e:14
:00:00:0a:f8:00:22:af:05:00:ff:00:02:08:00:00:00:01:10:d7:0a
[DEBUG] liquidctl.driver.usb: explicitly release interface
Corsair Hydro H80i v2
├── Liquid temperature       31.0  °C
├── Fan speed                   0  rpm
├── Pump speed               1350  rpm
└── Firmware version      2.8.0.0

[DEBUG] liquidctl.driver.usb: explicitly release interface

jonasmalacofilho, 2021-03-14:
That firmware version seems new, and there's another value that could be a fan speed.  It's a long shot, but maybe the fan speed is now in another place.

Can you show me the data in the `[DEBUG] liquidctl.driver.usb: read 32 bytes: <...>` lines after setting the fan speed to 0%, 50% and 100%?

MennoLabs, 2021-03-14:
OK, will check tomorrow. Midnight in this time zone. Thanks a lot for your
willingness to help!

Kind regards,

Menno

MennoLabs, 2021-03-15:
$ liquidctl  set fan speed 0
$ liquidctl status --debug
[DEBUG] liquidctl.driver.usb: read 32 bytes:
00:00:01:92:00:06:00:9e:04:1a:14:14
:00:00:02:f8:00:22:af:05:00:ff:00:02:08:00:00:00:01:10:c7:0a
$ liquidctl  set fan speed 50
$ liquidctl status --debug
[DEBUG] liquidctl.driver.usb: read 32 bytes:
00:00:01:8c:00:06:00:9e:03:fc:14:14
:00:00:08:f8:00:22:af:05:00:ff:00:02:08:00:00:00:01:10:c7:0a
$ liquidctl  set fan speed 100
[DEBUG] liquidctl.driver.usb: read 32 bytes:
00:00:01:87:00:06:00:9e:04:1a:15:14
:00:00:02:f8:00:22:af:05:00:ff:00:02:08:00:00:00:01:10:c7:0a

jonasmalacofilho, 2021-03-15:
Unfortunately the long shot didn't pay off.  The device really doesn't appear to be reporting the fan speed.

MennoLabs, 2021-03-15:
Thanks for looking into it.
Does "the device" mean every Corsair Hydro H80i v2 with this firmware, or
is it just my Corsair which is either defective or incorrectly installed?
I'll try switching back the fan splitter and see if that makes a
difference..

jonasmalacofilho, 2021-03-16:
I don't think this behavior is expected for the H80i v2 model: the device appears to support fan control, and it doesn't make sense to not also monitor its speed (even if the latter isn't necessary for the former).

So it's probably an issue with your device of setup.  By the way, make sure to specifically check the cooler's fan splitter; there could be other splitters on that system.

MennoLabs, 2021-03-16:
Hello Jonas,

It's all been installed very neat and tightly, black wires tied to a black
motherboard, so it's quite hard to make out what is what and where it goes.
I Will try to check it out some time.

Thank you very much for your helpfulness, really appreciated!

Kind regards,

Menno

MennoLabs, 2021-03-16:
Also, it might be BIOS-related:
https://forum.corsair.com/forums/showthread.php?t=168410

jonasmalacofilho, 2021-03-16:
> Also, it might be BIOS-related:

That's unlikely to be related to your issue.  Although the header powering the cooler should be set to DC/100% for other reasons, it should have no impact on the ability of the device to report the fan speed.

MennoLabs, 2021-03-16:
Well, turns out it was the BIOS-setting:
$ liquidctl status
Corsair Hydro H80i v2
├── Liquid temperature       30.4  °C
├── Fan speed                 840  rpm
├── Pump speed               1980  rpm
└── Firmware version      2.8.0.0

The strange thing however is that I can't set the pump slower now.
I used to use opencorsairlink to set it to mode 3, which made it run around
1380 rpm.
AFAIK liquidctl can not set modes for the pump, only speeds.
I can set the pump speed higher, but not lower.

$ liquidctl  set pump speed 100
$ liquidctl status
├── Pump speed               2940  rpm
$ liquidctl  set pump speed 20
$ liquidctl status
├── Pump speed               1980  rpm
$ liquidctl  set pump speed 10
$ liquidctl status
├── Pump speed               1980  rpm

Opencorsairlink also is not able anymore to set it lower.
Quite mysterious....

jonasmalacofilho, 2021-03-16:
That's quite odd.  But, assuming you're right:

- apparently your BIOS settings were undervolting the cooler, which would also explain lower than usual pump speeds;
- it's very curious that fan control worked, even though the voltage was low enough to cause the cooler to stop reporting the fan speed.

Anyway, glad to hear that you managed to get it to work.