liquidctl/liquidctl #309: liquidctl list fails with AttributeError on latest hidraw release

ehaupt, 2021-03-15:
With `liquidctl 1.5.0` a dependency to [hidraw](https://github.com/daym/hidraw) was introduced. The new dependency however was not compatibly with Python 3.x (Python 2.7 only). I've contacted the author (@daym) and he's kindly [released](https://github.com/daym/hidraw/tree/0.2.1) an updated version compatible with Python 3.x.

Trying `liquidctl 1.5.0` with `hidraw 0.2.1` fails to run with the following error on FreeBSD:

```
$ liquidctl list
AttributeError: module 'hidraw' has no attribute 'enumerate'
```

Plattform:
- FreeBSD 13.0-RC2

jonasmalacofilho, 2021-03-15:
liquidctl has never depended on that `hidraw` package.

What we actually use is the `hidapi` package (source code in [cython-hidapi](https://github.com/trezor/cython-hidapi)), which, depending on how and for which platform it's compiled, exposes one or two modules: `hid` and, optionally, `hidraw`.

When both are available, `hid` will use the libusb-based fallback implementation, so we prefer `hidraw`.  But we already handle that cleanly:

https://github.com/liquidctl/liquidctl/blob/234e380faf924dbae1d622a1aabfc346b7ce059e/liquidctl/driver/usb.py#L62-L65

_(And none of this changed in 1.5.0)._

Unfortunately, two conflicting `hidraw` modules for Python 3 exist now.  And I don't think Python has a good way to handle this.

I think I'll need to add a heuristic to detect when the `hidraw` is not what we expect, and fallback those users to (potentially libusb-based) `hid`...

daym, 2021-03-15:
I would gladly drop my conflicting hidraw from pypi, but I've lost access to that account a decade ago.

jonasmalacofilho, 2021-03-16:
I don't think it's possible (or necessary) to drop an existing package from PyPI. 

But not pushing the Python 3 version to PyPI (and maybe untagging the release on GitHub) should avoid it ending up in end-users machines.

So if you're fine with doing that, wow, thanks!

---

_(While `hidapi`'s surprising top level module names are probably at fault, they are already widely used, and they seem to cover your use case as well.  So I think you suggestion makes sense but, once again, thank you!)._

ehaupt, 2021-03-16:
Thank you @jonasmalacofilho for the clarification.