liquidctl/liquidctl #310: Support for Gigabyte RGB Fusion 2.0 8297 Controller (IT8297BX-GBX599-MASTER)

issamsaid, 2021-03-18:
It seems that IT8297BX-GBX599-MASTER is not supported.

I am using it with Corsair iCUE H100i ELite on Ubuntu 20.04 LTS

The device appears to support the following features:

 - [ ] monitoring of temperatures or other environment sensors
 - [ ] monitoring of fan or pump speeds or duty cycles
 - [ ] monitoring of voltages, currents or power
 - [ x] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [ ] configurable lighting of embedded LEDs
 - [ ] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [ ] USB
- [ ] PCI-E
- [ x] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ ] HID
- [ ] USB
- [ x] I²C or SMBus
- [ ] other (please elaborate)
- [ ] unknown

Please also include any useful additional information, such as USB vendor and product IDs, the output of `lsusb -v` (Linux) or `system_profiler SPUSBDataType` (Mac OS), links to external resources or already collected traffic data.

`Bus 007 Device 004: ID 1b1c:0c1c Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1c
  bcdDevice            1.00
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
    Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
   bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
 bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
       bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1

```
---

Finally, I can help with:¹

- [x ] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [ ] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2021-03-18:
The 8297 controller is a USB device; the chip itself is made by ITE.  Can you show the complete output from `lsusb` and `lsusb -v -d 048d:5702`?

Another thing that comes to mind is that it is a LED controller (hence RGB Fusion 2.0), but you selected "configurable fan or pump speeds" as the only available features.  Can you elaborate?

Finally, we already support a Gigabyte RGB Fusion 2.0 8297 Controller with the [`rgb_fusion2`](https://github.com/liquidctl/liquidctl/blob/master/docs/gigabyte-rgb-fusion2-guide.md) driver.  Can you show the complete output from `liquidctl list --debug`?

issamsaid, 2021-03-18:
@jonasmalacofilho  i am actually trying to find a way to control the fans as they are on all the time.

isaid@nvdxb-musk:~$ lsusb -v -d 048d:5702
isaid@nvdxb-musk:~$ 
the lsusb is empty an so is `liquidctl status`

isaid@nvdxb-musk:~$ sudo liquidctl initialize 
Gigabyte RGB Fusion 2.0 8297 Controller
├── Hardware name       IT8297BX-GBX599-MASTER  
└── Firmware version                   1.0.1.0  

isaid@nvdxb-musk:~$ sudo liquidctl status 
isaid@nvdxb-musk:~$ sudo liquidctl status --verbose
INFO: status reports not available from Gigabyte RGB Fusion 2.0 8297 Controller

jonasmalacofilho, 2021-03-18:
The fans are controlled by another chip, probably a ITE (but maybe a Nuvoton) Super I/O chip; doesn't `sensors-detect` (from lm-sensors) detect anything?

I'm assuming, of course, that your fans are connected to the motherboard.  And in that case you should also be able to set fan curves in the BIOS settings.

---

I'm going to close the issue as the 8297 RGB controller is already supported.

jonasmalacofilho, 2021-03-18:
> I am using it with Corsair iCUE H100i ELite on Ubuntu 20.04 LTS

If you're trying to control the H100i Elite Capellix (with the external controller that comes with it), you might want to follow/subscribe to #218.

closed 2021-03-18 by jonasmalacofilho