liquidctl/liquidctl #313: $ python -m liquidctl.cli <args>... ?

sgehrman, 2021-04-02:
what is <args>  I can't install.

jonasmalacofilho, 2021-04-02:
The snippet where you copied that line from is more focused on those wanting to contribute to the project, and assumes minimal familiarity with Python.

I agree that some users may arrive there because of the "for the latest changes" part, even though it isn't immediately clear for them what follows.  But the conciseness is also beneficial...

Anyway, to make it clearer, here's another way to read that section:

> You can directly execute the code,
```
$ python -m liquidctl.cli <args>...
```
> or install it from that local copy.
```
# pip install .
```

The `<args>...` are the arguments to `liquidctl`.  For example: `python -m liquidctl.cli list --verbose`.

Please see the [python(1)](https://man.archlinux.org/man/python.1.en) and [pip(1)](https://man.archlinux.org/man/pip.1) man pages for more information.

---

P.S. Too much noise is obviously not good, but please try to be more descriptive.  This issue tracker is not a telegraph, and using it as if it was one gives the appearance that you care less about my time than yours.  After all, you probably wanted a better answer than "just executes use pip install .".

sgehrman, 2021-04-03:
ok, got it.  I'm a coder, I'm hoping to contribute.  I assumed this python command was to build the code or something.  I was trying to 'build' it.

jonasmalacofilho, 2021-04-03:
Happy to hear that!

Yeah, no need to build it, and that command will be useful when you are working on the code base, and want execute liquidctl without (re)installing it.

The actual `liquidctl` executable in your path is only created by setuptools during the installation procedure (which solves some cross-platform issues).

Feel free to join our [developer's Discord server](https://discord.gg/GyCBjQhqCd).

closed 2021-04-03 by jonasmalacofilho

sgehrman, 2021-04-03:
Quick question. Is there an option to get json output from the tool?  Or do I have to parse the text?

I'm writing a gui app in Flutter desktop.  I don't know python, and don't want to mess with that, so I'm only going to use liquidctl.