liquidctl/liquidctl #315: hydro platinum 'all fan' mode is called 'fan' and not 'sync'

MarshallAsch, 2021-04-04:
Most (all?) of the device drivers in `liquidctl` use `sync` for the set fan speed commands that impact all the fans. However, all of the devices in the `hydro_platinum.py` driver use the label `fan` instead. This should be changed for consistency. 

To maintain backwards compatibility `fan` should be kept but with a depreciation notice.

jonasmalacofilho, 2021-04-05:
This is a bit tricky with AIOs:

1. Some devices don't have individual fan speed control, so `fan` is used _instead_ of `fan1`, `fan2`, ...
2. Most devices have some type of pump speed control, _generally_ using `set pump speed`; thus `sync`, if it were to be supported, would mean `fan* | pump`.

For these reasons, `sync` is only used on AIOs for lighting and, in that case, implies all channels, regardless of physical location.

IMO `fan` is the better name for a convenience channel to set the speed of all fans, on devices that, natively, use individual fan control.

jonasmalacofilho, 2021-04-05:
We discussed this a bit on Discord:

- we think `sync` is generally unexpected, and we would like to support `fan` where applicable.

I also propose we consider:

- adding `fans` as an alias to `fan`;
- adding better aliases to `sync` (e.g. `pump` for `ring+logo` in Kraken X.2 lighting);
- implementing "all" behavior for more devices (e.g. Kraken X.2 cooling `fan+pump`);
- documenting the naming logic in the style guide.

Note: "sync" has another problem: it implies synchronization, which needs hardware support.  From this perspective, I'm also not sure we're always using it correctly.

jonasmalacofilho, 2022-07-03:
Triage: closing in favor of #458, which although a duplicate of this issue, is now more complete and up to date.

closed 2022-07-03 by jonasmalacofilho