liquidctl/liquidctl #320: NZXT Fancontroller either unsupported or broken

tbaumann, 2021-04-05:
I have a fancontroller in my NZXT H510i case and it works great.
But I needed a few more fan controller ports so I bought a second one.

That one does not work. I wonder if I received an unsupported model (the revision is not the same) or if I got a broken device.
I don't have windows to confirm.

I unplugged the working controller as to not muddy the waters
```
tilli@zuse:~$ liquidctl list -v
Device ID 0: NZXT Kraken X (X53, X63 or X73)
├── Vendor ID: 0x1e71
├── Product ID: 0x2007
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw10
└── Driver: KrakenX3

```

The first one in lsusb -v is the Kraken (same product ID)

```
Bus 009 Device 006: ID 1e71:2007 NZXT NZXT USB Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2007 
  bcdDevice            2.00
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     412
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

Bus 009 Device 009: ID 1e71:200e NZXT NZXT USB Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x200e 
  bcdDevice            1.0c
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 8D874882565
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     463
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

Do I have a unsupported device or just a broken one?

[image]

Sorry if this is just a dumb user support question. This software worked so out of the box the the other devices that Ithought it would be easiest if I just add another one of those controllers.

BTW, I'm working on some fancontroller daemon. First version is just de-crudded the bash script in contrib folder but I think I should add it in python. I hope I will find the time to finish this...

tbaumann, 2021-04-05:
Just for completeness

This is how the other controller looks when I plug it in

```
tilli@zuse:~$ liquidctl list -v
Device ID 0: NZXT Smart Device V2
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw11
└── Driver: SmartDevice2
``
```
Bus 009 Device 010: ID 1e71:2006 NZXT Smart Device V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2006 Smart Device V2
  bcdDevice            2.00
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     463
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

tbaumann, 2021-04-05:
The driver is supposed to match this product ID
https://github.com/liquidctl/liquidctl/blob/8ef2e79fd5f385487ff4a1a8e4ee92f0670a6d7f/liquidctl/driver/smart_device.py#L332

tbaumann, 2021-04-05:
Oh no. Sorry to waste your time. 

My debian package version was just very outdated.

closed 2021-04-05 by tbaumann

jonasmalacofilho, 2021-04-05:
As you've already figured out, it's a new variant.

Already supported in the master branch, and soon in liquidctl 1.6.0.

> My debian package version was just very outdated.

Well, I don't know what version you have, but I'd also be very surprised if they had the new version 36 hours *before* we're scheduled to release it : )

jonasmalacofilho, 2021-04-05:
> BTW, I'm working on some fancontroller daemon. First version is just de-crudded the bash script in contrib folder but I think I should add it in python. I hope I will find the time to finish this...

Also take a look at `extra/yoda`... it has some rough edges, but could help you as a reference to our Python APIs.

tbaumann, 2021-04-05:
Ah that yoda script looks much more like what I was going for. I'll play with it.