liquidctl/liquidctl #321: Code linter and automatic formatter

MarshallAsch, 2021-04-06:
Sometimes humans write bad looking code, we do not want bad looking code. Luckily for us some people have already written code that will take our code and make it look nice and ✨shiny✨, or yell at us when it is bad. 

This will much improve the developers quality of life by reducing the number of PR comments that look like
- "fix your white space"
- "why do you have so many empty lines?????????"
- "name your things consistently, stop going back and forth between, cammelCase, snake_case, and BoGlEcAsE, ....."

Potential tools to use (I have not looked into any of these yet)
- PyType
- pylint
- flake8
- autopep8

jonasmalacofilho, 2021-04-07:
In the area of code formatting tools, I generally prefer the "format everything" approach, instead of the "just fix format 'issues'" ones.

I suggest something like:

 - [psf/black](https://github.com/psf/black)
 - [google/yapf](https://github.com/google/yapf)

As far as I know, black is the current favorite in the Python community... (not that this matters a lot).

---

The other type of formatters, which are based on linters, don't change code that they see as already acceptable.  But what they see as acceptable has little to do with looking good, or even being PEP 8 compliant, in all but the most trivial cases.

The "format everything" approach is more honest: yes, it may sacrifice some super elegant hand formatting, but it's consistent and saves *everyone's* time, not just the time of the more style-disinterested developers.

Also, code style is more than just formatting, so style-disinterested developers are probably still going to get some "hey, please do something sensible here" comments.

Finally, just for the record, I don't think neither of us has ever been as obnoxious as the examples you mentioned : )

MarshallAsch, 2022-02-16:
> Also, this would allows to stop adding more files that would later need to be converted.

I agree with that, something is better than nothing. 

I took a quick glance at black again, I looked at it more when this was originally discussed and I still think its good.

jonasmalacofilho, 2022-02-27:
We have adopted `black` and there's now a style check action running in CI.

However, I don't think we'll get around to converting old modules to the new style all at once: it's a significant amount of work (t not make them worse in the process), and also the type of thing that has to happen when there is no other work in progress.

Since we also don't want to make `git blame` useless, the remaining alternative is to gradually update modules _when, for different reasons, we're making large scale changes to them._

That said, if anyone volunteers to do the big one-time-conversion, accepting that they'll have to wait for the timing to be right, that would awesome!