liquidctl/liquidctl #323: Add mypy type annotations

jonasmalacofilho, 2021-04-07:
This will have some limitations, particularly on the generic public API, but it's probably still useful.

And we could try to add this in more places, like the public driver APIs, and our core functions.

https://mypy.readthedocs.io/

Requested-by: @Sporknife

Sporknife, 2021-04-19:
Working on annotations in my [fork](https://github.com/Sporknife/liquidctl/tree/mypy-type-annotations).

jonasmalacofilho, 2021-05-30:
@Sporknife,

I really appreciate you looking into this stuff.  But as of your last commit (59cb1b2a006cce40716c02b42fe4b126512b4e75), I don't think you are exactly on the right path.

First, I think that you are focusing too much on easy things like constants, but typing them (assuming you did it correctly) adds very little value: if all mypy can do for us in this codebase is type check a few constants, then the churn and noise is probably not worth it.

There are also some problems like `BaseDriver.vendor_id(self)`(and friends) being type annotated to return `None`.  I get that methods are complicated to type annotate in Python because of`*args` and `**kwargs` (and we [ab]use them quite a lot here), but the return types obviously (IMHO) have to be compatible between methods in a base class and its subclasses, for some reasonable definition of "compatible".

Additionally, I would have expected some typing bugs to have been found by now: it is very unlikely that we have got typing 100% right by hand, and if mypy is not revealing any bugs then, maybe, the annotations that you are adding are not rigorous enough.

My suggestion is: start by completely annotating `BaseDriver`, and only then, gradually (perhaps method by method), start having mypy check that the subclasses are indeed compatible with their parents.

jonasmalacofilho, 2021-08-06:
**Triage: closing without fixing**

Given the very untyped architecture our codebase currently uses, and the tests that have been performed so far, mypy doesn't appear to be a good fit (=result in a ratio between added value and clutter/clutter that's greater than one).  And, at this point, if we were to rewrite the entire codebase to be more friendly to static typing, we would not do it in Python.

@Sporknife, thanks for suggesting/pushing this, and also for doing the tests/experiments I just mentioned.

---

P.S. I personally prefer statically typed languages for almost all applications.  But I also still stand by my choice of Python (and fairly naive Python at that) at the time liquidctl  was created: it was important to allow as much people as possible to help with the drivers.

closed 2021-08-06 by jonasmalacofilho