liquidctl/liquidctl #327: Persistence & device matching issue

Str-Gen, 2021-04-12:
Hi,
I've just started using liquidctl to control my Corsair Hydro H100i Platinum SE.
I use version 1.6.0 obtained from the Arch User Repository.

The issue is basically this:
This works fine (I've created the udev rule to get access from an unprivileged user):

`liquidctl --vendor 1b1c --product 0c19 set fan speed 20 30 30 50 40 70 45 85 50 100`

But embedding the same command inside the /etc/systemd/system/liquidcfg.service file fails on boot

> [Unit]
> Description=AIO startup service
> 
> [Service]
> Type=oneshot
> ExecStart=/usr/bin/liquidctl --vendor 1b1c --product 0c19 set fan speed 20 30 30 50 40 70 45 85 50 100
> 
> [Install]
> WantedBy=default.target
> 

I have tried different matching options, but none have been successful.

from `systemctl --failed`
>   UNIT              LOAD   ACTIVE SUB    DESCRIPTION        
> ● liquidcfg.service loaded failed failed AIO startup service
> 
> LOAD   = Reflects whether the unit definition was properly loaded.
> ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
> SUB    = The low-level unit activation state, values depend on unit type.
> 1 loaded units listed.
> [dhoogla@rgbcore ~]$ systemctl status liquidcfg
> × liquidcfg.service - AIO startup service
>      Loaded: loaded (/etc/systemd/system/liquidcfg.service; enabled; vendor preset: disabled)
>      Active: failed (Result: exit-code) since Mon 2021-04-12 22:14:04 CEST; 6min ago
>     Process: 609 ExecStart=/usr/bin/liquidctl --vendor 1b1c --product 0c19 set fan speed 20 30 30 50 40 70 45 85 50 100 (code=exited, status=1/FAILURE)
>    Main PID: 609 (code=exited, status=1/FAILURE)
>         CPU: 69ms
> 
> Apr 12 22:14:03 rgbcore systemd[1]: Starting AIO startup service...
> Apr 12 22:14:04 rgbcore liquidctl[609]: Error: no devices matches available drivers and selection criteria
> Apr 12 22:14:04 rgbcore systemd[1]: liquidcfg.service: Main process exited, code=exited, status=1/FAILURE
> Apr 12 22:14:04 rgbcore systemd[1]: liquidcfg.service: Failed with result 'exit-code'.
> Apr 12 22:14:04 rgbcore systemd[1]: Failed to start AIO startup service.
> 

**One secondary question:** the docs don't mention liquidctl initialization in the systemd unit file. Is this step required? (Which in my case would be `liquidctl --vendor 1b1c --product 0c19 initialize --pump-mode balanced`

jonasmalacofilho, 2021-04-13:
It seems that the service is running before the device has been made available to liquidctl through hidraw.

Our README is lacking a bit in that regard, but there is some information on possible ways of dealing with this in `docs/linux`: [Making systemd units wait for devices](https://github.com/liquidctl/liquidctl/blob/master/docs/linux/making-systemd-units-wait-for-devices.md).

---

> the docs don't mention liquidctl initialization in the systemd unit file. Is this step required?

Yes, it is.  Nice catch!

Str-Gen, 2021-04-13:
Thanks for the response. It's fixed with the following config

**/etc/udev/rules.d/72-liquidctl-persistence.rules**
> \# Corsair Hydro H100i Platinum SE
> ACTION=="add", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c19", SYMLINK+="h100i", TAG+="systemd"

**/etc/systemd/system/liquidcfg.service**
> [Unit]
> Description=AIO startup service
> Requires=dev-h100i.device
> After=dev-h100i.device
> 
> [Service]
> Type=oneshot
> ExecStart=/usr/bin/liquidctl --vendor 1b1c --product 0c19 initialize --pump-mode balanced
> ExecStart=/usr/bin/liquidctl --vendor 1b1c --product 0c19 set fan speed 20 30 30 50 40 70 45 85 50 100
> 
> [Install]
> WantedBy=default.target
>

closed 2021-04-13 by Str-Gen