liquidctl/liquidctl #331: Error message for HUE 2 device while initialization

captain1701e, 2021-04-20:
Hi,

because of CAM having some issues lately 😂, I was looking into liquidctl to get my HUE 2 back in the game.

Unfortunately already the initialization step results in an error on my end. Here is what I did:
> G:\test\liquidctl-1.6.0-bin-windows-x86_64>liquidctl list
Device #0: NZXT HUE 2
Device #1: Gigabyte RGB Fusion 2.0 5702 Controller
>
>G:\test\liquidctl-1.6.0-bin-windows-x86_64>liquidctl initialize -m nzxt
ERROR: Unexpected OS error with NZXT HUE 2: OSError: Could not write to device

Do you have a suggestion where I made the mistake, because according to the supported devices list HUE 2 should work?

Thank you an advance.

Regards,
Martin

captain1701e, 2021-04-20:
thanks for your fast reply.
Unfortunately it doesn't matter if I ran the command prompt as "user" or "admin", every time the same result.

jonasmalacofilho, 2021-04-20:
Do you the CAM app or even just a service started by it running?  Those can block access from other applications to the devices.

captain1701e, 2021-04-21:
Yeah, my fault, service was still running. But even after stopping and disabling it, the same error. Also a complete Windows reboot doesn't change the picture.

jonasmalacofilho, 2021-04-21:
I suspect something else from CAM may still be running, or perhaps you have another software that, like CAM, takes exclusive control of the device.

That said, can you show me the output of

```
> liquidctl initialize -m nzxt --debug
```

captain1701e, 2021-04-21:
The only thing which goes into this direction is the Razor software for the keyboard. But I thought they don't support each other, then blocking would not be good manner, wouldn't it? (just the opinion from an old C++ developer 😉)

Sure, here we go:
[image]

captain1701e, 2021-04-21:
After closing all Razor apps and stopping there corresponding services, I still get the same error

jonasmalacofilho, 2021-04-21:
Thanks.

> then blocking would not be good manner, wouldn't it? 

It wouldn't, but CAM used to do it anyway, and so I have to assume other programs could do it too. : )

> After closing all Razor apps and stopping there corresponding services, I still get the same error

There's one little thing odd in that output, but I can't easily tell if it's just a quirk of the protocol this device uses or if it's indeed a bug introduced by any recent-ish change.

Could you maybe [capture some USB traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md) between CAM and the device?

Additionally, can you also try a few past releases?  Specifically, I would try [1.5.1](https://github.com/liquidctl/liquidctl/releases/tag/v1.5.1), [1.4.2](https://github.com/liquidctl/liquidctl/releases/tag/v1.4.2) and [1.3.3](https://github.com/liquidctl/liquidctl/releases/tag/v1.3.3).

captain1701e, 2021-04-21:
Fast replay, I have tried the past releases and still same error just in a different formatted output (as far as I got it):
[image]

For the traffic capture I will need some time, I'll report back as soon as I have any news.

jonasmalacofilho, 2021-04-21:
I failed to clarify that the output with `--debug` would have been more helpful, especially with the 1.3.3 version.  But that's ok.

---

It's unlikely that this is a new bug in liquidctl, as it would need to have gone unnoticed for way too many versions, and this device is part of a family (and any of them would have triggered the bug, if it existed).

Besides there still being some unexpected program claiming this device for whatever reason, another possibility is that the device itself is gone a bit haywire.

This is usually solved by resetting it, which you can achieve by completely powering it off.  Technically you only need power off the 5V _Stand-by_ rail, but since that depends on motherboard settings, my advice is to turn off the PSU for a few minutes, until you're sure any remaining change has been drained from the bulk capacitors (some motherboards have a LED to indicate this).

Also, you mentioned CAM issues... any chance the device is misbehaving with it too?

Oh, and I suppose it could also be a weird USB issue, although I would expect the OS to unbind the device if that happened.  Still, does Device Manager indicate any issues with it?

captain1701e, 2021-04-22:
Since the issue with CAM is not only related to my system and configuration as it can be [read here on 18 pages](https://linustechtips.com/topic/1223050-new-pc-strange-windows-freezes-probable-cause-nzxt-cam/), I assume it is not my hardware. Also my device works with CAM running as expected but CAM freezes the PC after some time.

In regard to resetting, I fully disconnect my PC each time I shut it down. This happens at least once a day.

Please find attached the requested traffic capture with wireshark. The filter should be `usb.src ~ "1.33" || usb.dst ~ "1.33"` if I didn't got something wrong. I did two captures one for the initialization command with lidquidctl and one for "normal" CAM use. Please let me know in case something additional is needed.
[usb_capture.zip](https://github.com/liquidctl/liquidctl/files/6359096/usb_capture.zip)

jonasmalacofilho, 2021-04-22:
Thanks for captures, and we can indeed rule out a hardware problem.

But I would need to see the part where CAM and/or its helpers/services initialize the device, not just the "normal" use.  In practice, you need to start the capture before CAM and its helpers/services.

Additionally, your capture already shows that the firmware version is 2.6.0, which is know to us and known to be compatible with at least some of the liquidctl versions you tested.

P.S. It's not necessary to capture the traffic from liquidctl in this instance, `--debug` already provides us enough information.

captain1701e, 2021-04-22:
Okay, good to know for the liquidctl part. :)

For the CAM trace I started it before I started the service and CAM, so there should be everything in it.

jonasmalacofilho, 2021-04-22:
> For the CAM trace I started it before I started the service and CAM, so there should be everything in it. But I can rerun it if you would like.

The capture is missing the two writes that initiialize the device, which leads me to believe that there is still some CAM part/helper/service that you're not seeing.  That would explain both the missing writes in the capture as well as the issue with liquidctl not being able to communicate with the device.

(as far as know, CAM 4 uses a secondary process to handle all device communication)

captain1701e, 2021-04-22:
Understood, I will recapture then and check all processes & services.

captain1701e, 2021-04-22:
Here we go. I have checked with Sysinternals ProcessExplorer that nothing CAM related was running and stopped the service once more.
Then I started CAM application which showed a UAC windows to start cam_helper. I excepted this, so I hope everything is in the log now. Device ID changed to 35 for this run, since I unplugged the HUE 2 device in the meantime.
[CAM_second.zip](https://github.com/liquidctl/liquidctl/files/6360145/CAM_second.zip)

One funny thing, the CAM service wasn't running after starting CAM application, so why is there one? But I think this is another topic. 😀

jonasmalacofilho, 2021-04-22:
> Then I started CAM application which showed a UAC windows to start cam_helper. I excepted this, so I hope everything is in the log now. Device ID changed to 35 for this run, since I unplugged the HUE 2 device in the meantime.

Oddly no.

To be honest, the first (and maybe second) write(s) in `initialize` are probably not necessary for the HUE 2.  However, the traffic capture from `liquidctl initialize` shows that it wasn't the device that refused those writes, but rather that the OS refused to even forward them to the device.  So it's unlikely that removing them will make any difference, although you could try commenting [these two lines](https://github.com/liquidctl/liquidctl/blob/ad758606753d44150ced13ad2e088568d5e67234/liquidctl/driver/smart_device.py#L419-L420).

This bug is super weird!  Everything points to it being a conflict with CAM (or some other program), and yet there doesn't seem to be anything from CAM still running.  Because I checked: the service part doesn't lock the devices, and the cam_helper instances are automatically terminated when CAM closes; it would very hard to miss the fact that CAM is still running.

(FYI, I can reproduce the error with my devices while CAM is open, but as soon as close it liquidctl regains the ability to access them, as expected).

> One funny thing, the CAM service wasn't running after starting CAM application, so why is there one? But I think this is another topic. grinning

I also noticed that... quite amusing indeed.

captain1701e, 2021-04-25:
Took some time to setup the whole dev environment (incl. Git and all the rest), but with your detailed setup guide it worked like a charm.

Once I commented the two mentioned lines, initialization ends without error and setting effects of the LEDs works as expected afterwards.

jonasmalacofilho, 2021-04-25:
Does bringing back the two lines also cause the issue to reapear?

I would like to confirm this because another HUE 2 with the same (reported) firmware version is known to have worked even with those lines.  (and because from the captured traffic from liquidctl that first write wasn't even reaching the device).

So can I ask you to do these tests (commenting them out again; uncommenting them) a few times, and let me know if the results are always the same (works; fails on that first write)?

Thanks!

captain1701e, 2021-04-26:
I can confirm, that initialization works once the lines are commented and doesn't work anymore once I uncommented them again.

One additional observation from today: Yesterday I shutdown Windows and fully disconnected my PC from power line (switchable multiple socket outlet). I switched it on today and ran only the following command on your release version 1.6.0 for Windows: `liquidctl.exe set led2 color marquee-5 2f6017 --direction backward -m nzxt`. Result is as expected a marquee effect on the LED strip.
Is that the expected behavior? I understood from the documentation that an initialization is always needed after full power disconnect (mechanical off).

jonasmalacofilho, 2021-04-26:
Thank you so much for doing these additional tests.  I'll commit the fix to our main branch in a few minutes.

> Is that the expected behavior? I understood from the documentation that an initialization is always needed after full power disconnect (mechanical off).

The behavior is actually device-dependent, and even then sometimes tricky to specify.  Also, failing to initialize some devices can have some significant consequences.

So we find it easier to recommend (and for users to remember) to always initialize the device, preferably even after just a soft off (since, depending on motherboard power saving settings, it could be equivalent to a mechanical off for most chips in a system).

captain1701e, 2021-04-26:
Understand and thank you very much for all the insight and support!

commit 0d16d026 (2021-04-26): smart_device: check if fan controller before initializing fan reporting

HUE 2 devices (with firmware 2.6.0) can sometimes have issues with the
writes necessary to initialize fan reporting.  This behavior has been
found to be reproducible (#331), but may also vary due to unknown
factors[2].

Still, since HUE 2 (and HUE 2 Ambient) devices do not feature fan
reporting, these writes are not necessary and removing them has no
apparent drawbacks.

Fixes: #331 ("Error message for HUE 2 device while initialization")
Tested-by: @captain1701e

[2]: https://github.com/liquidctl/liquidctl/issues/109#issu [... cut]
commit 992f175f (2021-05-01): smart_device: check if fan controller before initializing fan reporting

HUE 2 devices (with firmware 2.6.0) can sometimes have issues with the
writes necessary to initialize fan reporting.  This behavior has been
found to be reproducible (#331), but may also vary due to unknown
factors[2].

Still, since HUE 2 (and HUE 2 Ambient) devices do not feature fan
reporting, these writes are not necessary and removing them has no
apparent drawbacks.

Fixes: #331 ("Error message for HUE 2 device while initialization")
Tested-by: @captain1701e

[2]: https://github.com/liquidctl/liquidctl/issues/109#issu [... cut]
commit dad8aeb7 (2021-05-01): smart_device: check if fan controller before initializing fan reporting

HUE 2 devices (with firmware 2.6.0) can sometimes have issues with the
writes necessary to initialize fan reporting.  This behavior has been
found to be reproducible (#331), but may also vary due to unknown
factors[2].

Still, since HUE 2 (and HUE 2 Ambient) devices do not feature fan
reporting, these writes are not necessary and removing them has no
apparent drawbacks.

Fixes: #331 ("Error message for HUE 2 device while initialization")
Tested-by: @captain1701e

[2]: https://github.com/liquidctl/liquidctl/issues/109#issu [... cut]
closed 2021-04-26 by captain1701e