liquidctl/liquidctl #332: Corsair AX1500i

keckj, 2021-04-22:
Hello,

I would like to know how I can help to add support for the `Corsair AX1500i` power supply to your amazing tool.
The device appears to support the same features as the `Corsair HX1200i` power supply (corsair link USB connection).

 I can help with:
- [x] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [x] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

I think that the power supply corresponds to this device (obtained with `lsusb -v`):
```
Bus 003 Device 003: ID 1b1c:1c02 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c02 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

jonasmalacofilho, 2021-04-22:
Hi,

That's awesome, let's do that.

> I think that the power supply corresponds to this device (obtained with lsusb -v):

Actually, that doesn't appear to be the case at all.  The HXi and RMi modesl are USB HIDs, and this AX1500i uses a Vendor Specific Class.  But that's not in itself a major problem.

Can you [capture some USB traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md) between iCue (or, if necessary, CorsairLink) and the PSU?  Try to start the capture before opening iCue/CorsairLink, and also send me a screenshot of the device infos (voltages, currents, etc.) during the the capture.

Thanks

keckj, 2021-04-24:
I finally managed to capture some USB traffic by following your guide (QEMU/KVM Windows 10 virtual machine with X forwarding). The first capture should contain data when I open iCUE (but I think it that some iCUE services already start at boot). The second capture has been done while iCUE was already opened. I included screenshots of the iCUE interface with sensors and parameters.

Link: [https://cloud.keckj.fr/index.php/s/HAFDx49CAF4YUKo](https://cloud.keckj.fr/index.php/s/HAFDx49CAF4YUKo)

I also confirm that my `lsusb -v` output matches with obtained vendor/product id:
```
  idVendor           0x1b1c Corsair
  idProduct          0x1c02 
```

jonasmalacofilho, 2021-04-25:
At first I though the captured data was corrupted, since it looked very weird and there were also some malformed packets.  But, in the end, it seems that this PSU just uses a rather [unusual encoding](https://github.com/ka87/cpsumon/blob/fd639684d7f9fe7aef9c8662d5f91e3fc8745e52/libcpsumon/src/cpsumon.c#L80-L123) (for a USB protocol) with balanced codes.  The malformed packets are something else, but maybe not important, at least for now.

I'll decode the data and try to make sense of it.  In the meantime, can you start a capture before starting QEMU, open it and iCue, then close both before stopping the capture?  The USB-to-serial bridge used by the device may require some specific commands during initialization and shutdown, and I would like to see those with this capture.

keckj, 2021-04-25:
The two first capture were obtained by using Wireshark **inside the Windows VM**, so that may explain some errors or weird behaviors. I am going to install Wireshark on my linux machine and capture some USB traffic directly from host (including traffic from QEMU boot).

jonasmalacofilho, 2021-04-25:
> The two first capture were obtained by using Wireshark inside the Windows VM

Oh, I missed that.

> that may explain some errors or weird behaviors.

Oddly the "malformed" packets remain.  Could just be an artifact of Wireshark or passing through the USB device to the VM.

> Edit: I did the capture, look for the file linux_capture_including_QEMU_start.pcapng here.

Thanks.

I'll come back here once I get the balanced codes decoded and have some idea of how the protocol looks like.

jonasmalacofilho, 2021-06-09:
Update: I made a Wireshark dissector that decodes the balanced codes ([`Corsair.AXi.lua`](https://github.com/liquidctl/collected-device-data/blob/master/dissectors/Corsair.AXi.lua)), but I haven't really made any progress so far on the actual protocol (although it's probably PMBus).

xkszltl, 2021-06-24:
> Update: I made a Wireshark dissector that decodes the balanced codes ([`Corsair.AXi.lua`](https://github.com/liquidctl/collected-device-data/blob/master/dissectors/Corsair.AXi.lua)), but I haven't really made any progress so far on the actual protocol (although it's probably PMBus).

https://www.corsair.com/us/en/Categories/Products/Accessories-%7C-Parts/PC-Components/Power-Supplies/AXI-I2C-800mm-PMBus-Cable/p/CP-8920117
[image]
[image]

xkszltl, 2021-06-24:
Emmm...Seems someone put a driver in kernel:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/corsair-psu.c
https://news.ycombinator.com/item?id=27257755

jonasmalacofilho, 2021-06-24:
@xkszltl,

The `corsair-psu` driver only supports HXi and RMi models, which use a different carrier protocol.  They may also use a slightly different subset of PMBus, although we can't know that yet.

chiefgecko, 2021-08-21:
Hi, If this thread moves forward again, I can help with AX1600i testing. 

=====
Bus 001 Device 006: ID 1b1c:1c11 Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c11
  bcdDevice            1.00
  iManufacturer           1 SILABS
  iProduct                2 USB API
  iSerial                 3 AR5W2
  bNumConfigurations      1
  Configuration Descriptor:

chiefgecko, 2021-08-21:
[https://lkml.org/lkml/2020/11/28/241](url)
Some LKML traffic about hwmon & AX1500i/AX1600i from 11/2020

jonasmalacofilho, 2022-01-31:
> I have the AX1200i, and it _kinda_ works with https://github.com/ka87/cpsumon

Just so I'm on the same page, which parts don't work?

> It looks like it's using the same protcol as AX760i, AX860i, and AX1500i
> 
> The dongle for my PSU appears to the system as a usb serial device, and claims to be a "USB to SMB Bridge (Firmware by Ross Fosler)"
> 
> I'm trying to wrap my head around the protocol by making a port to python,

Is there anything I can do to help?

(Also, feel free to reach out in [Discord](https://discord.gg/GyCBjQhqCd) our by email).

> would that be possible to include in this project?

As long as you can adapt it to our device/API model, we'll be super happy to include your work.

> At the moment I don't have a way of monitoring the smbus data the dongle outputs, so it's a bit hit and miss for now.

Despite being a serial bridge (which your OS may or may not have special support for), you can still listen to the USB traffic using Wireshark.  I also made a [Wireshark dissector](https://github.com/liquidctl/liquidctl/issues/332#issuecomment-857433881) to decode the balanced code used on top of the "base" protocol.

audunmg, 2022-01-31:
> > I have the AX1200i, and it _kinda_ works with https://github.com/ka87/cpsumon
> 
> Just so I'm on the same page, which parts don't work?
> 

It can read the correct values most of the time, but it seems like it gets misaligned or ends up in a unknown state after a while and returns nonsense values and the wrong PSU. If i wait a bit, it will output correct values again.

It very rarely outputs values for the 12V rails correctly, if at all, so it crashes before that.

Maybe the solution is here: https://github.com/Hagbard-Celin/cpsumon/commit/2e2ecc4

> Is there anything I can do to help?

That wireshark dissector will probably help a lot :)

Thanks!

> 
> (Also, feel free to reach out in [Discord](https://discord.gg/GyCBjQhqCd) or by email).
> 
> > would that be possible to include in this project?
> 
> As long as you can adapt it to our device/API model, we'll be super happy to include your work.
> 

I hope I'm able to, it would be really great

> > At the moment I don't have a way of monitoring the smbus data the dongle outputs, so it's a bit hit and miss for now.
> 
> Despite being a serial bridge (which your OS may or may not have special support for)

Looks like I'm just lucky and my dongle vid/pid (1b1c:1c00) is in the kernel. the others are probably just missing:
https://github.com/torvalds/linux/blob/83b67041f3eaf33f98a075249aa7f4c7617c2f85/drivers/usb/serial/cp210x.c#L211

If someone on Linux have a dongle with a different pid, maybe they could test loading the cp210x driver and report back if it works as a ttyUSB:

    # modprobe cp210x
    # # List out USB devices with corsair vendor id. If you have more than one, choose the correct one.
    # for f in /sys/bus/usb/devices/*; do if grep -q 1b1c $f/idVendor; then ls $f | grep : ; fi ; done 2> /dev/null 
    1-1.4:1.0
    # echo 1-1.4:1.0 > /sys/bus/usb/drivers/cp210x/bind

maybe the original instructions in https://github.com/ka87/cpsumon/issues/4#issuecomment-358851126 is better

With confirmation that it works, it probably wouldn't be hard getting the missing pids added to the driver.

jonasmalacofilho, 2022-01-31:
> Looks like I'm just lucky and my dongle vid/pid (1b1c:1c00) is in the kernel. the others are probably just missing:
https://github.com/torvalds/linux/blob/83b67041f3eaf33f98a075249aa7f4c7617c2f85/drivers/usb/serial/cp210x.c#L211

Good to know.  But I think we still want to work at the USB level, given that we're cross-platform.

Knowing that's a CP210x (or CP210x-like) could maybe help with the initialization... I'll take a look at past captures before saying more, just to avoid polluting this conversation with noise if it turns out that I'm wrong.

chiefgecko, 2022-02-02:
> > > I have the AX1200i, and it _kinda_ works with https://github.com/ka87/cpsumon
> > 
> > 
> > Just so I'm on the same page, which parts don't work?
> 
> It can read the correct values most of the time, but it seems like it gets misaligned or ends up in a unknown state after a while and returns nonsense values and the wrong PSU. If i wait a bit, it will output correct values again.
> 
> It very rarely outputs values for the 12V rails correctly, if at all, so it crashes before that.
> 
> Maybe the solution is here: [Hagbard-Celin/cpsumon@2e2ecc4](https://github.com/Hagbard-Celin/cpsumon/commit/2e2ecc4)
> 
> > > It looks like it's using the same protcol as AX760i, AX860i, and AX1500i
> > > The dongle for my PSU appears to the system as a usb serial device, and claims to be a "USB to SMB Bridge (Firmware by Ross Fosler)"
> > > I'm trying to wrap my head around the protocol by making a port to python,
> > 
> > 
> > Is there anything I can do to help?
> 
> That wireshark dissector will probably help a lot :)
> 
> Thanks!
> 
> > (Also, feel free to reach out in [Discord](https://discord.gg/GyCBjQhqCd) or by email).
> > > would that be possible to include in this project?
> > 
> > 
> > As long as you can adapt it to our device/API model, we'll be super happy to include your work.
> 
> I hope I'm able to, it would be really great
> 
> > > At the moment I don't have a way of monitoring the smbus data the dongle outputs, so it's a bit hit and miss for now.
> > 
> > 
> > Despite being a serial bridge (which your OS may or may not have special support for)
> 
> Looks like I'm just lucky and my dongle vid/pid (1b1c:1c00) is in the kernel. the others are probably just missing: https://github.com/torvalds/linux/blob/83b67041f3eaf33f98a075249aa7f4c7617c2f85/drivers/usb/serial/cp210x.c#L211
> 
> If someone on Linux have a dongle with a different pid, maybe they could test loading the cp210x driver and report back if it works as a ttyUSB:
> 
> ```
> # modprobe cp210x
> # # List out USB devices with corsair vendor id. If you have more than one, choose the correct one.
> # for f in /sys/bus/usb/devices/*; do if grep -q 1b1c $f/idVendor; then ls $f | grep : ; fi ; done 2> /dev/null 
> 1-1.4:1.0
> # echo 1-1.4:1.0 > /sys/bus/usb/drivers/cp210x/bind
> ```
> 
> maybe the original instructions in [ka87/cpsumon#4 (comment)](https://github.com/ka87/cpsumon/issues/4#issuecomment-358851126) is better
> 
> With confirmation that it works, it probably wouldn't be hard getting the missing pids added to the driver.

[ AX1600i ]
root@pve1:~# lsusb | grep 'Corsair'
Bus 003 Device 005: ID 1b1c:0c12 Corsair H150i Platinum
Bus 001 Device 003: ID 1b1c:1c11 Corsair USB API
root@pve1:~# modprobe -v cp210x && lsmod | grep cp210x
insmod /lib/modules/5.15.7-1-pve/kernel/drivers/usb/serial/cp210x.ko
cp210x                 32768  0
usbserial              57344  1 cp210x
root@pve1:~# for f in /sys/bus/usb/devices/*; do if grep -q 1b1c $f/idVendor; then ls $f | grep : ; fi ; done 2> /dev/null
1-5.1:1.0
3-6:1.0
root@pve1:~# echo "3-6:1.0" 1>/sys/bus/usb/drivers/cp210x/bind
-bash: echo: write error: No such device
root@pve1:~# cat /etc/*release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@pve1:~#
[end]

Tim...

jonasmalacofilho, 2022-02-02:
@chiefgecko, you should also try the other device:

```
$ echo "1-5.1:1.0" 1>/sys/bus/usb/drivers/cp210x/bind
```

I don't think the order in `lsusb` has to match the order in `/sys/bus/usb/devices/*`.

---

P.S. Please try to format your comments a bit better, as your original version was effectively unreadable.

chiefgecko, 2022-02-02:
Sorry, I tried and failed to determine what triggered the strikeout.

Tim...
[snip]
root@pve1:~# echo "1-5.1:1.0" 1>/sys/bus/usb/drivers/cp210x/bind
-bash: echo: write error: No such device
root@pve1:~#
[end]

jonasmalacofilho, 2022-02-02:
@chiefgecko, can you provide the complete output of `lsusb -v -d 1b1c:1c11`?

---

> Sorry, I tried and failed to determine what triggered the strikeout.

The strikethrough was caused by the `~` tildes.  Btw, [Markdown code blocks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks) will render better than just marking with `[snip]` or similar; more information on using Markdown on Github can be found in their [docs](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/about-writing-and-formatting-on-github).

jonasmalacofilho, 2022-02-02:
Thanks!  I'm not sure how relevant it is to the cp210x driver, but it's notable that the AX1600i uses different USB endpoints.

chiefgecko, 2022-02-02:
As a noob poking around ...
CP210x module in kernel 5.15.17-1-pve does not have a modalias entry for 1b1c:1c11

Tim...
```
# cat /sys/bus/usb/devices/1-5.1:1.0/uevent
DEVTYPE=usb_interface
PRODUCT=1b1c/1c11/100
TYPE=0/0/0
INTERFACE=255/0/0
MODALIAS=usb:v1B1Cp1C11d0100dc00dsc00dp00icFFisc00ip00in00
root@pve1:~# modinfo cp210x | grep 'usb:v1B1C'
alias:          usb:v1B1Cp1C00d*dc*dsc*dp*ic*isc*ip*in*
root@pve1:~#
```

chiefgecko, 2022-02-02:
I forked cp210x.c and added usb dongle device id: 1b1c:1c11 ... I have no idea how to request pulls or get a patch accepted. The is end of my know how!

Ciao, Tim...

[https://github.com/torvalds/linux/compare/master...chiefgecko:patch-1](url)

audunmg, 2022-02-03:
There's a good chance the AX1600i is not cp210x, since it's quite new compared to the others.

To know for sure, it would be great to have a trace. The other way would be to look at which driver it uses under Windows, if any. The AX1200i uses a cp210x driver under Windows.

This is what my AX1200i dongle looks like, for comparison:
```
# lsusb -d 1b1c:1c00 -v
Bus 001 Device 004: ID 1b1c:1c00 Corsair Controller for Corsair Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c00 Controller for Corsair Link
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 Corsair Link TM USB Dongle
  iSerial                 3 R26K0297
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 Corsair Link TM USB Dongle
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

chiefgecko, 2022-02-03:
My lsusb output is a few posts up.
You are thinking signature below may not be CP210x?

Over next 1-2 days I will spin up a windows vm and see what transpires.
Tim...
```
  idVendor           0x1b1c Corsair
  idProduct          0x1c11
  bcdDevice            1.00
  iManufacturer           1 SILABS
  iProduct                2 USB API
  iSerial                 3 AR5W2
```

audunmg, 2022-02-03:
After looking at the windows driver, I'm pretty sure it's a cp210x chip, but for some reason your driver doesn't seem to attach to your dongle, and I'm not really sure why. 

We were thinking to try working around that driver anyway, and getting a packet dump would really help there.

For some reason my PSU resets my computer when I try to pass the USB device through to a windows VM, so maybe be prepared that it might happen.

chiefgecko, 2022-02-03:
https://github.com/torvalds/linux/blob/3f667b5d4053ad54aee13dab5c94f04ff75ddfdf/drivers/usb/serial/cp210x.c
As of the latest commit of cp210x.c (link above) on Nov 26, 2021 and [1c1b:1c11] is not in the usb device list.

Hence my novice attempt to fork CP210x and add [1c1b:1c11]  to usb device list. 
I have no Github experience with forks/pulls and no idea how to request pull for my patch.

G'nite

jonasmalacofilho, 2022-02-03:
@audunmg 

> There's a list of usb ids and descriptions in the windows driver which use the Corsair-provided Silicon Labs SiUSBXp driver

It's interesting that all known AXi models use the same driver, but it's also notable that the v2/Pro coolers also appear on the same list: while I knew they used a bridge from Silicon Labs, I don't think it's necessarily the CP210x; and the captured control transfers from a AX1500i do look a lot different than what we see when analyzing one of those coolers.

That said, this still means that this driver will not be very far from other drivers we already have (`asetek` and `asetek_pro`), which simplifies things.

---

@chiefgecko 

> Hence my novice attempt to fork CP210x and add [1c1b:1c11] to usb device list.

Have you compiled a new module and tested it?  Were you able to bind the patched kernel driver to your PSU?

audunmg, 2022-02-03:
You're probably right, the AX1200i I have is very old compared to the newer PSUs, especially AX1600i.

Sorry for jumping to conclusions there.

jonasmalacofilho, 2022-02-13:
Before it gets lost in noise, from Discord: https://github.com/audunmg/wip-liquidctl-ax1200i

chiefgecko, 2022-02-17:
Secured a Win10 laptop, ordered a usb mini b cable, installed iCue 4.20.169, and AX1600i P/S shows up in iCue.
Win10 device info:
USB Device Name: [Corsair USBXp Driver]
Device Description: [AX1600i]
Device Id: [USB\VID_1B1C&PID_1C11\AR5W2]
Driver Info: 7/14/2017, V3.3.0

I will spin up a Linux VM soon try to remember how to install kernel source. Last time was 15y ago.

Remind me what was the endgame under Win10?
Sniff usb traffic?

Tim...

user-d9dfd17, 2022-04-02:
I have done some research on this with AX1600i, Windows 10, Corsair Link 4, QEMU and USB passthrough.

- The driver is [Silicon Labs USB Express](https://www.silabs.com/search#q=usb%20express)
- Capture can be done via [/sys/kernel/debug/usb/usbmon](https://www.kernel.org/doc/html/latest/usb/usbmon.html)
- There are a couple of workarounds; [here](http://www.etheus.net/SiUSBXp_Linux_Driver) and [here](https://github.com/fMeow/silabs_usb_xpress)

Example output:

```
ffff927ea57fde00 1158834213 S Bi:3:004:2 -115 4096 <
ffff927ea57fde00 1158839736 C Bi:3:004:2 0 2 = a800
ffff927ea55c4780 1158839801 S Bi:3:004:2 -115 4096 <
ffff927ea55c4780 1158844870 C Bi:3:004:2 0 2 = a800
ffff927ea55c5e00 1158844920 S Bi:3:004:2 -115 4096 <
ffff927ea55c5e00 1158847552 C Bi:3:004:2 0 4 = a8555500
ffff927ea55c5200 1158847593 S Bi:3:004:2 -115 4096 <
ffff927ea55c5200 1158849633 C Bi:3:004:2 0 16 = a8566595 66565a69 5a555a55 5a966900
ffff927ea55c5bc0 1158849669 S Bi:3:004:2 -115 4096 <
ffff927ea55c5bc0 1158858396 C Bi:3:004:2 0 2 = a800
ffff927ea55c5c80 1158858456 S Bi:3:004:2 -115 4096 <
ffff927ea55c5c80 1158860856 C Bi:3:004:2 0 4 = a8555500
ffff927ea55c5440 1158860901 S Bi:3:004:2 -115 4096 <
ffff927ea55c5440 1158862832 C Bi:3:004:2 0 16 = a8566595 66565a69 5a555a55 5a966900
ffff927ea55c5d40 1158862867 S Bi:3:004:2 -115 4096 <
ffff927ea55c5d40 1158867059 C Bi:3:004:2 0 2 = a800
ffff927ea55c4600 1158867109 S Bi:3:004:2 -115 4096 <
ffff927ea55c4600 1158869509 C Bi:3:004:2 0 4 = a8555500
ffff927ea55c46c0 1158869542 S Bi:3:004:2 -115 4096 <
ffff927ea55c46c0 1158871155 C Bi:3:004:2 0 8 = a8565556 55555500
ffff927ea55c5140 1158871187 S Bi:3:004:2 -115 4096 <
ffff927ea55c5140 1158873661 C Bi:3:004:2 0 2 = a800
ffff927ea55c5a40 1158873701 S Bi:3:004:2 -115 4096 <
ffff927ea55c5a40 1158875145 C Bi:3:004:2 0 4 = a8555500
ffff927ea55c4180 1158875179 S Bi:3:004:2 -115 4096 <
ffff927ea55c4180 1158876733 C Bi:3:004:2 0 8 = a8565556 55555500

```

jonasmalacofilho, 2022-04-02:
Thanks @user-d9dfd17 ,

> * The driver is [Silicon Labs USB Express](https://www.silabs.com/search#q=usb%20express)

Indeed.  But it's important to remember that's very low level and only handles the very general USB to serial bridge these devices use.  There are more protocol layers on top to understand before we can get to actual hardware monitoring data.

Another thing is that the captures I've seen so far of these PSUs show a slightly different behavior than what I'm used to seeing from other devices that use the Silicon Labs USB Express driver, or from Craig Shelley's implementation (that you also linked).

By the way, we already support a few devices that normally use that same Silicon Labs Windows driver: [`asetek`](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/asetek.py) and [`asetek_pro`](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/asetek_pro.py).

> * Capture can be done via [/sys/kernel/debug/usb/usbmon](https://www.kernel.org/doc/html/latest/usb/usbmon.html)

I suggest you use Wireshark as a frontend to it, and its pcap format for storage of the captures.

A [few comments above](https://github.com/liquidctl/liquidctl/issues/332#issuecomment-857433881) you can also find a Wireshark dissector that I started for this device.

> * There are a couple of workarounds; [here](http://www.etheus.net/SiUSBXp_Linux_Driver) and [here](https://github.com/fMeow/silabs_usb_xpress)

Thanks for linking them, the Rust one is new to me, and both are helpful references.

But on our end, that's just a very minor part in writing our higher level driver, as you can see in our `asetek` and `asetek_pro` drivers.

user-d9dfd17, 2022-04-02:
Thank you for the overview. I was not able to get the cp210x serial driver running:
```
[    6.034293] usbserial: USB Serial support registered for cp210x
[    6.034312] cp210x 3-2:1.0: cp210x converter detected
[    6.038064] cp210x ttyUSB0: failed get req 0x4 size 2 status: -32
[    6.038072] cp210x: probe of ttyUSB0 failed with error -32
```

I now have tshark installed and can provide captures if needed.

user-d9dfd17, 2022-04-08:
Does anyone know where to find more information on the encoding/decoding tables and process used in [Corsair.AXi.lua](https://github.com/liquidctl/collected-device-data/blob/master/dissectors/Corsair.AXi.lua) and [cpsumon.c](https://github.com/ka87/cpsumon/blob/fd639684d7f9fe7aef9c8662d5f91e3fc8745e52/libcpsumon/include/cpsumon.h#L81-L100)?

jonasmalacofilho, 2022-04-08:
@user-d9dfd17 

If it helps, it's called balanced code.

jonasmalacofilho, 2022-04-08:
@user-d9dfd17

Also take a look at:

- Knuth.  _[Efficient Balanced Codes](https://ieeexplore.ieee.org/document/1057136)._  1986.
- Scripcariu. _[The Design of Perfectly balanced Constant Weight Substitution Codes](https://www.bulipi-eee.tuiasi.ro/archive/2017/fasc.4/p4_f4_2017.pdf)._ 2017.[^1]

[^1]: The exact code used by these PSUs is on Table 2.

user-d9dfd17, 2022-04-09:
That's interesting, both efficient and information-theoretically secure. The question though, is it possible to reuse the decoding from the example? I can get the same output as from the decode method of [cpusumon.c](https://github.com/ka87/cpsumon/blob/master/libcpsumon/src/cpsumon.c#L80-L101) with some tweaking. There is a difference though, for the dissector, decoding starts at 0 while in the example at index 1.

jonasmalacofilho, 2022-04-09:
@user-d9dfd17 ,

(It's been a while since I looked at cpsumon in detail but) I think that's because it access the PSU as a serial device, while the Wireshark dissector is at the lower abstraction level of the USB HID.  And HID reports can have report numbers.

This is also relevant when implementing liquidctl drivers, as liquidctl also access USB HIDs at the HID layer, and the APIs for that also have to account for the report numbers.

Good references for this are the HID spec, as well as the documentation for hidapi and Linux hidraw:

- [Device Class Definition for HID 1.11](https://www.usb.org/document-library/device-class-definition-hid-111)
- [libusb/hidapi](https://github.com/libusb/hidapi/), [(documented) `hidapi.h`](https://github.com/libusb/hidapi/blob/master/hidapi/hidapi.h)
- [(Linux) HIDRAW - Raw Access to USB and Bluetooth Human Interface Devices](https://www.kernel.org/doc/html/latest/hid/hidraw.html)

Tangentially related: #312

okias, 2023-08-27:
Hello! I have interest in having AX860i supported, if anyone come up with some architecture fir AX*i, I would be happy to RE serial/usb communication from Win and contribute support, thou don't ask me for the designing of the interface :)

Thank you

SkeletonGamer, 2026-09-10:
# AX1500i: protocol worked out, and it only needs a one-line kernel patch

I have an AX1500i working under Linux, reading input voltage/current, input and output
power, internal temperature and fan speed. The captures @keckj contributed to
`collected-device-data` were the key: everything below is derived from them plus
measurements on my own unit, so it is verified against two different AX1500i units.

## 1. It is a CP2103, and `cp210x` only needs the ID

`1b1c:1c02` is a CP2103 (`GET_PARTNUM` returns `0x03`). With the ID added to
`drivers/usb/serial/cp210x.c`, a plain `/dev/ttyUSB*` at **115200 8N1** is all that is
needed. No libusb, no driver detaching:

```
/sys/bus/usb/devices/3-14/idVendor       1b1c
/sys/bus/usb/devices/3-14/idProduct      1c02
/sys/bus/usb/devices/3-14/product        Corsair Link TM USB Dongle
/sys/bus/usb/devices/3-14/manufacturer   Silicon Labs
/sys/bus/usb/devices/3-14:1.0/driver ->  cp210x
/dev/ttyUSB0
```

A one-line patch adding the ID has been **submitted to linux-usb** (not merged yet):

https://lore.kernel.org/linux-usb/20260910162609.7373-1-jeremy.carrat@icloud.com/

It is also reproduced at the end of this comment.

**Correction to an earlier comment in this thread:** the reports of
`echo "…:1.0" > /sys/bus/usb/drivers/cp210x/bind` failing with `No such device` are a
methodology artifact. `bind` only works for IDs already in the driver's table; to add one at
runtime you need `new_id`:

```sh
sudo modprobe cp210x
echo 1b1c 1c02 | sudo tee /sys/bus/usb-serial/drivers/cp210x/new_id
```

So the AX1600i (`1b1c:1c11`) was very likely never actually tested. It is probably the same
kind of device, but I cannot confirm it: I do not have one.

**Important, and it cost me a day:** if the bridge does not answer, **power-cycle the PSU at
the mains** (rear switch off, cord unplugged ~30 s). The CP2103 is bus-powered and enumerates
and answers all its control requests even when the Corsair Link side is wedged, so the device
looks perfectly healthy while being completely mute. A USB reset, a different port and a
different hub all changed nothing; only removing AC did.

## 2. Initialization: iCUE sends exactly five vendor requests

From the capture. Note it **never touches DTR/RTS**, and `IFC_ENABLE` uses `0xFFFF`, not
`0x0001` as the kernel driver does (either works):

| bRequest | Name | wValue | Data |
|---|---|---|---|
| `0x00` | `IFC_ENABLE` | `0xFFFF` | none |
| `0x01` | `SET_BAUDDIV` | `0x2000` | none |
| `0xFF` | `VENDOR_SPECIFIC` / `GET_PARTNUM` | `0x370B` | reply `03` |
| `0x12` | `PURGE` | `0x000C` | none |
| `0x1E` | `SET_BAUDRATE` | `0x0000` | `00 c2 01 00` (115200) |

No `SET_LINE_CTL`, no `SET_MHS`, no `SET_FLOW`.

## 3. Framing

The balanced code is the one your dissector already decodes (569a nibbles). What is worth
adding:

- request frames start with **`0x54`**, replies with **`0xA8`**;
- payload is 2 bytes per useful byte, **low nibble first**;
- frames end with **`0x00`**.
[... 140 lines cut ...]
 	{ USB_DEVICE(0x1B1C, 0x1C00) }, /* Corsair USB Dongle */
+	{ USB_DEVICE(0x1B1C, 0x1C02) }, /* Corsair AX1500i Power Supply */
 	{ USB_DEVICE(0x1B93, 0x1013) }, /* Phoenix Contact UPS Device */
 	{ USB_DEVICE(0x1BA4, 0x0002) },	/* Silicon Labs 358x factory default */
 	{ USB_DEVICE(0x1BE3, 0x07A6) }, /* WAGO 750-923 USB Service Cable */
```

## 10. A note on implementing this in liquidctl

One tension worth flagging. liquidctl talks to devices over raw USB with pyusb, to stay
cross-platform. Putting the ID in `cp210x` makes the kernel claim interface 0 and expose a
`/dev/ttyUSB*` instead. Both routes work, but **not at the same time**: only one thing can hold
the interface. I hit this directly, and it is easy to handle: a libusb program has to call
`detach_kernel_driver(0)` first, and can call `attach_kernel_driver(0)` afterwards to hand the
tty back.

That does not block a liquidctl driver, because the raw-bulk route is entirely self-contained.
liquidctl can issue the five CP210x vendor control requests from section 2 itself, then do
ordinary bulk writes and reads on endpoints `0x01` and `0x81` using the framing in sections 3
and 4. No tty, no kernel driver, and it stays cross-platform. That is exactly how my own
libusb implementation works, and it reads the same values as the tty route.

So the kernel patch is for people who just want a `ttyUSB` and a short script. It is not a
prerequisite for a liquidctl driver, and the two coexist fine. I am happy to help with either
one.