liquidctl/liquidctl #334: Not recognizing any devices

pmleffers, 2021-04-23:
Have a number of RGB devices including a Kraken X52 water pump.
Installed version liquidctl-1.6.0-bin-windows-x86_64.zip, didn't see driver in HID's so installed WinUSB driver from Zadib. Already setup path and liquidctl can execute from path.  Also manually installed the libusb files from libusb/releases. On the face of it liquidctl appears to work; however I can't ever seem to get it to recognize my watercooler. I have tried uninstalling the driver from zadib and gone back to default, and re-installed multiple times. Every time the same problem, liquidctl doesn't recognize any devices.

jonasmalacofilho, 2021-04-23:
>  I have tried uninstalling the driver from zadib and gone back to default

That's the right setup for this and most other devices supported by liquidctl.

> liquidctl doesn't recognize any devices.

Can you show me the complete output from `liquidctl list --debug`?

---

P.S. In the future, please don't bypass the appropriate issue template when reporting possible bugs.

jonasmalacofilho, 2021-04-23:
> didn't see driver in HID's

I just noticed that I don't really know what you meant by that.  Can you clarify?

pmleffers, 2021-04-24:
[DEBUG] liquidctl.cli: running liquidctl v1.6.0
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 413c:2107
[DEBUG] liquidctl.driver.usb: found HID device 045e:0813
[DEBUG] liquidctl.driver.usb: found HID device 045e:0813
[DEBUG] liquidctl.driver.usb: found HID device 045e:0813
[DEBUG] liquidctl.driver.usb: found HID device 045e:0813
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18ec
[DEBUG] liquidctl.driver.usb: found HID device 045e:0813
[DEBUG] liquidctl.driver.usb: found HID device 22d4:131e
[DEBUG] liquidctl.driver.usb: found HID device 045e:028e
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18ec
[DEBUG] liquidctl.driver.usb: found USB device 045e:028e
[DEBUG] liquidctl.driver.usb: found USB device 1022:145f
[DEBUG] liquidctl.driver.usb: found USB device 0bda:8771
[DEBUG] liquidctl.driver.usb: found USB device 1022:145f
[DEBUG] liquidctl.driver.usb: found USB device 413c:2107
[DEBUG] liquidctl.driver.usb: found USB device 1e71:170e
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0620
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 22d4:131e
[DEBUG] liquidctl.driver.usb: found USB device 1b21:2142
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ad6
[DEBUG] liquidctl.driver.usb: found USB device 1022:43ba
[DEBUG] liquidctl.driver.usb: found USB device 10de:1ad6
[DEBUG] liquidctl.driver.usb: found USB device 0a12:1004

jonasmalacofilho, 2021-04-24:
The log shows that, at the OS level, the device is found *only* as a USB device, but not as HID.  This suggests that the correct driver (Microsoft Generic HID Driver, IIRC) still hasn't been restored, even though you already tried to remove the WinUSB driver.  

Try doing that through Device Manager: you need to find the correct device, and "Uninstall" the current/incorrect driver.

> I'm looking for is on address 1e71:170e although I'm not sure how I can tell liquidctl how to connect the device via the address.

Those are actually the vendor and product IDs, not an address.

You can tell liquidctl to *filter* only the devices that match such IDs with `--vendor` and `--product`, and you can filter devices on a specific `--bus` or `--address`.  *But these options don't make liquidctl connect to a device, they only ask it to ignore devices that don't match these criteria.*

Your problem is that that device isn't available at the OS level (on the correct bus), and liquidctl can't override the OS.

We try in more than once place to clarify that Zadig should only be used for very specific devices, but clearly there are still some issues in our documentation.  Do you have any suggestions of what we could change there that could have prevented you from replacing the driver when that was unnecessary and undesirable?

---

I would still like to understand what you meant by:

> didn't see driver in HID's

jonasmalacofilho, 2021-04-24:
> Do you have any suggestions of what we could change [...]

I forgot to reference the issue in the commit message, but ffb2d87b6121 already makes the most important notes stand out (from less important notes) in the table of supported devices.

jonasmalacofilho, 2021-04-29:
@pmleffers, let me know if you managed to get this sorted out.

closed 2021-05-01 by jonasmalacofilho