liquidctl/liquidctl #337: corsair h150i pro - pump profile - invalid literal int base 10

fallen1011, 2021-04-28:
Please describe the issue or request as clearly and completely as possible.
can you please provide the proper syntax for setting a h150i pro pump mode/profile to performance. Mine is stuck on balanced. Ive initialized all, but when i attempt to set it, i get invalid literal int base 10. ive read the man pages and the documentation here, i cannot get the syntax right to get it to set the pump profile to performance mode /extreme.

MarshallAsch, 2021-04-28:
Would you be able to provide the command that you tried and the output it provided?

Also I believe that [this file](https://github.com/liquidctl/liquidctl/blob/main/docs/asetek-pro-guide.md) has the documentation for the device that you are using.

fallen1011, 2021-04-28:
i figured it out, corsair labels the profile as extreme, and the doc mentioned extreme. however it only allows performance mode. im guessing those are one in the same? i could be wrong. but i was able to set it with liquidctl initialize --pump-mode performance

jonasmalacofilho, 2021-04-28:
Reopening becaude I noticed a small typo in the documentation, that I can't fix right now.

Also, where in the does is "extreme" mentioned?

commit f6a77027 (2021-04-28): docs: asetek_pro: fix typo in pump mode

Closes: #337
closed 2021-04-28 by jonasmalacofilho