liquidctl/liquidctl #339: Support for Corsair Hydro H115i Platinum is incomplete

andrejpodzimek, 2021-05-16:
```
$ liquidctl --vendor=0x1b1c --product=0x0c17 status --verbose --debug
[DEBUG] liquidctl.cli: running liquidctl v1.6.1 (Arch Linux; 1.6.1-1)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-20
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-10
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-19
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-17
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-15
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-13
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-21
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-11
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-18
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-16
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-14
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-12
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c17
[DEBUG] liquidctl.driver.hydro_platinum: instanced driver for Corsair Hydro H115i Platinum (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c17
[DEBUG] liquidctl.cli: device: Corsair Hydro H115i Platinum (experimental)
[DEBUG] liquidctl.keyval: data in /run/user/1984/liquidctl/vid1b1c_pid0c17/loc0
[DEBUG] liquidctl.keyval: loaded sequence=8 (from /run/user/1984/liquidctl/vid1b1c_pid0c17/loc0/sequence)
[DEBUG] liquidctl.keyval: replaced with sequence=9 (stored in /run/user/1984/liquidctl/vid1b1c_pid0c17/loc0/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:48:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:f8:11:0f:00:f0:01:50:1d:00:00:a9:e8:03:a9:42:05:00:a1:e8:03:a1:16:05:01:aa:00:00:aa:18:09:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:21
Corsair Hydro H115i Platinum (experimental)
├── Liquid temperature    29.3  °C
├── Fan 1 speed           1346  rpm
├── Fan 1 duty              66  %
├── Fan 2 speed           1302  rpm
├── Fan 2 duty              63  %
├── Pump speed            2328  rpm
└── Pump duty               67  %
```

On this^^^ device, the only supported LED modes are `fixed` and `superfixed`. This makes it impossible to (e.g.) configure something similar to the factory defaults with “rainbows” circling around the cooler and fans. No animated LED modes are supported. It is highly likely that this device has some `breathing` / `blinking` / `fading` / `rainbow` modes (such as the factory default).

The (two) supported modes work as expected. For example, this sets the cooler to red, one fan to green and another fan to blue:

```
liquidctl --vendor=0x1b1c --product=0x0c17 set led color super-fixed ff0000{,,,,,,,,,,,,,,,} 00ff00{,,,} 0000ff{,,,}
```

Additional information about the system:

- Affected device: Corsair Hydro H115i Platinum
- Operating system: Arch Linux
- Liquidctl version: Arch Linux `liquidctl` package, version `1.6.1-1`

If I can help by trying some (as of yet) unsupported HID commands on the device, I’ll be happy to give it a try.

jonasmalacofilho, 2021-05-16:
I think this is a duplicate of #193.

But based on the info there, it's still unclear whether other hardware modes exist, other than fixed and rainbow

MarshallAsch, 2021-05-17:
I've got some of this done in my testing branch, ill try yo get an update with something tomorrow night. If you dont hear from me by Wednesday please poke me again. 

Hardware modes do exist, and they cover way more options than what icue shows. Ill try to limit the inital implementation to a few default modes rather than every at once.

MarshallAsch, 2021-05-20:
Im going to try to take a look at this tomorrow evening. And add a couple of the preset hardware modes.

jonasmalacofilho, 2021-05-30:
Closing this as it is, indeed, a duplicate of #193.

@MarshallAsch, take you time, I know you're pretty busy these days...

closed 2021-05-30 by jonasmalacofilho