liquidctl/liquidctl #342: extra/yoda TypeError on execute

S4ndf1re, 2021-05-19:
Today i tried using extra/yoda for fan speed control but instead of executing normally, this happend:
```
...$ yoda --match grid show-sensors
...$ TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
```
I have installed liquidctl as described and is worked as expected. No errors

I debugged a little and it seems that the exception gets raised while parsing --pick (i don't think that I have to include it while running yoda)

Does anyone have a clue about this?

S4ndf1re, 2021-05-19:
I tried that before, but the output is just the doc text. Maybe I am using --debug the wrong way?

```
...$ yoda --debug --match grid show-sensors
Usage:
  yoda [options] show-sensors
  yoda [options] control (<channel> with <profile> on <sensor> [and])...
  yoda --help
  yoda --version
Options:
  --interval <seconds>     Update interval in seconds [default: 2]
  -m, --match <substring>  Filter devices by description substring
  -n, --pick <number>      Pick among many results for a given filter
  --vendor <id>            Filter devices by vendor id
  --product <id>           Filter devices by product id
  --release <number>       Filter devices by release number
  --serial <number>        Filter devices by serial number
  --bus <bus>              Filter devices by bus
  --address <address>      Filter devices by address in bus
  --usb-port <port>        Filter devices by USB port in bus
  -v, --verbose            Output additional information
  -g, --debug              Show debug information on stderr
  --hid <module>           Override API for USB HIDs: usb, hid or hidraw
  --legacy-690lc           Use Asetek 690LC in legacy mode (old Krakens)
  --version                Display the version number
  --help                   Show this message
Examples:
  yoda --match grid show-sensors
  yoda --match grid control fan1 with '(20,20),(35,100)' on nct6793.systin
  yoda --match kraken show-sensors
  yoda --match kraken control pump with '(20,50),(50,100)' on istats.cpu and fan with '(20,25),(34,100)' on _internal.liquid
Requirements:
  all platforms  liquidctl, including the Python APIs (pip install liquidctl)
  Linux/FreeBSD  psutil (pip install psutil)
  macOS          iStats (gem install iStats)
  Windows        none, system sensors not yet supported
Changelog:
  0.0.4  Fix casing of log and error messages
  0.0.3  Remove duplicate option definition
  0.0.2  Add low-pass filter and basic error handling.
  0.0.1  Generalization of krakencurve-poc 0.0.2 to multiple devices.
Copyright (C) 2020–2021  Jonas Malaco
SPDX-License-Identifier: GPL-3.0-or-later
```

jonasmalacofilho, 2021-05-19:
That definitively isn't right.

I can't reproduce it

```
yoda --debug --match grid show-sensors                          127 ↵
[DEBUG] __main__: yoda v0.0.4
[DEBUG] __main__: liquidctl v1.6.0
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: ignoring 0-0037, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: ignoring 0-0053, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[...]
```

so it seems that `docopt` is broken on your system.

Can you show me the output from `pip show docopt`?

S4ndf1re, 2021-05-19:
```
Name: docopt
Version: 0.6.2
Summary: Pythonic argument parser, that will make you smile
Home-page: http://docopt.org
Author: Vladimir Keleshev
Author-email: vladimir@keleshev.com
License: MIT
Location: /home/jank/.local/lib/python3.8/site-packages
Requires: 
Required-by: liquidctl

```

jonasmalacofilho, 2021-05-19:
And you haven't modified `yoda` in any way?

S4ndf1re, 2021-05-19:
No. But i can try to clone it again. I mean that shouldn't change much, but I can try.

S4ndf1re, 2021-05-19:
Ok now it works for some magical reason. I just cloned the repository again, and it works just fine. Thank you for your time though.

closed 2021-05-19 by S4ndf1re

jonasmalacofilho, 2021-05-19:
I'm just glad to hear that it's working now : )