liquidctl/liquidctl #344: Suddenly a working liquidctl results in ...

RCTORONTO, 2021-05-27:
ImportError: /usr/lib/x86_64-linux-gnu/libhidapi-libusb.so.0: undefined symbol: libusb_handle_events_completed

RCTORONTO, 2021-05-27:
sudo apt-get install --reinstall libhidapi-dev libhidapi-libusb0 libhidapi-hidraw0 python3-hidapi 

no dice.. does this project need to be updated to work with current revisions of it's dependencies?

RCTORONTO, 2021-05-27:
does anyone have a simple utility to turn on the fan in the rmi series power supplies - stupidly the default state is off, which is likely why corsair no longer is manufacturing "smart" usb interfaces on their power supplies (my guess is that a non-spinning fan caused some failures)

jonasmalacofilho, 2021-05-28:
>  I'd be willing to pay for a utility written in c/c++ that seeks to only support the fans.

I can write you a statically linked binary in Rust (or, if you must, C) that does that, while only depending on the kernel hidraw ABI remaining stable (and that interface not being disabled by a different-than-default kernel driver).

Write me an email and we'll discuss details and price.

closed 2021-05-28 by jonasmalacofilho

RCTORONTO, 2021-05-28:
Hi Jonas, how much would I be looking at for a simple command line utility written in c that can set the fans for the corsair power supplies ?  I'd have a whack at it myself but I can't be bothered to parse though liquidctls python mania to get at the meat and potatoes

Closed #344<https://github.com/liquidctl/liquidctl/issues/344>.

jonasmalacofilho, 2021-05-28:
> Hi Jonas, how much would I be looking at for a simple command line utility written in c that can set the fans for the corsair power supplies ?  I'd have a whack at it myself but I can't be bothered to parse though liquidctls python mania to get at the meat and potatoes

1000 USD.

RCTORONTO, 2021-05-28:
> For the record though, your distro appears to be broken: libhidapi-libusb.so.0 has been compiled to require a symbol not supplied by libusb, at least not in the versions your distro has shipped you.

Its also quite possible my server this is to he used on has shared object issues, its been upgraded in place with dist upgrades since I think Ubuntu 14 - currently its 18.04 (ive been holding off on 20/21 because I'm still getting my head around all the changes introduced in Ubuntu 20 with the adoption of netplan and other things....

No, mainly because it doesn't need to be compiled.

And it's also tested with dependencies substantially ahead what's usually in Debian/Ubuntu.

RCTORONTO, 2021-05-28:
Its weird because it was working fine, I had a conky that showed me the PS details on the desktop and until a couple of days ago it worked, I blame the apt package manager its always screwing me

RCTORONTO, 2021-05-28:
Hahahhjajahjahhaha I can't stop laughing , I'll spin the goddam fan by hand

jonasmalacofilho, 2021-05-28:
Good idea!