liquidctl/liquidctl #346: Corsair Obsidian 1000D

FellTarsier, 2021-05-30:
Corsair's Obsidian 1000D case comes with a slight variation on the Commander Pro.  They're essentially the same aside from the branding being different so that it shows up in iCue as the 1000D.

The device appears to support the following features:

 - [x] monitoring of temperatures or other environment sensors
 - [x] monitoring of fan or pump speeds or duty cycles
 - [x] monitoring of voltages, currents or power
 - [x] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [x] configurable lighting of embedded LEDs
 - [x] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [x] USB
- [ ] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ ] HID
- [x] USB
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [ ] unknown

`lsusb -v` output:

```Bus 003 Device 011: ID 1b1c:1d00 Corsair 1000D
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1d00 
  bcdDevice            0.09
  iManufacturer           1 Corsair
  iProduct                2 1000D
  iSerial               128 0105003D83211E20
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

---

Finally, I can help with:¹

- [x] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [x] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

I was able to get liquidctl to work with it by adding the following code
```        (0x1b1c, 0x1d00, None, 'Corsair Obsidian 1000D (experimental)',
            {'fan_count': 6, 'temp_probs': 4, 'led_channels': 2}),
``` 
to commander_pro.py  and then deleting this line a little further down.
```
        if self.device.product_id != 0x0c10:
```

I'm not great at python otherwise I'd fix the hack and submit a PR instead.  I was also racing against the clock to get the fans to spin before my CPU overheated and crashed again.

jonasmalacofilho, 2021-05-30:
Thanks!

Already added it to the `main` branch, can you give that a test?

closed 2021-05-30 by jonasmalacofilho

FellTarsier, 2021-05-30:
@jonasmalacofilho can confirm it works and the unit tests pass.

```
================================================================= 301 passed in 3.13s =================================================================
Device #0: Corsair Obsidian 1000D (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x1d00
├── Release number: 0x0009
├── Serial number: <removed>
├── Bus: hid
├── Address: /dev/hidraw9
└── Driver: CommanderPro

Device #1: Corsair Commander Pro (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c10
├── Release number: 0x0009
├── Serial number: <removed>
├── Bus: hid
├── Address: /dev/hidraw10
└── Driver: CommanderPro
```

Thanks for getting this support added so quickly!

jonasmalacofilho, 2021-05-30:
You did most of the work! : )

Thanks again!

commit 9057ba2e (2021-05-30): commander_pro: enable support for Corsair Obsidian 1000D

As reported in #346:

> Corsair's Obsidian 1000D case comes with a slight variation on the
> Commander Pro. They're essentially the same aside from the branding
> being different so that it shows up in iCue as the 1000D.

Fixes: #346